espressif/esp-idf #4567: phy_init_data not building with CMake

ducalex, 2020-01-01:
## Environment

- Development Kit:      none
- Kit version (for WroverKit/PicoKit/DevKitC): n/a
- Module or chip used:  ESP32-WROVER-B
- IDF version: v3.3.1
- Build System:         CMake
- Compiler version: crosstool-ng-1.22.0-80-g6c4433a5
- Operating System:     Windows
- Power Supply:         USB

## Problem Description

The build fails when PHY calibration data is selected to be built. It works correctly with the make build system but I understand it is being deprecated.

### Expected Behavior

Successful build.

### Actual Behavior

Failure to find sdkconfig.h

### Steps to repropduce
1. Enable `CONFIG_ESP32_PHY_CALIBRATION_AND_DATA_STORAGE=y`
   `CONFIG_ESP32_PHY_INIT_DATA_IN_PARTITION=y`
2. idf.py build

### Code to reproduce this issue

None needed

## Debug Logs

```
[185/829] Generating ../../esp32/phy_init_data.bin
FAILED: esp32/phy_init_data.bin
cmd.exe /C "cd /D D:\ESP32\project\build\esp-idf\esp32 && C:\Users\alex\Apps\esp-idf\tools\tools\xtensa-esp32-elf\1.22.0-80-g6c4433a5-5.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe -x c -c -I C:/Users/alex/Apps/esp-idf/3.3/components/esp32 -I C:/Users/alex/Apps/esp-idf/3.3/components/esp32/include -I D:/ESP32/project/build -o phy_init_data.obj C:/Users/alex/Apps/esp-idf/3.3/components/esp32/phy_init_data.h && C:\Users\alex\Apps\esp-idf\tools\tools\xtensa-esp32-elf\1.22.0-80-g6c4433a5-5.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-objcopy.exe -O binary phy_init_data.obj D:/ESP32/project/build/esp32/phy_init_data.bin"
C:/Users/alex/Apps/esp-idf/3.3/components/esp32/phy_init_data.h:18:23: fatal error: sdkconfig.h: No such file or directory
```

## Other items if possible
I have found a similar bug report referencing the issue elsewhere:     
https://sysprogs.com/w/forums/topic/unable-to-locate-sdkconfig-h-when-generating-phy_init_data-bin/

renzbagaporo, 2020-01-08:
Sorry for the late reply @ducalex. Can you try out this patch?
[0001-esp32-fix-generation-of-phy_init_data.bin.txt](https://github.com/espressif/esp-idf/files/4034118/0001-esp32-fix-generation-of-phy_init_data.bin.txt)

ducalex, 2020-01-08:
@renzbagaporo Thank you, it works!

Alvin1Zhang, 2020-01-13:
@ducalex Thanks for the updates.

closed 2020-01-13 by Alvin1Zhang