espressif/esp-idf #4568: Name servece lookup fails with ethernet when using static IP

AxelLin, 2020-01-02:
[ethernet_static_ip_dns.txt](https://github.com/espressif/esp-idf/files/4014829/ethernet_static_ip_dns.txt)

## Environment

- Development Kit: ESO32-Ethernet-Kit_A_V1.1
- Module or chip used: ESP32-WROVER-B
- IDF version: v4.1-dev-1572-g30372f5a4ff2
- Build System: CMake
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System:  Linux
- Power Supply:  USB

## Problem Description

I test it using examples/ethernet/basic.
The attached patch modify the example to disable dhcpc and use static IP,
then use dns_setserver to set DNS server.
With the patched code, the getaddrinfo() always fails.
If you comment out the "#define DISABLE_DHCP 1" line, then getaddrinfo() works.

Also note, if I call dns_setserver when dhcpc is enabled, the getaddrinfo() also fails.
The static ip settings are hard-coded in the test patch, you may need to change it.

### Expected Behavior
1. When using static IP, it should be ok to set DNS via dns_setserver()
2. Even when using dhcpc, it should be ok to change DNS via dns_setserver()

### Actual Behavior
getaddrinfo() always fails if dns_setserver() is called.

### Code to reproduce this issue
see attached file.

negativekelvin, 2020-01-02:
https://github.com/espressif/esp-idf/issues/705

AxelLin, 2020-01-02:
@negativekelvin 
I know that and I tried setting DNS at various places. It does NOT work.
BTW, it works if using wifi interface.

AxelLin, 2020-01-20:
Looks like bug in my application code.

closed 2020-01-20 by AxelLin