espressif/esp-idf #4569: wifi&blemesh Coexist

wivin001, 2020-01-02:
能否给一个blemesh中hsl ctl 等其他model的demo，目前最新的sdk依然不太方面自己创建一个模型，需要较高的水平才能游刃有余的写一个model出来。能否提供一个参考？而不仅仅是onoff的模型？目前移植了之前代码到最新的github的sdk上有出现以下的报错：
[0;31mE (1389) BLE_MESH: generic_server_init, NULL Generic OnPowerUp State[0m
[0;31mE (1389) BLE_MESH: generic_server_init, NULL Generic OnPowerUp State[0m
[0;31mE (1392) BLE_MESH: light_server_init, NULL Light Lightness State[0m
[0;31mE (1399) BLE_MESH: light_server_init, NULL Light Lightness State[0m
[0;31mE (1406) BLE_MESH: light_server_init, NULL Light HSL State[0m
[0;31mE (1412) BLE_MESH: light_server_init, NULL Light HSL State[0m
[0;33mW (1418) BLE_MESH: bt_mesh_light_ctl_srv_init, Light CTL Setup Server is not present[0m
[0;31mE (1426) BLE_MESH: light_server_init, NULL Light CTL State[0m
[0;31mE (1433) BLE_MESH: light_server_init, NULL Light HSL State[0m
[0;31mE (1439) BLE_MESH: light_server_init, NULL Light HSL State[0m
[0;31mE (1445) BLE_MESH: light_server_init, NULL Light CTL State[0m

整个log如下：

 306- 309 sec       0.00 Mbits/sec
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:7124
load:0x40078000,len:14824
load:0x40080400,len:4468
entry 0x400806ec
[0;32mI (72) boot: Chip Revision: 1[0m
[0;32mI (72) boot_comm: chip revision: 1, min. bootloader chip revision: 0[0m
[0;32mI (39) boot: ESP-IDF v4.1-dev-1086-g93a8603c5-dirty 2nd stage bootloader[0m
[0;32mI (39) boot: compile time 11:11:11[0m
[0;32mI (40) boot: Enabling RNG early entropy source...[0m
[0;32mI (46) boot: SPI Speed      : 40MHz[0m
[0;32mI (50) boot: SPI Mode       : DIO[0m
[0;32mI (54) boot: SPI Flash Size : 4MB[0m
[0;32mI (58) boot: Partition Table:[0m
[0;32mI (61) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (69) boot:  0 nvs              WiFi data        01 02 00009000 00004000[0m
[0;32mI (76) boot:  1 otadata          OTA data         01 00 0000d000 00002000[0m
[0;32mI (84) boot:  2 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (91) boot:  3 factory          factory app      00 00 00010000 00200000[0m
[0;32mI (99) boot: End of partition table[0m
[0;32mI (103) boot: Defaulting to factory image[0m
[0;32mI (107) boot_comm: chip revision: 1, min. application chip revision: 0[0m
[0;32mI (115) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x42720 (272160) map[0m
[0;32mI (221) esp_image: segment 1: paddr=0x00052748 vaddr=0x3ffbdb60 size=0x05384 ( 21380) load[0m
[0;32mI (230) esp_image: segment 2: paddr=0x00057ad4 vaddr=0x40080000 size=0x00400 (  1024) load[0m
[0;32mI (230) esp_image: segment 3: paddr=0x00057edc vaddr=0x40080400 size=0x0813c ( 33084) load[0m
[0;32mI (251) esp_image: segment 4: paddr=0x00060020 vaddr=0x400d0020 size=0xfc248 (1032776) map[0m
[0;32mI (620) esp_image: segment 5: paddr=0x0015c270 vaddr=0x4008853c size=0x179e4 ( 96740) load[0m
[0;32mI (680) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (680) boot: Disabling RNG early entropy source...[0m
[0;32mI (681) cpu_start: Pro cpu up.[0m
[0;32mI (685) cpu_start: Application information:[0m
[0;32mI (689) cpu_start: Project name:     ble_mesh_node_deep_sleep_timer[0m
[0;32mI (696) cpu_start: App version:      187722d-dirty[0m
[0;32mI (702) cpu_start: Compile time:     Jan  2 2020 11:11:09[0m
[0;32mI (708) cpu_start: ELF file SHA256:  07042f3da1a02128...[0m
[0;32mI (714) cpu_start: ESP-IDF:          v4.1-dev-1086-g93a8603c5-dirty[0m
[0;32mI (721) cpu_start: Starting app cpu, entry point is 0x400815d4[0m
[0;32mI (0) cpu_start: App cpu up.[0m
[0;32mI (731) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (738) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM[0m
[0;32mI (744) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM[0m
[0;32mI (750) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM[0m
[0;32mI (756) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM[0m
[0;32mI (762) heap_init: At 3FFD61F8 len 00009E08 (39 KiB): DRAM[0m
[0;32mI (769) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM[0m
[0;32mI (775) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (781) heap_init: At 4009FF20 len 000000E0 (0 KiB): IRAM[0m
[0;32mI (788) cpu_start: Pro cpu start user code[0m
[0;32mI (806) spi_flash: detected chip: generic[0m
[0;32mI (806) spi_flash: flash io: dio[0m
[0;32mI (807) cpu_start: Starting scheduler on PRO CPU.[0m
[0;32mI (0) cpu_start: Starting scheduler on APP CPU.[0m
[0;32mI (820) BLE_MESH_WIFI_COEXIST_DEMO: Initializing...[0m
[0;32mI (826) gpio: GPIO[16]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 [0m
ledc_set_fade_with_time --> esp_err 0
ledc_fade_start -->esp_err 0
ledc_set_fade_with_time --> esp_err 0
ledc_fade_start -->esp_err 0
ledc_set_fade_with_time --> esp_err 0
ledc_fade_start -->esp_err 0
ledc_set_fade_with_time --> esp_err 0
ledc_fade_start -->esp_err 0
[0;32mI (858) BOARD: realese press Sta: 16[0m
[0;32mI (863) BOARD: sub ,Long press Gpio: 17,Duty:0
[0m
[0;32mI (868) BOARD: sub ,Long press Gpio: 18,Duty:0
[0m
[0;32mI (873) BOARD: realese press Sta: 5[0m
[0;32mI (877) BOARD: sub ,Long press Gpio: 19,Duty:0
[0m
[0;32mI (883) gpio: GPIO[17]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 [0m
[0;32mI (892) BOARD: realese press Sta: 17[0m
[0;32mI (896) BOARD: sub ,Long press Gpio: 18,Duty:0
[0m
[0;32mI (901) BOARD: sub ,Long press Gpio: 19,Duty:0
[0m
[0;32mI (906) gpio: GPIO[18]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 [0m
[0;32mI (916) BOARD: realese press Sta: 18[0m
[0;32mI (920) BOARD: sub ,Long press Gpio: 19,Duty:0
[0m
[0;32mI (925) gpio: GPIO[5]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 [0m
[0;32mI (934) gpio: GPIO[19]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 [0m
[0;32mI (935) BOARD: realese press Sta: 19[0m
[0;32mI (974) BLE_MESH_WIFI_COEXIST_DEMO: Quantitative_Change, int Xs:5000, int Xe:5400,int Ys:3066,int Ye:2237,int in:5000,int out:3066,float fin:-2.072500 [0m
[0;32mI (978) BLE_MESH_WIFI_COEXIST_DEMO: light_ctl_pwm_get, color_ctl_pwm->s2000K:0, s2700K:3066, s7000K:8000, ALL_CURRENT:0 [0m
ledc_set_fade_with_time --> esp_err 0
ledc_fade_start -->esp_err 0
ledc_set_fade_with_time --> esp_err 0
ledc_fade_start -->esp_err 0
ledc_set_fade_with_time --> esp_err 0
ledc_fade_start -->esp_err 0
ledc_set_fade_with_time --> esp_err 0
ledc_fade_start -->esp_err 0
[0;32mI (1014) BTDM_INIT: BT controller compile version [340d4d4][0m
[0;32mI (1021) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
[0;32mI (1118) phy: phy_version: 4180, cb3948e, Sep 12 2019, 16:39:13, 0, 0[0m
[0;31mE (1389) BLE_MESH: generic_server_init, NULL Generic OnPowerUp State[0m
[0;31mE (1389) BLE_MESH: generic_server_init, NULL Generic OnPowerUp State[0m
[0;31mE (1392) BLE_MESH: light_server_init, NULL Light Lightness State[0m
[0;31mE (1399) BLE_MESH: light_server_init, NULL Light Lightness State[0m
[0;31mE (1406) BLE_MESH: light_server_init, NULL Light HSL State[0m
[0;31mE (1412) BLE_MESH: light_server_init, NULL Light HSL State[0m
[0;33mW (1418) BLE_MESH: bt_mesh_light_ctl_srv_init, Light CTL Setup Server is not present[0m
[0;31mE (1426) BLE_MESH: light_server_init, NULL Light CTL State[0m
[0;31mE (1433) BLE_MESH: light_server_init, NULL Light HSL State[0m
[0;31mE (1439) BLE_MESH: light_server_init, NULL Light HSL State[0m
[0;31mE (1445) BLE_MESH: light_server_init, NULL Light CTL State[0m
[0;32mI (3334) BLE_MESH_WIFI_COEXIST_DEMO: ESP_BLE_MESH_PROV_REGISTER_COMP_EVT, err_code: 0[0m
[0;32mI (3336) BLE_MESH_WIFI_COEXIST_DEMO: BLE Mesh Wi-Fi Coexist Node initialized[0m
[0;32mI (3345) BLE_MESH_WIFI_COEXIST_DEMO: ESP_BLE_MESH_NODE_PROV_ENABLE_COMP_EVT, err_code: 0[0m
[0;32mI (3357) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
[0;32mI (3358) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m

 ==================================================
 |       Steps to test WiFi throughput            |
 |                                                |
 |  1. Print 'help' to gain overview of commands  |
 |  2. Configure device to station or soft-AP     |
 |  3. Setup WiFi connection                      |
 |  4. Run iperf to test UDP/TCP RX/TX throughput |
 |                                                |
 =================================================

[5n

WCCWCC, 2020-01-02:
Hi, @wivin001 

     你需要一个什么样的 demo，需求能够描述一下吗？
     1. 你需要哪些 server and client models ?
     2. wifi 的具体行为是什么？ 比如想验证共存时 wifi 的 tx ， rx速率 ?
     我会尽快给你提供一个参考 demo.

wivin001, 2020-01-02:
我需要HSL或者 CTL，而且能否结合下Mesh_Model_Specification v1.0.pdf 文档中，并注释下如何根据这个协议中创建对应model框架，可以拿HSL或者CTL来写注释，谢谢。
WIFI的验证共存时候，需要实现WIFI在TX RX时，blemesh能够正常使用，无论是中继还是其他其他的功能应该尽可能的不受影响。比如开启Run iperf to test UDP/TCP RX/TX throughput时候，能够正常的使用blemesh的功能，比如正常BLE配网，或者FAST配网，并在上述的CTL或者HSL的MODEL中能够正常控制，而不仅仅控制GENEL模型。

Campou, 2020-08-20:
Hi @wivin001 

Sorry for the late reply.

From the above log, in order to implement the SIG-defined server models in the application, please make sure the corresponding states are defined for each server model. Currently these states are missing for the server models, so during mesh initialization, the error log will be output like the above.

Thanks.

closed 2025-03-14 by Alvin1Zhang