espressif/esp-idf #4572: Ble Secure Connection - dhkey chcks do no match

BonifaceBassey, 2020-01-02:
## Environment

- Development Kit:     ESP32-DevKitC
- Kit version (DevKitC): v2
- Module or chip used:  ESP32-WROOM-32
- IDF version: v4.0-beta2-174-g99fb9a3f7
- Build System:         [Make|CMake]
- Compiler version:     xtensa-esp32-elf-gcc.exe (crosstool-NG esp-2019r2) 8.2.0
- Operating System:     Windows
- Power Supply:         USB

## Problem Description

When creating a secure connection with "**ESP_LE_AUTH_REQ_SC_MITM_BOND**", I am consistently getting 
> BT_SMP: dhkey chcks do no match

I am using nrf Connect. My code is based on [ESP-IDF Gatt Security Server Demo](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/ble/gatt_security_server). The strange this is that, the connection will be successful after 3 or 4 or several trials. 

What  could the problem be?

## Debug Logs

```
I (211070) BTLowEnergy: ESP_GATTS_CONNECT_EVT, conn_id = 0
I (211071) BTLowEnergy: 7f 59 74 e9 2c 6d
E (211402) BT_SMP: Value for numeric comparison = 441067
I (211407) BTLowEnergy: NC_REQ_EVT Please compare the numeric value: 441067 callback: 1
I (211490) BTLowEnergy: Update Connection Params : status = 0, min_int = 6, max_int = 12,conn_int = 12,latency = 0, timeout = 400
W (215468) BT_SMP: dhkey chcks do no match

E (215487) BTLowEnergy: Pairing failed
I (215540) BTLowEnergy: authentication complete, status: 0
W (216471) BT_BTM: btm_sec_clr_temp_auth_service() - no dev CB

W (216472) BT_APPL: bta_gattc_conn_cback() - cif=3 connected=0 conn_id=3 reason=0x0013
E (216477) BT_BTM: Device not found

E (216481) BT_APPL: bta_gattc_mark_bg_conn unable to find the bg connection mask for: 7f:59:74:e9:2c:6d
I (216490) BTLowEnergy: ESP_GATTS_DISCONNECT_EVT, reason = 0x13
I (216516) BTLowEnergy: Start advertising success

```

GYC-Espressif, 2020-01-03:
Hello, could you help use `git pull` to update to the last IDF version? If the still have this issue, please tell me the phone you used. Thank you.

BonifaceBassey, 2020-01-03:
Hi, I have used git pull to update to the latest IDF version (v4.0-beta2-320-gfb81b516d) but the problem still persists.
I am using windows 10 and Xiaomi Note7 same behaviour.

GYC-Espressif, 2020-01-06:
Sorry for the late reply. I forget to remind you to use `git submodule update --init` command to update the submodule of IDF. Have you used this command? And from my side, I will find a phone you used to see if this problem still exists.

BonifaceBassey, 2020-01-07:
I have done git submodule update --init, still the same behaviour.
Mostly, i am getting **_W (215468) BT_SMP: dhkey chcks do no match_** 2 times and on the 3rd trial, the connection succeeds.

BonifaceBassey, 2020-02-25:
Any update on this? I am still having the same issue

GYC-Espressif, 2020-03-03:
Hello. We don't reproduce this issue yet.Could you help change the function from "/esp-idf/components/bt/host/bluedroid/stack/btm/btm_sec.c"
```
void btm_sec_clr_temp_auth_service (BD_ADDR bda)
……
```
To
```
void btm_sec_clr_temp_auth_service (BD_ADDR bda)
{
    tBTM_SEC_DEV_REC   *p_dev_rec;

    if ((p_dev_rec = btm_find_dev (bda)) == NULL) {
        BTM_TRACE_WARNING ("btm_sec_clr_temp_auth_service() - no dev CB\n");
        ets_printf("bda: [%02x:%02x:%02x:%02x:%02x:%02x]\nlist:\n",bda[0], bda[1], bda[2], bda[3], bda[4], bda[5]);

        tBTM_SEC_DEV_REC *p_dev_rec = &btm_cb.sec_dev_rec[0];

        if (bda) {
            for (uint8_t i = 0; i < BTM_SEC_MAX_DEVICE_RECORDS; i++, p_dev_rec++) {
                if (p_dev_rec->sec_flags & BTM_SEC_IN_USE) {
                    ets_printf("p_dev_rec->bd_addr: [%02x:%02x:%02x:%02x:%02x:%02x] in use\n");

                #if BLE_INCLUDED == TRUE
                    // If a LE random address is looking for device record
                    ets_printf("p_dev_rec->ble.pseudo_addr: [%02x:%02x:%02x:%02x:%02x:%02x] in use\n");
                #endif
                
                } else {
                    ets_printf("not use\n");
                }
            }
        }
        return;
    }

    /* Reset the temporary authorized flag so that next time (untrusted) service is accessed autorization will take place */
    if (p_dev_rec->last_author_service_id != BTM_SEC_NO_LAST_SERVICE_ID && p_dev_rec->p_cur_service) {
        BTM_TRACE_DEBUG ("btm_sec_clr_auth_service_by_psm [clearing device: %02x:%02x:%02x:%02x:%02x:%02x]\n",
                         bda[0], bda[1], bda[2], bda[3], bda[4], bda[5]);

        p_dev_rec->last_author_service_id = BTM_SEC_NO_LAST_SERVICE_ID;
    }
}
```
and send the log to me? Thanks.

fzboffice, 2020-06-16:
I have the same issue

fzboffice, 2020-06-16:
@Alvin1Zhang

fzboffice, 2020-06-17:
@BonifaceBassey Is there a solution ^^

Alvin1Zhang, 2020-06-17:
@fzboffice Thanks for the updates and sorry for slow turnaround. As you might have seen the comment by @GYC-Espressif that we could not reproduce the issue locally, would you please help share more logs with us for further debugging? Thanks.

fzboffice, 2020-06-17:
> @fzboffice Thanks for the updates and sorry for slow turnaround. As you might have seen the comment by @GYC-Espressif that we could not reproduce the issue locally, would you please help share more logs with us for further debugging? Thanks.

@Alvin1Zhang 我的环境是esp-idf v3.3 最开始出现问题的时候我初始化了gatts和gattc，因为我的项目需要同时用到他们，遇到了这个问题之后，我做了一些尝试，只保留了gatts的初始化，这个问题消失了

fzboffice, 2020-06-17:
但是我的项目不得不同时用到gatts和gattc，所以这个问题依旧困扰着我

WCCWCC, 2020-06-18:
Hi， @fzboffice

   为了解决这个问题，我需要知道详细的复现步骤，最好也能够提供复现代码。

fzboffice, 2020-06-20:
> Hi， @fzboffice
> 
> 为了解决这个问题，我需要知道详细的复现步骤，最好也能够提供复现代码。

@WCCWCC 我做了很多的尝试，现在已经彻底解决了问题，之前我的蓝牙地址使用了BLE_ADDR_TYPE_RANDOM，那么会出现上文的错误，但是当我使用了BLE_ADDR_TYPE_PUBLIC时，就再也不会出现这个错误了

fzboffice, 2020-06-20:
@WCCWCC 复现代码的话，我只是把官方例子的gatts和gattc合并 然后都启用，没有什么特殊的

younglee123, 2020-06-29:
Hi, @fzboffice 
I merged gatts and gattc from the official example, but still cannot reproduce this problem. Here are my debug logs and code.

### Debug logs
```c
I (890) SEC_GATTS_DEMO: advertising start success
I (6570) SEC_GATTS_DEMO: ESP_GATTS_CONNECT_EVT
I (6570) SEC_GATTC_DEMO: EVT 40, gattc if 3
W (18580) BT_SMP: FOR LE SC LTK IS USED INSTEAD OF STK
I (18760) SEC_GATTS_DEMO: key type = ESP_LE_KEY_LENC
I (18760) SEC_GATTS_DEMO: key type = ESP_LE_KEY_PENC
I (18760) SEC_GATTS_DEMO: key type = ESP_LE_KEY_LID
I (19000) SEC_GATTS_DEMO: key type = ESP_LE_KEY_PID
I (19000) SEC_GATTS_DEMO: remote BD_ADDR: 402add4dfe57
I (19000) SEC_GATTS_DEMO: address type = 1
I (19000) SEC_GATTS_DEMO: pair status = success
I (19010) SEC_GATTS_DEMO: auth mode = ESP_LE_AUTH_REQ_SC_BOND
I (19010) SEC_GATTS_DEMO: Bonded devices number : 1

I (19020) SEC_GATTS_DEMO: Bonded devices list : 1

I (19020) SEC_GATTS_DEMO: 40 2a dd 4d fe 57 
```
### Gatts gattc demo
[example_ble_sec_gatts_gattc_demo.zip](https://github.com/espressif/esp-idf/files/4845312/example_ble_sec_gatts_gattc_demo.zip)

fzboffice, 2020-06-30:
@younglee123 广播时 地址类型是random

WCCWCC, 2020-06-30:
Hi, @fzboffice 

* 我们复现这个问题可能要使用到 你出现问题时使用代码，idf 的版本信息（commit id），手机的型号， 你可以帮我们提供一下吗？

* 复现问题的代码在 adv 时的地址类型确实是 random 的，如下图所示:
[image]

fzboffice, 2020-06-30:
@WCCWCC 我现在，代码改动颇多，连自己都无法复现了，就我目前的代码来说完全正常，我只能提供commit https://github.com/espressif/esp-idf/tree/6c9c253ed1bf8a0a3e29c9ed7f7bb8c1fcf7f50c

fzboffice, 2020-06-30:
@WCCWCC 手机型号为iphone8 和 pixel2 XL 都遇到过这个问题

Alvin1Zhang, 2020-11-23:
@fzboffice Thanks for reporting. Will close due to the issue can't reproduce at both your side and our side, feel free to reopen if the issue happens again.

closed 2020-11-23 by Alvin1Zhang