espressif/esp-idf #4574: Bluetooth set random address not working anymore

aircable, 2020-01-02:
@Weijian-Espressif Most likely there is easier way to do it or more elegant, but this is my address generator. Maybe its not pretty but i think it works just like i expecting:

```
void BLEAdvertising::setPrivateAddress(esp_ble_addr_type_t type) {
	esp_bd_addr_t addr;
	m_advParams.own_addr_type = type;

	if(type == BLE_ADDR_TYPE_RPA_PUBLIC) {
		esp_ble_gap_config_local_privacy(true);
		return;
	}
	else{
		mbedtls_ctr_drbg_context ctr_drbg;
		mbedtls_entropy_context entropy;
		mbedtls_md_context_t ctx;

		char pers[] = "aes generate key";
		int ret;
		unsigned char key[BLOCK_SIZE] = {0};
		const char inp[] = "random static address";
		unsigned char outp[BLOCK_SIZE/2];

		mbedtls_entropy_init( &entropy );
		mbedtls_ctr_drbg_init( &ctr_drbg );

		if( ( ret = mbedtls_ctr_drbg_seed( &ctr_drbg, mbedtls_entropy_func, &entropy,
			(unsigned char *) pers, strlen( pers ) ) ) != 0 )
		{
			printf( " failed\n ! mbedtls_ctr_drbg_init returned -0x%04x\n", -ret );
			goto exit;
		}

		if( ( ret = mbedtls_ctr_drbg_random( &ctr_drbg, key, BLOCK_SIZE ) ) != 0 )
		{
			printf( " failed\n ! mbedtls_ctr_drbg_random returned -0x%04x\n", -ret );
			goto exit;
		}

		mbedtls_md_init(&ctx);
		mbedtls_md_setup(&ctx, mbedtls_md_info_from_type(HMAC), true);
		mbedtls_md_hmac_starts(&ctx, (const unsigned char *)key, BLOCK_SIZE);
		mbedtls_md_hmac_update(&ctx, (const unsigned char *)inp, strlen(inp));
		mbedtls_md_hmac_finish(&ctx, outp);
		mbedtls_md_free(&ctx);
		ESP_LOG_BUFFER_HEX("random key", key, BLOCK_SIZE);
		ESP_LOG_BUFFER_HEX("HASH step 2", outp, BLOCK_SIZE/2);

		memcpy(addr, outp, 6);
	}
	if(type == BLE_ADDR_TYPE_RANDOM) {
		addr[0] &= 0x3F;  // <--- Format of non-resolvable private address 00xx xxxx
	}
	else{
		addr[0] |= 0xC0;  // <--- Format of static address 11xx xxxx
	}

	esp_ble_gap_set_rand_addr(addr);
	ESP_LOG_BUFFER_HEX("random address", addr, 6);
	exit:
	return;
}
```

_Originally posted by @chegewara in https://github.com/espressif/esp-idf/issues/2559#issuecomment-430057879_

chegewara, 2020-01-02:
Hi @aircable 
could you try to call `esp_ble_gap_config_local_privacy(true);` before `setPrivateAddress` in your code?

aircable, 2020-01-03:
I already did, I tried both ways , true and false.
I think false should be the correct one for random, but it crashes as well.
Setting to false immediate reboots without comment, setting to true crashes
with Mediator error.

For RANDOM it still says:

BT_BTC: Invalid random address, the high bit should be 0b11, bits of the
random part shall not be all 1 or 0
and crashes. But, should it not be set to 0b00 for RANDOM?
Anyway, it both crashes the way I described it.

I use esp_ble_gap_config_local_privacy(true);  for the public address. That
works fine for me.

My code is actually based on esp-idf and I made sure I have the latest
version.
 esp_ble_gap_config_local_privacy(true);
return esp_ble_gap_set_rand_addr(addr);

GYC-Espressif, 2020-01-03:
static device address, The two most significant bits of the address shall be equal to 1, 11
resolvable private address, The two most significant bits of prand shall be equal to 0 and 1, 10
Non-resolvable private address, The two most significant bits of the address shall be equal to 0, 00
But consider there is less use of this function, we do not support set Non-resolvable private address in this function. So you will see the error log when set a Non-resolvable private address.
And you can only call one of the two functions in one time: esp_ble_gap_config_local_privacy and setPrivateAddress.Or it will cause the error.

aircable, 2020-01-03:
Thank you for your reply. The bug report is against the IDF component bt/host/bluedroid library, not esp32-arduino, which is a different github repo:  espressif/arduino-esp32. The function setPrivateAddress is from Arduino.

The only function the IDF has to set Bluetooth addresses is
```esp_err_t esp_ble_gap_set_rand_addr(esp_bd_addr_t rand_addr)```.

Unfortunately, this function crashes with various error messages whenever on of the two high bits is set to 0. Meaning any address 0b01xxxxxx, 0b10xxxxxx and 0x00xxxxxx is crashing.

But it seems that this function should not be called at all. Can you confirm that this would be the correct way to set a random address for the BLE advertisement or use the build-in one?

```
esp_err_t bluetooth_setaddr( esp_bd_addr_t addr, esp_ble_addr_type_t type) {
	if(type == BLE_ADDR_TYPE_PUBLIC) {
		esp_efuse_mac_get_default( addr );
		esp_base_mac_addr_set( addr );
		esp_ble_gap_clear_rand_addr();
		return esp_ble_gap_config_local_privacy(false);
	} else {
		esp_ble_gap_config_local_privacy(true);
		esp_base_mac_addr_get( addr ); // get generated address
		ESP_LOG_BUFFER_HEX(TAG, "random address", addr, 6);
	}
}
```

aircable, 2020-01-03:
Only that the function ```esp_ble_gap_config_local_privacy(true);``` does not change the advertisement MAC address. Still set to the build-in one.

chegewara, 2020-01-04:
Its because you are trying to read base MAC which is burned in efuse:
https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/system/system.html#_CPPv421esp_base_mac_addr_setPK7uint8_t

Try this function:
https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/bluetooth/esp_gap_ble.html#_CPPv431esp_ble_gap_get_local_used_addr13esp_bd_addr_tP7uint8_t

aircable, 2020-01-04:
Thanks for all your help.

I solved the problem by setting the adv parameter to RANDOM.
and then call ```esp_ble_gap_config_local_privacy(true);``` to generate a
new random address.
This works nicely. No need to set the address manually.

Alvin1Zhang, 2020-08-20:
Thanks for reporting and updates, feel free to reopen if the issue still happens or please file a new ticket for new issue. Thanks.

closed 2020-08-20 by Alvin1Zhang

aircable, 2023-04-11:
I'd like to reopen the issue against IDF-5.1-dev since it is not working there.