espressif/esp-idf #4576: multiple parameter store into nvs creating issue.

dipaknagvdia, 2020-01-03:
Hi I am new to IDF I am trying to store multiple parameter in nvs I am unable to do it.
I am using ESP32_Devkitc_V4  -board  

I want to store 	
	int32_t ee_factory_data_check,
	int32_t ee_temp_previous_value,
	int32_t ee_force_publish_count,
	int32_t ee_total_data_send_count,
	int32_t ee_sleep_multiplier_count,
these varible into nvs but not happning. Kindly help.

My code snippet is given below: 

nvs_handle my_handle;
int32_t ee_factory_data_check = 0; // value will default to 0, if not set yet in NVS
int32_t ee_temp_previous_value = 0; // value will default to 0, if not set yet in NVS
int32_t ee_force_publish_count = 0; // value will default to 0, if not set yet in NVS
int32_t ee_total_data_send_count = 0; // value will default to 0, if not set yet in NVS
int32_t ee_sleep_multiplier_count = 0; // value will default to 0, if not set yet in NVS

// Initialize NVS
 esp_err_t err = nvs_flash_init();
if (err == ESP_ERR_NVS_NO_FREE_PAGES || err == ESP_ERR_NVS_NEW_VERSION_FOUND) 
{
        ESP_ERROR_CHECK(nvs_flash_erase());
        err = nvs_flash_init();
}
ESP_ERROR_CHECK( err );

// Open
printf("\n");
printf("Opening Non-Volatile Storage (NVS) handle... ");
err = nvs_open("storage", NVS_READWRITE, &my_handle);
if (err != ESP_OK) 
{
     printf("Error (%s) opening NVS handle!\n", esp_err_to_name(err));
} 
else 
{
        printf("Done\n");
        // Read
	ee_factory_data_check = 0;
	ee_sleep_multiplier_count =0;
	ee_force_publish_count =0;
        err = nvs_get_i32(my_handle, "ee_factory_data_check", &ee_factory_data_check);
        switch (err) 
	{
            case ESP_OK:
                printf("Done\n");
                printf("ee_factory_data_check = %d\n", ee_factory_data_check);
                break;
            case ESP_ERR_NVS_NOT_FOUND:
                printf("The value is not initialized yet!\n");
                break;
            default :
                printf("Error (%s) reading!\n", esp_err_to_name(err));
        }
	if(ee_factory_data_check != 12345)
	{
		printf("Factory data write into NVS");
		ee_factory_data_check = 12345;
		err = nvs_set_i32(my_handle, "ee_factory_data_check", ee_factory_data_check);
		printf((err != ESP_OK) ? "ee_factory_data_check write Failed!\n" : "Done\n");
		ee_sleep_multiplier_count =1;
		err = nvs_set_i32(my_handle, "ee_sleep_multiplier_count", ee_sleep_multiplier_count);
		printf((err != ESP_OK) ? "ee_sleep_multiplier_count write Failed!\n" : "Done\n");
		ee_force_publish_count =0;
		err = nvs_set_i32(my_handle, "ee_force_publish_count", ee_force_publish_count);
		printf((err != ESP_OK) ? "ee_force_publish_count write Failed!\n" : "Done\n");
		printf("Committing updates in NVS ... ");
		err = nvs_commit(my_handle);
		printf((err != ESP_OK) ? "Failed!\n" : "Done\n");
	}	
	else
	{
		printf("Factory data not writ in NVS ... ");
		err = nvs_get_i32(my_handle, "ee_sleep_multiplier_count", &ee_sleep_multiplier_count);
		switch (err) {
                  case ESP_OK:
                         printf("Done\n");
                         printf("ee_sleep_multiplier_count = %d\n", ee_sleep_multiplier_count);
                         break;
                 case ESP_ERR_NVS_NOT_FOUND:
                         printf("The value is not initialized yet!\n");
                         break;
                  default :
                         printf("Error (%s) reading!\n", esp_err_to_name(err));
		}
		err = nvs_get_i32(my_handle, "ee_force_publish_count", &ee_force_publish_count);
		switch (err) {
                    case ESP_OK:
                          printf("Done\n");
                          printf("ee_force_publish_count = %d\n", ee_force_publish_count);
                          break;
                    case ESP_ERR_NVS_NOT_FOUND:
                          printf("The value is not initialized yet!\n");
                          break;
                    default :
                           printf("Error (%s) reading!\n", esp_err_to_name(err));
		}	
	}
}

0xjakob, 2020-01-06:
Hi @dipaknagvdia 
Could you please elaborate on why you think it's not happening? Please keep in mind that we don't have your device here and can't see your flash. Furthermore, it would be nice if you could format your code. There is an issue template which guides you through the steps of creating an issue. The more information you provide here, the more likely someone here will help. Thanks!

Alvin1Zhang, 2020-08-19:
Thanks for reporting, will close due to short of updates, feel free to reopen with more details. Thanks.

closed 2020-08-19 by Alvin1Zhang