espressif/esp-idf #4577: ESP32 Smart config not recieving BSSID

j1and1, 2020-01-03:
## Environment

- Development Kit:      ESP32-DevKit
- Kit version (for WroverKit/PicoKit/DevKitC): v1
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): ESP-IDF v4.1-dev-1572-g30372f5a4
- Build System:         [Make|CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System:    Windows & Linux
- Power Supply:         USB

## Problem Description

I was investigating how to configure the ESP to connect to wifi network. So I found out that there's a example in wifi called [smart config](https://github.com/espressif/esp-idf/blob/master/examples/wifi/smart_config/main/smartconfig_main.c) that I can use to connect to acces point. So I flashed it on to my devboard and tried the Esp touch app... and it didn't work for 2 reasons:

1. the received `smartconfig_event_got_ssid_pswd_t` object has no `BSSID` and `BSSID_set` property set if I dump the memory then theres only SSID and password set, but the ESP touch app always sends BSSID if I understood the code correctly.
2. The WiFi I'm trying to connect to is hidden and as the [getting started](https://github.com/espressif/esp-idf/blob/master/examples/wifi/getting_started/station/main/station_example_main.c) code is also having troubles connecting to my hidden network when I provide SSID/BSSID/BSSID_set and password values in the `wifi_config_t` there's another issue

### Expected Behavior

BSSID is present in `smartconfig_event_got_ssid_pswd_t` object and the ESP succesfuly connects to a hidden network 

### Actual Behavior

`smartconfig_event_got_ssid_pswd_t` is missing BSSID as the app sends it all of the time and it does not connect to WiFi with hidden SSID even if I pass the BSSID manually

### Steps to repropduce

1. just flash the ESP32 devboard with smart config example and hide your WiFi SSID
2. use the esp touch app to send config
3. dump out the memory of `smartconfig_event_got_ssid_pswd_t` and see that theres no BSSID

liuzfesp, 2020-01-20:
@j1and1, could you help to check the channel number of your AP?  I suspect the hidden AP is in channel 12~14.

By default, the esp32 station is configured to WIFI_COUNTRY_POLICY_AUTO, so it does active scan from channel 1~11, and passive scan from 12 to 13. If the hidden AP is in 12~14, the station has no way to find it.

In order to find the hidden API in channel 12~14, you can configure the country policy to WIFI_COUNTRY_POLICY_MANUAL.

j1and1, 2020-01-23:
@liuzfesp I have a ASUS RT-AC51U router that has country set to Latvia and the channels are set to Auto and channel bandwidth is set to 20/40 Mhz  and by its logs it is currently running on channel 13... I'll try to set the WIFI_COUNTRY_POLICY_MANUAL tonight and see if it connects and report back to you.

Also if the channel is the issue wouldn't it be wise to add country or something to the Smart config protocol?

j1and1, 2020-01-24:
@liuzfesp When I set country to manual and add the country code to my code the board connects to my hidden WiFi successfully.

Before I close this issue is there a way to pass the country mode or code to the Smart Config protocol?

Alvin1Zhang, 2020-01-29:
Thanks for updates and sorry for the slow turnaround. Sorry that we are now during the holiday of Chinese Lunar New Year. Will update to you when we are back, thanks.

liuzfesp, 2020-02-17:
HI @j1and1, I think what you need to do is always set country mode to manual mode if you want to connect the hidden AP in channel [12, 14].

j1and1, 2020-02-18:
Ok thanks for the info... will do so.

closed 2020-02-18 by j1and1