espressif/esp-idf #4578: Pppos client ipcp req terminate by server

superxxd, 2020-01-04:
Hardware esp32+air720h,config Cmnet atdt*99***1#,LCP pap is ok,but ipcp req,the ppp server send terminate LCP packet

xiangxud, 2020-01-04:
the log:
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:7064
load:0x40078000,len:14204
load:0x40080400,len:4472
0x40080400: _init at ??:?

entry 0x400806f0
I (63) boot: Chip Revision: 1
I (65) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (40) boot: ESP-IDF v4.1-dev-1572-g30372f5a4-dirty 2nd stage bootloader
I (40) boot: compile time 10:33:08
I (41) boot: Enabling RNG early entropy source...
I (47) boot: SPI Speed      : 80MHz
I (51) boot: SPI Mode       : DIO
I (55) boot: SPI Flash Size : 4MB
I (59) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (92) boot: End of partition table
I (96) boot_comm: chip revision: 1, min. application chip revision: 0
I (103) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x174e4 ( 95460) map
I (141) esp_image: segment 1: paddr=0x0002750c vaddr=0x3ffb0000 size=0x0245c (  9308) load
I (145) esp_image: segment 2: paddr=0x00029970 vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at C:/esp-idf/components/freertos/xtensa_vectors.S:1778

I (148) esp_image: segment 3: paddr=0x00029d7c vaddr=0x40080404 size=0x0629c ( 25244) load
I (166) esp_image: segment 4: paddr=0x00030020 vaddr=0x400d0020 size=0x613e4 (398308) map
0x400d0020: _stext at ??:?

I (286) esp_image: segment 5: paddr=0x0009140c vaddr=0x400866a0 size=0x05284 ( 21124) load
0x400866a0: timer_ll_wdt_set_protect at c:\esp-idf\components\soc\esp32\include\hal/timer_ll.h:394
 (inlined by) disableAllWdts at C:/esp-idf/components/esp32/panic.c:432
 (inlined by) panicHandler at C:/esp-idf/components/esp32/panic.c:313

I (301) boot: Loaded app from partition at offset 0x10000
I (302) boot: Disabling RNG early entropy source...
I (302) cpu_start: Pro cpu up.
I (306) cpu_start: Application information:
I (310) cpu_start: Project name:     App_pppos_task
I (316) cpu_start: App version:      v4.1-dev-1572-g30372f5a4-dirty
I (323) cpu_start: Compile time:     Jan  4 2020 10:28:46
I (329) cpu_start: ELF file SHA256:  5e481e452c39dead...
I (335) cpu_start: ESP-IDF:          v4.1-dev-1572-g30372f5a4-dirty
I (342) cpu_start: Starting app cpu, entry point is 0x4008133c
0x4008133c: call_start_cpu1 at C:/esp-idf/components/esp32/cpu_start.c:276

I (0) cpu_start: App cpu up.
I (352) heap_init: Initializing. RAM available for dynamic allocation:
I (359) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (365) heap_init: At 3FFB4758 len 0002B8A8 (174 KiB): DRAM
I (372) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (378) heap_init: At 3FFE4350 len 00013CB0 (79 KiB): D/IRAM
I (384) heap_init: At 4008B924 len 000146DC (81 KiB): IRAM
I (390) cpu_start: Pro cpu start user code
I (407) esp_apptrace: Initialized TRAX on CPU0
I (408) spi_flash: detected chip: generic
I (408) spi_flash: flash io: dio
I (411) cpu_start: Starting scheduler on PRO CPU.
I (39) esp_apptrace: Initialized TRAX on CPU1
I (0) cpu_start: Starting scheduler on APP CPU.
I (438) esp-modem: MOMEM UART init,port:1 tx:25 rx:26 bdrate 921600

I (438) uart: queue free spaces: 30
[... 84 lines cut ...]
LCP terminated by peer
ppp phase changed[1]: phase=6
pppos_send_config[1]: out_accm=FF FF FF FF
ppp_send_config[1]
pppos_recv_config[1]: in_accm=0 0 0 0
ppp_recv_config[1]
sent [LCP TermAck id=0x2]
I (4028) esp-modem: wrt 18:
I (4038) esp-modem: 7e ff 7d 23 c0 21 7d 26 7d 22 7d 20 7d 24 94 7d
I (4038) esp-modem: 2d 7e
pppos_write[1]: len=8
I (4138) esp-modem: rec 23:
I (4138) esp-modem: 0d 0a 2b 43 47 45 56 3a 20 4d 45 20 44 45 41 43
I (4138) esp-modem: 54 20 31 2c 31 0d 0a
pppos_input[1]: got 23 bytes
I (4188) esp-modem: rec 14:
I (4188) esp-modem: 0d 0a 4e 4f 20 43 41 52 52 49 45 52 0d 0a
pppos_input[1]: got 14 bytes
ppp phase changed[1]: phase=12
Connection terminated.
ppp_link_terminated[1]
ppp_link_end[1]
ppp phase changed[1]: phase=0
ppp_link_terminated[1]: finished.
I (10038) pppos_example: Modem Disconnect from PPP Server

xiangxistu, 2020-01-04:
Sometimes ppp will terminate by sever, I don't know why the server switch off my ppp link state.
If you get the reason ,please share with me. Thanks.

superxxd, 2020-01-04:
Is there no upward reason for LCP negotiation code identifier? Is LwIP not good at handling this

superxxd, 2020-01-18:
The same program, I changed the 4G module, pppos is OK, anyone know why

Mluis31, 2020-07-27:
Any news on this? I'm having the same problem connecting a SIM800L

lhespress, 2020-12-10:
@Mluis31 @superxxd @xiangxistu Could you update the IDF to tag v4.2 and check again? It's better to modify on PPPOS client example or share your code for check if have the same problem on v4.2.

Alvin1Zhang, 2021-01-05:
@superxxd @xiangxud @xiangxistu @Mluis31 Sorry for the very slow turnaround, would you please help share if any updates of the issue? Thanks.

Mluis31, 2021-06-21:
Helle @Alvin1Zhang still not working for me.
`void app_main(void)
{
    ESP_LOGI(TAG, "[APP] Startup..");
    ESP_LOGI(TAG, "[APP] Free memory: %d bytes", esp_get_free_heap_size());
    ESP_LOGI(TAG, "[APP] IDF version: %s", esp_get_idf_version());

    esp_log_level_set("*", ESP_LOG_INFO);
    esp_log_level_set("MQTT_CLIENT", ESP_LOG_VERBOSE);
    esp_log_level_set("TRANSPORT_BASE", ESP_LOG_VERBOSE);
    esp_log_level_set("TRANSPORT", ESP_LOG_VERBOSE);
    esp_log_level_set("OUTBOX", ESP_LOG_VERBOSE);

#if CONFIG_LWIP_PPP_PAP_SUPPORT
    esp_netif_auth_type_t auth_type = NETIF_PPP_AUTHTYPE_PAP;
#elif CONFIG_LWIP_PPP_CHAP_SUPPORT
    esp_netif_auth_type_t auth_type = NETIF_PPP_AUTHTYPE_CHAP;
#elif !defined(CONFIG_EXAMPLE_MODEM_PPP_AUTH_NONE)
#error "Unsupported AUTH Negotiation"
#endif

    ESP_ERROR_CHECK(nvs_flash_init());
    ESP_ERROR_CHECK(esp_netif_init());
    ESP_ERROR_CHECK(esp_event_loop_create_default());
    ESP_ERROR_CHECK(esp_event_handler_register(IP_EVENT, ESP_EVENT_ANY_ID, &on_ip_event, NULL));
    ESP_ERROR_CHECK(esp_event_handler_register(NETIF_PPP_STATUS, ESP_EVENT_ANY_ID, &on_ppp_changed, NULL));

    event_group = xEventGroupCreate();

    /* create dte object */
    esp_modem_dte_config_t config = ESP_MODEM_DTE_DEFAULT_CONFIG();
    /* setup UART specific configuration based on kconfig options */
    config.tx_io_num = CONFIG_EXAMPLE_MODEM_UART_TX_PIN;
    config.rx_io_num = CONFIG_EXAMPLE_MODEM_UART_RX_PIN;
    config.rts_io_num = CONFIG_EXAMPLE_MODEM_UART_RTS_PIN;
    config.cts_io_num = CONFIG_EXAMPLE_MODEM_UART_CTS_PIN;
    config.rx_buffer_size = CONFIG_EXAMPLE_MODEM_UART_RX_BUFFER_SIZE;
    config.tx_buffer_size = CONFIG_EXAMPLE_MODEM_UART_TX_BUFFER_SIZE;
    config.pattern_queue_size = CONFIG_EXAMPLE_MODEM_UART_PATTERN_QUEUE_SIZE;
    config.event_queue_size = CONFIG_EXAMPLE_MODEM_UART_EVENT_QUEUE_SIZE;
    config.event_task_stack_size = CONFIG_EXAMPLE_MODEM_UART_EVENT_TASK_STACK_SIZE;
    config.event_task_priority = CONFIG_EXAMPLE_MODEM_UART_EVENT_TASK_PRIORITY;
    config.line_buffer_size = CONFIG_EXAMPLE_MODEM_UART_RX_BUFFER_SIZE / 2;

    modem_dte_t *dte = esp_modem_dte_init(&config);
    /* Register event handler */
    ESP_ERROR_CHECK(esp_modem_set_event_handler(dte, modem_event_handler, ESP_EVENT_ANY_ID, NULL));

    // Init netif object
    esp_netif_config_t cfg = ESP_NETIF_DEFAULT_PPP();
    esp_netif_t *esp_netif = esp_netif_new(&cfg);
    assert(esp_netif);

    void *modem_netif_adapter = esp_modem_netif_setup(dte);
    esp_modem_netif_set_default_handlers(modem_netif_adapter, esp_netif);

    modem_dce_t *dce = NULL;
    /* create dce object */
    dce = sim800_init(dte);
    assert(dce != NULL);
    ESP_ERROR_CHECK(dce->set_flow_ctrl(dce, MODEM_FLOW_CONTROL_NONE));
    ESP_ERROR_CHECK(dce->store_profile(dce));
    /* Print Module ID, Operator, IMEI, IMSI */
    ESP_LOGI(TAG, "Module: %s", dce->name);
    ESP_LOGI(TAG, "Operator: %s", dce->oper);
    ESP_LOGI(TAG, "IMEI: %s", dce->imei);
    ESP_LOGI(TAG, "IMSI: %s", dce->imsi);
    /* Get signal quality */
    uint32_t rssi = 0, ber = 0;
    ESP_ERROR_CHECK(dce->get_signal_quality(dce, &rssi, &ber));
    ESP_LOGI(TAG, "rssi: %d, ber: %d", rssi, ber);
#if !defined(CONFIG_EXAMPLE_MODEM_PPP_AUTH_NONE) && (defined(CONFIG_LWIP_PPP_PAP_SUPPORT) || defined(CONFIG_LWIP_PPP_CHAP_SUPPORT))
    esp_netif_ppp_set_auth(esp_netif, auth_type, CONFIG_EXAMPLE_MODEM_PPP_AUTH_USERNAME, CONFIG_EXAMPLE_MODEM_PPP_AUTH_PASSWORD);
#endif
    /* attach the modem to the network interface */
    esp_netif_attach(esp_netif, modem_netif_adapter);
    /* Wait for IP address */
    xEventGroupWaitBits(event_group, CONNECT_BIT, pdTRUE, pdTRUE, portMAX_DELAY);

    mqtt_app_start();
    xEventGroupWaitBits(event_group, GOT_DATA_BIT, pdTRUE, pdTRUE, portMAX_DELAY);
    while (1)
    {
        esp_mqtt_client_publish(client, "/topic/qos0", "data", 0, 0, 0);
        vTaskDelay(60000 / portTICK_RATE_MS);
    }
}`
That's a a version of my main app. Based on your example.
Still getting this after a few seconds:
  rcvd [LCP TermReq id=0x3]
  LCP terminated by peer

ESP-YJM, 2021-06-24:
@Mluis31 Could you please help enable PPP debug log(Component config > LWIP > Enable PPP debug log output) to share your whole log output.

Mluis31, 2021-06-24:
@ESP-YJM here it is:

I (27) boot: ESP-IDF v4.4-dev-744-g1cb31e509 2nd stage bootloader
I (27) boot: compile time 14:22:53
I (27) boot: chip revision: 1
I (31) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (38) boot.esp32: SPI Speed      : 40MHz
I (43) boot.esp32: SPI Mode       : DIO
I (47) boot.esp32: SPI Flash Size : 4MB
I (52) boot: Enabling RNG early entropy source...
I (57) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (68) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (83) boot:  2 factory          factory app      00 00 00010000 00100000
I (91) boot: End of partition table
I (95) boot_comm: chip revision: 1, min. application chip revision: 0
I (102) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=1e464h (124004) map
I (157) esp_image: segment 1: paddr=0002e48c vaddr=3ffb0000 size=01b8ch (  7052) load
I (160) esp_image: segment 2: paddr=00030020 vaddr=400d0020 size=6c7a4h (444324) map
I (331) esp_image: segment 3: paddr=0009c7cc vaddr=3ffb1b8c size=00fb0h (  4016) load
I (333) esp_image: segment 4: paddr=0009d784 vaddr=40080000 size=0da08h ( 55816) load
I (360) esp_image: segment 5: paddr=000ab194 vaddr=50000000 size=00010h (    16) load
I (368) boot: Loaded app from partition at offset 0x10000
I (368) boot: Disabling RNG early entropy source...
I (380) cpu_start: Pro cpu up.
I (381) cpu_start: Starting app cpu, entry point is 0x400813a4
0x400813a4: call_start_cpu1 at /home/miguel/esp/esp-idf/components/esp_system/port/cpu_start.c:158

I (0) cpu_start: App cpu up.
I (395) cpu_start: Pro cpu start user code
I (395) cpu_start: cpu freq: 160000000
I (395) cpu_start: Application information:
I (399) cpu_start: Project name:     pppos_client
I (405) cpu_start: App version:      1
I (409) cpu_start: Compile time:     Jun 21 2021 14:40:13
I (415) cpu_start: ELF file SHA256:  170d2447d0930ee8...
I (421) cpu_start: ESP-IDF:          v4.4-dev-744-g1cb31e509
I (428) heap_init: Initializing. RAM available for dynamic allocation:
I (435) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (441) heap_init: At 3FFB4810 len 0002B7F0 (173 KiB): DRAM
I (447) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (453) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (460) heap_init: At 4008DA08 len 000125F8 (73 KiB): IRAM
I (467) spi_flash: detected chip: generic
I (471) spi_flash: flash io: dio
I (476) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (485) pppos_mqtt: [APP] Startup..
I (485) pppos_mqtt: [APP] Free memory: 284332 bytes
I (495) pppos_mqtt: [APP] IDF version: v4.4-dev-744-g1cb31e509
netif: netmask of interface  set to 255.0.0.0
netif: GW address of interface  set to 127.0.0.1
netif_set_ipaddr: netif address being changed
netif: added interface lo IP addr 127.0.0.1 netmask 255.0.0.0 gw 127.0.0.1
I (555) uart: queue free spaces: 30
netif: netmask of interface  set to 255.255.255.255
netif: added interface pp IP addr 0.0.0.0 netmask 255.255.255.255 gw 0.0.0.0
ppp phase changed[1]: phase=0
I (605) pppos_mqtt: Module: SIMCOM_SIM800L
I (605) pppos_mqtt: Operator: "TMN"
I (605) pppos_mqtt: IMEI: 862643037685637
I (605) pppos_mqtt: IMSI: 268069638683271
I (615) pppos_mqtt: rssi: 19, ber: 2
I (745) pppos_mqtt: Modem PPP Started
ppp_connect[1]: holdoff=0
ppp phase changed[1]: phase=3
I (745) pppos_mqtt: PPP state changed event 259
pppos_connect: unit 1: connecting
ppp_start[1]
[... 319 lines cut ...]
pppos_input[1]: got 7 bytes
pppos_input[1]: got 7 bytes
pppos_input[1]: got 6 bytes
pppos_input[1]: got 7 bytes
pppos_input[1]: got 7 bytes
pppos_input[1]: Dropping bad fcs 0x2bd5 proto=0xf7
pppos_input_drop: pbuf len=76, addr 0x3ffc3590
I (13185) MQTT_CLIENT: Publishing skipped: client is not connected
ppp phase changed[1]: phase=12
I (16775) pppos_mqtt: PPP state changed event 268
Connection terminated.
ppp_link_terminated[1]
ppp_link_end[1]
ppp phase changed[1]: phase=0
I (16775) pppos_mqtt: PPP state changed event 256
I (16785) esp-netif_lwip-ppp: Connection lost
I (16785) pppos_mqtt: Modem Disconnect from PPP Server
ppp_link_terminated[1]: finished.
netif: IPv6 address 0 of interface pp set to ::/0x00
netif: IPv6 address 0 of interface pp set to ::/0x00
netif: IPv6 address 1 of interface pp set to ::/0x00
netif: IPv6 address 1 of interface pp set to ::/0x00
netif: IPv6 address 2 of interface pp set to ::/0x00
netif: IPv6 address 2 of interface pp set to ::/0x00
netif: netmask of interface pp set to 0.0.0.0

ESP-YJM, 2021-07-02:
@Mluis31 Sorry for late reply. I originally wanted to use the log to see the cause of the LCP Terminal, but it seems that the peer sends LCP TermReq and does not carry any reason for the disconnection, so currently I cannot know why the link is broken. Could you change another communication module or SIM card and try?

closed 2021-09-13 by Alvin1Zhang