espressif/esp-idf #4580: Where are the esp_event_base_t defined?

DasVoelkel, 2020-01-04:
## Environment

- Development Kit:      [SBC-NodeMCU-ESP32]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version platformio
- Build System:         [Make|CMake]
- Compiler version platformio
- Operating System:     [Linux]
- Power Supply:         [USB]

## Problem Description

//Detailed problem description goes here.

### Expected Behavior
That the WIFI_EVENT define is found

### Actual Behavior
lib/gate_controller/mg_wifi_sta.cpp:154:48: error: 'WIFI_EVENT' was not declared in this scope

### Steps to repropduce

1. load the example
2. try to compile it

### Code to reproduce this issue
https://github.com/espressif/esp-idf/blob/30372f5a4ff2c0dfdaad95f544dc36bcdda30b75/examples/wifi/fast_scan/main/fast_scan.c

On the legacy list:
https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/system/esp_event_legacy.html#_CPPv417system_event_id_t
I can see all the event types, but if I want to use the esp_event.h I need to move away from esp_event_loop_init(...) and for that i need the esp_event_base_t for the wifi or ip events, I can't find them anywhere in the documentation.

DasVoelkel, 2020-01-04:
It even says in the notes:

> This API is part of the legacy event system. New code should use event library API in esp_event.h

but it doesn't really say what equivalent code with esp_event.h would look like.

chegewara, 2020-01-04:
https://github.com/espressif/esp-idf/blob/release/v3.3/components/esp_event/include/esp_event_base.h#L27

https://github.com/espressif/esp-idf/blob/release/v3.3/components/esp_event/include/esp_event.h#L145

DasVoelkel, 2020-01-04:
which says what type it is, not what variables are what values.
On this line:
https://github.com/espressif/esp-idf/blob/30372f5a4ff2c0dfdaad95f544dc36bcdda30b75/examples/wifi/fast_scan/main/fast_scan.c#L94
What is supposed to go into the parameters instead of WIFI_EVENT because that is not defined.
yes i know it is of type

`typedef const char*  esp_event_base_t; /**< unique pointer to a subsystem that exposes events */```

But what ID is the ID of the WIFI_EVENT base ? What actual value is behind it?

igrr, 2020-01-04:
Hi @DasVoelkel,

> IDF version platformio

As you have indicated, you are using ESP-IDF installed via Platformio. If I'm not mistaken, Platformio ships ESP-IDF v3.3.

However you are attempting to compile an example from the master branch, which was added in IDF v4.x. This explains the compilation error: WIFI_EVENT did not exist in IDF v3.3 and was introduced in IDF v4.0.

You have two options:

1) use examples which are provided with IDF version you have installed via Platformio
2) use a more recent IDF version; this may require following the ESP-IDF getting started guide, and not using Platformio.

DasVoelkel, 2020-01-04:
Ok, thanks!
I'll keep this open until i work up a way to switch from pio to the normal idf

closed 2020-02-09 by DasVoelkel