espressif/esp-idf #4581: uart_set_line_inverse() mask parameter not specified

JB-DX, 2020-01-04:
I need to invert TXD line of the UART.
This used to be achieved by  
`uart_set_line_inverse(n, UART_INVERSE_TXD);
`

But UART_INVERSE_TXD is not defined anymore in the new IDF. The doucmentation is also unclear:
```
Choose the wires that need to be inverted. Using the ORred mask of uart_signal_inv_t
```
uart_signal_inv_t is also undefined.

What value is required when TXD is to be inverted?

Why has UART_INVERSE_TXD been removed?

koobest, 2020-01-05:
Hi, @JB-DX 
`uart_signal_inv_t` is define in component/soc/include/hal/uart_types.h, and if you want to inverse TXD line, `UART_SIGNAL_TXD_INV` should  be passed to `uart_set_line_inverse`

thanks !!

JB-DX, 2020-01-05:
Thanks !
It now inverts the signal, but 

> uart_set_line_inverse(n, 0)

 doesn't put it back to normal (non-inverted)
I use that to generate a break signal.
```
uart_set_line_inverse(n, UART_SIGNAL_TXD_INV );
vTaskDelay(200 / portTICK_PERIOD_MS);
uart_set_line_inverse(n, 0);
```
earlier idf versions work fine.

koobest, 2020-01-06:
Hi,
The driver refactoring brings a breaking change to this API. we will fix this soon.
thanks !!

closed 2020-02-26 by espressif-bot