espressif/esp-idf #4582: SIM800L and OTA throwing error

Arthedian, 2020-01-04:
## Environment

- Development Kit: none
- Module or chip used: ESP32-WROOM-32D
- IDF version (run ``git describe --tags`` to find it): v4.0-beta2-296-g7d241e113
- Build System: CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System:    Fedora
- Power Supply: external 5V

## Problem Description

I combined OTA and SIM800L examples together. During downloading new version of FW, the error is ocurring and in the and it fails the download of the FW. I tried to tweak some UART settings but I wasnt succesfull.

### Actual Behavior

```
D (499961) HTTP_CLIENT: is_data_remain=1, is_chunked=0, content_length=979440
D (499971) HTTP_CLIENT: need_read=512, byte_to_read=512, rlen=512, ridx=512
D (499971) HTTP_CLIENT: http_on_body 512
D (499991) OTA: Written image length 243712
D (499991) HTTP_CLIENT: is_data_remain=1, is_chunked=0, content_length=979440
D (499991) HTTP_CLIENT: need_read=1024, byte_to_read=512, rlen=512, ridx=0
D (500001) HTTP_CLIENT: http_on_body 512
D (500011) HTTP_CLIENT: is_data_remain=1, is_chunked=0, content_length=979440
D (500011) HTTP_CLIENT: need_read=512, byte_to_read=512, rlen=512, ridx=512
D (500021) HTTP_CLIENT: http_on_body 512
W (500041) esp-modem: HW FIFO Overflow

```

### Steps to reproduce

1. combine OTA and SIM800L examples and run

Arthedian, 2020-01-09:
Ok I was testing more this problem. This is not the main problem. THe main problem is that each 16kb will stop downloding OTA file for 30 seconds. Each time it will split 512 part into 254 and 258 bytes (not sure why). And almost at the end (the OTA file has almost 1 MB) it has broken the OTA and stops downloading. Only temporary fix I have is changing timeout_ms in esp_http_client_config_t to '100000'. Why this is happening on GSM? If i switch to WIFI, than everything is ok. 
```
 (98191) HTTP_CLIENT: is_data_remain=1, is_chunked=0, content_length=979440
D (98191) HTTP_CLIENT: need_read=1024, byte_to_read=512, rlen=512, ridx=0
D (98191) HTTP_CLIENT: http_on_body 512
D (98191) HTTP_CLIENT: is_data_remain=1, is_chunked=0, content_length=979440
D (98201) HTTP_CLIENT: need_read=512, byte_to_read=512, rlen=254, ridx=512
D (98211) HTTP_CLIENT: http_on_body 254
D (98211) HTTP_CLIENT: is_data_remain=1, is_chunked=0, content_length=979440
D (135571) HTTP_CLIENT: need_read=258, byte_to_read=258, rlen=258, ridx=766
D (135571) HTTP_CLIENT: http_on_body 258

```

montirob, 2020-06-19:
Hi, @Arthedian , have you resolved the HW_FIFO_OVERFLOW issue?
I'm experiencing the same tryng to run the OTA with an LTE module.

lhespress, 2021-01-06:
@Arthedian Could you enable `RTS/CTS` of UART and check again?

TahirShaik, 2021-01-27:
I (323714) aws_mqtt: Processing the Command from AWS_MQTT
I (323724) aws_mqtt: OTA Update
I (330974) esp_https_ota: Starting OTA...
I (330974) esp_https_ota: Writing to partition subtype 16 at offset 0x310000
I (488834) esp_https_ota: Connection closed
I (488834) esp_image: segment 0: paddr=0x00310020 vaddr=0x3f400020 size=0x0554c ( 21836) map
I (488854) esp_image: segment 1: paddr=0x00315574 vaddr=0x00000000 size=0x00000 (     0) 
I (488854) esp_image: segment 2: paddr=0x0031557c vaddr=0x00000000 size=0x00000 (     0) 
I (488854) esp_image: segment 3: paddr=0x00315584 vaddr=0x00000000 size=0x00000 (     0) 
I (488864) esp_image: segment 4: paddr=0x0031558c vaddr=0x00000000 size=0x00000 (     0) 
I (488874) esp_image: segment 5: paddr=0x00315594 vaddr=0x00000000 size=0x00000 (     0) 
E (488884) esp_image: Checksum failed. Calculated 0x81 read 0x0
E (488894) aws_mqtt: Firmware Upgrade Failed return code: 1503

lorenzomerici, 2021-02-02:
I have the same issue, also with SIM800L. Do you guys have found a way to solve?  At first look seemed a simple problem of buffers but maybe it's not...

TahirShaik, 2021-02-03:
**Issue was resolved!!!!**

I used the following functions instead of using **

> esp_https_ota(&config)

**;..............................
**

> esp_https_ota_begin(&ota_config, &https_ota_handle);
> esp_https_ota_get_img_desc(https_ota_handle, &app_desc);
> validate_image_header(&app_desc);
> esp_https_ota_perform(https_ota_handle);
> esp_https_ota_is_complete_data_received(https_ota_handle)
> esp_https_ota_finish(https_ota_handle);

**

When the ota image size is small then  **

> esp_https_ota(&config)

** is working. If the image is larger then it is not working, So my suggestion is to go with the above functions to perform OTA...

closed 2021-04-15 by espressif-bot

ssravindhar, 2021-09-17:
> **Issue was resolved!!!!**
> 
> I used the following functions instead of using the **esp_https_ota(&config)** .....
> 
> > esp_https_ota_begin(&ota_config, &https_ota_handle);
> > esp_https_ota_get_img_desc(https_ota_handle, &app_desc);
> > validate_image_header(&app_desc);
> > esp_https_ota_perform(https_ota_handle);
> > esp_https_ota_is_complete_data_received(https_ota_handle)
> > esp_https_ota_finish(https_ota_handle);
> 
> When the ota image size is small then **esp_https_ota(&config)** is working. If the image is larger then it is not working, So my suggestion is to go with the above functions to perform OTA...

@TahirShaik Hi can u able to share app code for this ota via GSM 4G module.

vil-nakul, 2022-09-12:
Hello I'm also facing the same issue, I have increased the timeout to 10s but the OTA still fails

FrannyLu, 2022-09-29:
I fixed the "HW FIFO Overflow" by moving the UART ISR into RAM (setting CONFIG_UART_ISR_IN_IRAM). Afterwards I still got the following error: "Image validation failed, image is corrupted". I "fixed" this by setting CONFIG_SPI_FLASH_USE_LEGACY_IMPL. No idea why this is not working without the legacy flag set.