espressif/esp-idf #4583: rmt_set_tx_thr_intr_en() causes panic on newest master

StefanRvO, 2020-01-04:
##  Environment

- Development Kit: 

Own PCB, but uses TTGO-Micro-32 and ESP32-S. Issue is present on both modules

- IDF version:

Newest master(30372f5a4ff2c0dfdaad95f544dc36bcdda30b75)
Problem occurs on versions later than 8fd8695ea14b5c2d3a506cd0a8e18d83b6e59e71

- Build System:
Make
- Compiler version:
Problem occurs both using GCC 5.2.0 and  GCC 8.2.0
    
- Operating System: 
Linux
- Power Supply:
external 3.3V

## Problem Description

I'am using the RMT peripheral to control some WS2812 LEDs. I'm using the periphial in interrupt mode.

My code works fine on earlier versions of ESP-IDF, but if I change to versions after 8fd8695ea14b5c2d3a506cd0a8e18d83b6e59e71i get a LoadProhibited panic when calling rmt_set_tx_thr_intr_en()
 

### Expected Behavior
No Panic

### Actual Behavior
LoadProhibited panic

### Code to reproduce this issue
My full code is available here:
https://gitlab.com/StefanRvO/light-equalizer-platform/blob/965245f96f65d03a3f0a658b9200d05943d1da06/software/firmware/components/esp32_digital_led_lib/esp32_digital_led_lib.cpp#L269

The problem occurs on the linked line (269). I can attempt to provide a more minimal example example if needed, but it would probably take some time to write :)

## Debug Logs

```
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x4017b695  PS      : 0x00060833  A0      : 0x80107abe  A1      : 0x3ffdba30  
0x4017b695: rmt_set_tx_thr_intr_en at /home/stefan/esp/esp-idf/components/driver/rmt.c:389

A2      : 0x00000000  A3      : 0x00000001  A4      : 0x00000020  A5      : 0x00000008  
A6      : 0x7ff00000  A7      : 0x7ff00000  A8      : 0x00000000  A9      : 0x00000034  
A10     : 0xfffffe00  A11     : 0x00000000  A12     : 0x000a0f20  A13     : 0x3ffb1a00  
A14     : 0x00000054  A15     : 0x00020f20  SAR     : 0x0000001c  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400029ac  LEND    : 0x400029cb  LCOUNT  : 0x00000000  

ELF file SHA256: 4700b01b8c2f7899f448599425ee822553c88d406eff869aa2bea189e175a4ee

Backtrace: 0x4017b692:0x3ffdba30 0x40107abb:0x3ffdba60 0x400d84c2:0x3ffdbac0 0x400d48a9:0x3ffdbaf0 0x400d48c0:0x3ffdbb10
0x4017b692: rmt_ll_set_tx_limit at /home/stefan/esp/esp-idf/components/soc/esp32/include/hal/rmt_ll.h:177
 (inlined by) rmt_set_tx_thr_intr_en at /home/stefan/esp/esp-idf/components/driver/rmt.c:389                                          

0x40107abb: digitalLeds_addStrands at /home/stefan/Equalizer_Roskilde/software/firmware/components/esp32_digital_led_lib/esp32_digital_led_lib.cpp:269                                                                                                                      

0x400d84c2: light_control_module_init at /home/stefan/Equalizer_Roskilde/software/firmware/main/modules/light_control_module/light_control_module.c:119                                                                                                                     

0x400d48a9: node_task_init at /home/stefan/Equalizer_Roskilde/software/firmware/main/tasks/node_task/node_task.c:27

0x400d48c0: node_task at /home/stefan/Equalizer_Roskilde/software/firmware/main/tasks/node_task/node_task.c:36

I (2075) esp_core_dump_flash: Save core dump to flash...
I (2081) esp_core_dump_common: Found tasks: (17)!
I (2606) esp_core_dump_flash: Core dump has been saved to flash.
Rebooting...
```

koobest, 2020-01-06:
Hi, @StefanRvO 
You should call `esp_err_t rmt_driver_install(rmt_channel_t channel, size_t rx_buf_size, int intr_alloc_flags)` first.

StefanRvO, 2020-01-06:
Thank you. You are right, that fixes the problem. I wonder why it worked before then. Anyway, i will close this issue.

closed 2020-01-06 by StefanRvO

bbulkow, 2020-06-21:
The documentation in this area needs a serious update. The documentation specifically states that you MUST NOT call rmt_driver_install . The most common code in this area, FastLED, creates an interrupt handler but does not register an RMT interrupt handler.

From https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/rmt.html

If you already called rmt_driver_install to use system RMT driver, please do not register ISR handler again.