espressif/esp-idf #4584: esp_netif_get_dns_info returns garbage

itayperl, 2020-01-04:
Hi,

There's an incorrect memcpy in the implementation of `esp_netif_get_dns_info` causing garbage data to be returned to the caller.
The patch below fixes the issue.

```diff
diff --git a/components/esp_netif/lwip/esp_netif_lwip.c b/components/esp_netif/lwip/esp_netif_lwip.c
index 1ed30fb0b..680070ef2 100644
--- a/components/esp_netif/lwip/esp_netif_lwip.c
+++ b/components/esp_netif/lwip/esp_netif_lwip.c
@@ -1308,7 +1308,7 @@ static esp_err_t esp_netif_get_dns_info_api(esp_netif_api_msg_t *msg)
         const ip_addr_t*  dns_ip = NULL;
         dns_ip = dns_getserver(type);
         if(dns_ip != NULL) {
-            memcpy(&dns->ip.u_addr, &dns_ip, sizeof(ip4_addr_t));
+            memcpy(&dns->ip.u_addr, ip_2_ip4(dns_ip), sizeof(ip4_addr_t));
         }
     }
```

david-cermak, 2020-01-09:
Hi @itayperl 

Thanks for  reporting this issue and suggesting the fix! Will be corrected.

closed 2020-01-11 by espressif-bot

AxelLin, 2020-01-31:
Hi @david-cermak 
I'm checking the code on master tree, the fix looks wrong to me.
1. dns_getserver never return NULL, you should test dns_ip != IP_ADDR_ANY instead.
2. The dns->ip is "esp_ip_addr_t" but dns_ip (return from dns_get_server) is const ip_addr_t *.
   Are you sure it's 100% safe to do memcpy(&dns->ip, dns_ip, sizeof(ip_addr_t)); ??
   (If esp_ip_addr_t is exactly the same as ip_addr_t, why not use ip_addr_t instead.)
   (If esp_ip_addr_t is different from ip_addr_t in some configuration, above code is wrong.)

david-cermak, 2020-01-31:
Hi @AxelLin 

Thanks for looking at the code and raising the questions!
I still find the fix okay:

> dns_getserver never return NULL, you should test dns_ip != IP_ADDR_ANY instead.

* True, this could check could be skipped for this version of lwip, but it is fair I think to check for NULL when calling external API
* The `IP_ADDR_ANY` is a valid number, so no need for another check to have `esp_netif_get_dns_info_api()` return this number.

> The dns->ip is "esp_ip_addr_t" but dns_ip (return from dns_get_server) is const ip_addr_t *.

Yes it is safe when lwip is used as esp-idf network stack (which is true when the file `components/esp_netif/lwip/esp_netif_lwip.c` gets to compile. The ESP-NETIF aims to abstract the specific TCP/IP stack, that's why all these `esp_ip..` types.

yzazik, 2020-11-17:
About 2. In general it is not safe, looks very much like type punning. Besides, no other interface (or component apart from mdns) is using those "abstracted" esp_ip_* types, so it is not an abstraction, just different incompatible types and without translation are not usable. Not sure what kind of different IP address can be abstracted by those, bringing back IPX/SPX?