espressif/esp-idf #4585: Spurious interrupts on GPIO 36 when using WiFi

vonnieda, 2020-01-05:
## Environment

- Development Kit:  none
- Module or chip used: ESP32-WROVER-B
- IDF version: v3.1.1
- Build System: Make
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
- Operating System: Mac
- Power Supply: USB

## Problem Description

I'm getting 5 unexpected interrupts per second on GPIO 36 once WiFi connects. This does not happen before WiFi is connected.

I have seen https://github.com/espressif/esp-idf/issues/1096 and other posts but these seem to indicate that as long as I'm not using "the power switch which controls the temperature sensor, SARADC1 sensor, SARADC2 sensor, AMP sensor, or HALL sensor" this should not happen, but I am not using those and it still happens.

Example code below. It counts interrupts on GPIO 36 and 39 printing them every 500ms. After 5 counts it initializes WiFi and continues. In my testing, I get 0 interrupts before initializing WiFi and then get 5 per second once WiFi is connected.

```
#include "sdkconfig.h"
#include "esp_system.h"
#include "esp_log.h"
#include "esp_err.h"
#include "esp_wifi.h"
#include "esp_event_loop.h"
#include "nvs_flash.h"

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "freertos/semphr.h"

#include "driver/gpio.h"

static unsigned gpio_36_ints = 0;
static unsigned gpio_39_ints = 0;

static EventGroupHandle_t s_wifi_event_group;

static void IRAM_ATTR gpio_36_handler(void* arg) {
	gpio_36_ints++;
}

static void IRAM_ATTR gpio_39_handler(void* arg) {
	gpio_39_ints++;
}

static esp_err_t event_handler(void *ctx, system_event_t *event)
{
    switch(event->event_id) {
    case SYSTEM_EVENT_STA_START:
        esp_wifi_connect();
        break;
    case SYSTEM_EVENT_STA_GOT_IP:
        ESP_LOGI("", "got ip:%s", ip4addr_ntoa(&event->event_info.got_ip.ip_info.ip));
        break;
    case SYSTEM_EVENT_STA_DISCONNECTED:
		esp_wifi_connect();
		break;
    default:
        break;
    }
    return ESP_OK;
}

void init_wifi() {
	esp_err_t ret = nvs_flash_init();
	if (ret == ESP_ERR_NVS_NO_FREE_PAGES) {
		ret = nvs_flash_erase();
		if (ret == ESP_OK) {
			ret = nvs_flash_init();
		}
	}

	s_wifi_event_group = xEventGroupCreate();

	tcpip_adapter_init();
	ESP_ERROR_CHECK(esp_event_loop_init(event_handler, NULL) );

	wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
	ESP_ERROR_CHECK(esp_wifi_init(&cfg));
    ESP_ERROR_CHECK(esp_wifi_set_storage(WIFI_STORAGE_RAM));
	ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_STA));
	ESP_ERROR_CHECK(esp_wifi_start());
    ESP_ERROR_CHECK(esp_wifi_set_ps(WIFI_PS_MIN_MODEM));

	wifi_config_t wifi_config = {
		.sta = {
			.ssid = "AccessPoint",
			.password = "Password"
		},
	};
	ESP_ERROR_CHECK(esp_wifi_set_config(ESP_IF_WIFI_STA, &wifi_config));
	ESP_ERROR_CHECK(esp_wifi_connect());
}

void app_main()
{
	gpio_install_isr_service(0);

	gpio_config_t gpio_36_config = {
			.pin_bit_mask = (1ULL << GPIO_NUM_36),
			.mode = GPIO_MODE_INPUT,
			.pull_up_en = GPIO_PULLUP_DISABLE,
			.pull_down_en = GPIO_PULLDOWN_DISABLE,
			.intr_type = GPIO_INTR_POSEDGE,
	};
	gpio_config(&gpio_36_config);
	gpio_isr_handler_add(GPIO_NUM_36, gpio_36_handler, NULL);

	gpio_config_t gpio_39_config = {
			.pin_bit_mask = (1ULL << GPIO_NUM_39),
			.mode = GPIO_MODE_INPUT,
			.pull_up_en = GPIO_PULLUP_DISABLE,
			.pull_down_en = GPIO_PULLDOWN_DISABLE,
			.intr_type = GPIO_INTR_POSEDGE,
	};
	gpio_config(&gpio_39_config);
	gpio_isr_handler_add(GPIO_NUM_39, gpio_39_handler, NULL);

	init_wifi();

	int counter = 0;

	while (true) {
		printf("%u %u\n", gpio_36_ints, gpio_39_ints);
		vTaskDelay(pdMS_TO_TICKS(500));
		if (counter++ == 5) {
			init_wifi();
		}
	}
    return;
}
```

negativekelvin, 2020-01-05:
https://github.com/espressif/esp-idf/issues/3714

koobest, 2020-01-05:
Hi, @vonnieda 
This is a known hardware issue. Because WIFI will use ADC.

[image]

thanks !!

prayer521, 2020-01-05:
> Hi, @vonnieda
> This is a known hardware issue. Because WIFI will use ADC.
> 
> [image]
> 
> thanks !!

我用BLE蓝牙时出现了GPIO39不断中断触发的情况，难道BLE内部也用了ADC？

vonnieda, 2020-01-05:
Thank you @koobest.

Perhaps the documentation could be updated to explicitly state that the GPIO 36 and 39 issue is also caused by WiFi? It could have saved me a lot of time, and a board spin.

negativekelvin, 2020-01-05:
So is this only with power save or even without?  If adc is always on there will be no glitch right?

vonnieda, 2020-01-06:
@negativekelvin Just tested: If I explicitly set WIFI_PS_NONE it works fine. No spurious interrupts.

negativekelvin, 2020-01-06:
Ok good, I don't know why https://github.com/espressif/esp-idf/issues/3714 is closed, I don't think adc power lock has been implemented

igrr, 2020-01-07:
I'd like to close this as duplicate in favour of #4117 (IDFGH-1917). Although issue #3714 has the same cause and will be closed by the same fix, the described behavior is different. Will keep both open #4117 and #3714 open to make them both show in search.

closed 2020-01-07 by igrr