espressif/esp-idf #4586: SHA-256 of runing firmware is all zeroes

fpgamaster, 2020-01-05:
Secure boot is enabled!

## Environment
- Development Kit: ESP32-PICO-Kit
- Module or chip used:  ESP32-PICO-D4
- IDF version: v4.1-dev-1572-g30372f5a4
- Build System:  Make
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
- Operating System: Linux
- Power Supply: USB

## Problem Description
 esp_partition_get_sha256(esp_ota_get_running_partition(), sha_256);
returns all zeroes

### Expected Behavior
Actual SHA-256 of the partition

### Actual Behavior
ota: SHA-256 for current firmware: : 0000000000000000000000000000000000000000000000000000000000000000

### Steps to repropduce
Enable secure boot, print the result of sha_256 after execution of esp_partition_get_sha256(esp_ota_get_running_partition(), sha_256)

May be native_ota example with secure boot enabled can be used too.

StefanRvO, 2020-01-05:
This issue is known and will apparently soon be fixed: https://www.esp32.com/viewtopic.php?f=13&t=13250&start=10

projectgus, 2020-01-10:
Hi @fpgamastester ,

Thanks for the detailed bug report.

This should be fixed as of version `v4.1-dev-1686-g03f6179211`. Please check and we'll reopen if it's not the case.

Angus

closed 2020-01-10 by projectgus