espressif/esp-idf #4587: ethernet-lan8720: Auto-Negotiation & Auti-MDIX cannot be set from user code

JB-DX, 2020-01-06:
In IDF-v4.1 ethernet with lan8720  the speed defaults to 10Mb/s.
There's no way to set lan8720 to Auto-Negotiation-Enable .

Also Auto-MDIX seems to be disabled, at least it does not work.
I did not find anything in the documentation.

Both of the above worked with an earlier older IDF.

Are there any workarounds available?

suda-morris, 2020-01-06:
Auto-MDIX is enabled by default in [this register ](https://github.com/espressif/esp-idf/blob/master/components/esp_eth/src/esp_eth_phy_lan8720.c#L100), the driver doesn't happen to change its default value.

By default, Auto-negotiation is enabled by the MODE[2:0] configuration straps (so please check your schematic to see if you have disabled this feature in your hardware). 

if not, you can always manually issue a new round of negotiation by `phy->negotiate(phy)`.

If auto-negotiation is enabled by default, Lan8720 can also start auto-negotiation after a soft-reset [here](https://github.com/espressif/esp-idf/blob/master/components/esp_eth/src/esp_eth_phy_lan8720.c#L370).

JB-DX, 2020-01-07:
The hardware board has no configuration straps at all. 
The same board worked fine with an earlier IDF version. So it's only a software issue.
phy->negotiate(phy) does not help either. It is called right after esp_eth_driver_install().

There's a phy_reg_read(0,0,&v), how do I call it? 
I mean what what has to precede phy_reg_read?
Trying eth_handle->mediator.phy_reg_read(0, 0, &v) does not compile.
I want to check register-0

suda-morris, 2020-01-07:
Hi @JB-DX To read the phy register
```c
uint32_t reg_value = 0;
mac->read_phy_reg(mac,phy_addr,phy_reg,&reg_value);
```

JB-DX, 2020-01-09:
Thanks !
After more testing I found Auto-MDIX is indeed enabled. The other side was a laptop that had a larger activation delay when connecting the LAN cable. So it fooled me. I just should have waited longer.
Speed selection seems also to be correct, But I see it only rarely connecting at 100Mb/s, most of the time  it uses 10 Mb/s. I have no idea what the reason might be.
Is there a way to force it to 100Mb/s via IDF ?

--
A related question:
Our module should use either ethernet or Wifi.
After power-up it waits 3s for an ethernet connection,
if that fails (cable not connected) it switches to Wifi and stops using ethernet.
In that case I want to shut down ethernet to save power and to prevent LED1 to light-up when later on a cable is plugged in.
esp_eth_stop() is not sufficient.
esp_eth_driver_uninstall() reports errors that something is still attached.

What is a best way to achieve that?
I was considering pulling the LAN8720 reset to low. But is that wise?

suda-morris, 2020-01-09:
First question, for now IDF doesn't have a convenient way to force setting a link & duplex, but will have it in the near future. 

Second question, to totally uninstall the Ethernet driver, you can take this unit test as a reference: https://github.com/espressif/esp-idf/blob/master/components/esp_eth/test/test_emac.c#L204

JB-DX, 2020-01-13:
Thanks !
esp_eth_driver_uninstall() can't uninstall it even after a 4 sec. wait.
It seems something else needs to removed, but what?

The order of calls:
```
ESP_ERROR_CHECK(esp_eth_stop(eth_handle) );
ESP_ERROR_CHECK(esp_eth_del_netif_glue(eth_netif));
ESP_ERROR_CHECK(esp_eth_driver_uninstall(eth_handle));

``` 
Calling esp_event_handler_unregister() before uninstall did not help either.

This is the initialisation:
```
esp_netif_config_t cfg= ESP_NETIF_DEFAULT_ETH();
eth_netif= esp_netif_new(&cfg);
ESP_ERROR_CHECK(esp_eth_set_default_handlers(eth_netif));
ESP_ERROR_CHECK(esp_event_handler_register(ETH_EVENT, ESP_EVENT_ANY_ID, &eth_event_handler, NULL));

eth_mac_config_t mac_config= ETH_MAC_DEFAULT_CONFIG();
eth_phy_config_t phy_config= ETH_PHY_DEFAULT_CONFIG();
phy_config.phy_addr = 0; 
phy_config.reset_gpio_num= PIN_LNA8720_RESET; 

mac_config.smi_mdc_gpio_num= PIN_SMI_MDC; 
mac_config.smi_mdio_gpio_num= PIN_SMI_MDIO; 
mac= esp_eth_mac_new_esp32(&mac_config);
phy= esp_eth_phy_new_lan8720(&phy_config);
esp_eth_config_t config= ETH_DEFAULT_CONFIG(mac, phy);
ESP_ERROR_CHECK(esp_eth_driver_install(&config, &eth_handle));
ESP_ERROR_CHECK(esp_netif_attach(eth_netif, esp_eth_new_netif_glue(eth_handle)));
ESP_ERROR_CHECK(esp_eth_start(eth_handle));

```

suda-morris, 2020-01-14:
@JB-DX I recommend you to read the unit test first: https://github.com/espressif/esp-idf/blob/master/components/esp_eth/test/test_emac.c#L204

Your code is wrong, you should pass the "netif glue" to  `esp_eth_del_netif_glue()`, not "eth_netif".

JB-DX, 2020-01-14:
Oops that was my stupid mistake, that I didn't see.
It now works fine. That problem is solved.
Thank you very much for your patience.

BrunoMasut, 2020-03-04:
@JB-DX  You can set to accept only 100base-T writing reg 4:
1) Clear bits 5 and 6
2) Start negotiation

I don't know about your hw, but if it's not so good, maybe it will not work at 100mbps :(
Maybe in few boards (if you have many).

I have similar problem. My pcb is not so good and rarely works at 100mbps.
To solve, I trying force it to work at 10mbps!
But my lan8720 does not automatically swithing to 10mbps. Auto negotiation assert valid 100mbps, I can receive everything properly, but transmitted packets are not received by other side!

@suda-morris Note for your data rate selection implementation (10base-t or 100base-t): when I inform the '001' (10base-t only) mode of operation in register 18 of lan8720, the component is no longer able to detect the disconnection of the cable! I don't know if it's just my hardware, but the connection and disconnection of the cable is done correctly in 100base-t, disabling it, even maintaining auto negotiation, the system understands that the cable is always connected! It is not a software problem, reading the reg, the link status is 1.
If there's any solution, tell me. I'll try detect cable in second layer.

neorevx, 2020-03-04:
@JB-DX  You can set to accept only 100base-T writing reg 4:
1) Clear bits 5 and 6
2) Start negotiation

I don't know about your hw, but if it's not so good, maybe it will not work at 100mbps :(
Maybe in few boards (if you have many).

I have similar problem. My pcb is not so good and rarely works at 100mbps.
To solve, I trying force it to work at 10mbps!
But my lan8720 does not automatically swithing to 10mbps. Auto negotiation assert valid 100mbps, I can receive everything properly, but transmitted packets are not received by other side!

@suda-morris Note for your data rate selection implementation (10base-t or 100base-t): when I inform the '001' (10base-t only) mode of operation in register 18 of lan8720, the component is no longer able to detect the disconnection of the cable! I don't know if it's just my hardware, but the connection and disconnection of the cable is done correctly in 100base-t, disabling it, even maintaining auto negotiation, the system understands that the cable is always connected! It is not a software problem, reading the reg, the link status is 1.
If there's any solution, tell me. I'll try detect cable in second layer.

closed 2021-01-12 by Alvin1Zhang