espressif/esp-idf #4588: ESP32 receiving json file form local server

DheeranYuvaraj, 2020-01-06:
Hi,

I want to send json file from html local server to esp32. In this case esp32 working as station mode. 

Once it connects to the SSID it will starts html web server. That html page having 4 boxes for input the user data. After that if i press submit button in the html page esp has to receive those 4 datas in JSON format. 

please help me how can i achieve this.

suda-morris, 2020-01-06:
https://github.com/espressif/esp-idf/blob/master/examples/protocols/http_server/restful_server/main/rest_server.c#L109

DheeranYuvaraj, 2020-01-06:
Hi @suda-morris 

Our development is happening based on esp-idf-v3.3. Is this restful_server API's will work with idf-v3.3? If no how can we do it with idfv3.3?

suda-morris, 2020-01-06:
It's not in idf-v3.3, but I think it's still possible to work in 3.3 as long as you take care of the initialization stuff in the example (there're other http server examples in idf-v3.3 that can refer to).

DheeranYuvaraj, 2020-01-06:
Ok, Let me check.

closed 2020-01-27 by DheeranYuvaraj