espressif/esp-idf #4589: flash read err, 1000, RTCWDT_RTC_RESET

surekhaibot, 2020-01-06:
Hi, i am using ESP32-WROOM-32  flash size of 16MB, when power reset the module , i am getting  this issue.flash read err, 1000, RTCWDT_RTC_RESET

log follows: 

06-01-2020 18:37:27.909 [RX] - e??ts Jun  8 2016 00:22:57<CR><LF>
<CR><LF>
rst:0x1 (POWERON_RESET),boot:0x17 (SP??I_FAST_FLASH_BOOT)<CR><LF>
flash read err, 1000<CR><LF>
ets_main.c 371 <CR><LF>

06-01-2020 18:37:28.275 [RX] - ??ets? Jun  8 2016 00:22:57<CR><LF>
<CR><LF>
rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)<CR><LF>
configsip: 0, SPIWP:0xee<CR><LF>
clk_drv:0x00??,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00<CR><LF>
mo??d?e:DIO, clock div:2<CR><LF>
load:0x3fff0018,len:4<CR><LF>
load:0x3fff001c,len?:6276<CR><LF>
load:0x40078000,len:11624<CR><LF>
load:0x40080400,len:6648<CR><LF>
en?try 0x40080764<CR><LF>
<ESC>[0;32mI (29) boot: ESP-IDF GIT-NOTFOUND 2nd stage bootloader<ESC>[0m<CR><LF>

0xjakob, 2020-01-07:
@surekhaibot Hi, thanks for posting the bug! Do you have more information to reproduce this issue? Please take a look at the issue template if you don't know which information is important.

igrr, 2020-01-07:
This is a hardware issue described in section 3.8 of the ESP32 errata: https://www.espressif.com/sites/default/files/documentation/eco_and_workarounds_for_bugs_in_esp32_en.pdf. It is related to SPI flash startup time. Please see the document and also the explanations posted in https://github.com/espressif/esp-idf/issues/4441.

closed 2020-10-11 by projectgus