espressif/esp-idf #4590: new SIM800L example is not working (wont receive IP address)

Arthedian, 2020-01-06:
## Environment

- Development Kit:      none
- Module or chip used:  ESP32-WROOM-32D
- IDF version (run ``git describe --tags`` to find it): v4.1-dev-1757-g826b9f638
- Build System:    idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): (crosstool-NG esp-2019r2) 8.2.0
- Operating System:    Linux
- Using an IDE?: Clion, but compiling it from plain command line (idf.py)
- Power Supply: external 5V

## Problem Description

I am using some old PPPOS_client example (some v4.0, but not the latest), where if I have GSM signal, it will connect to the network and I can connect to the internet without problem. I had some other issues with it, so I updated esp idf to the newest version, and now pppos_client is not working. ESP32 wont receive IP address from the GSM network.

### Expected Behavior

work corectly and download things from the internet.

### Actual Behavior

THE ESP32 wont receive IP address from the GSM network and it will fail to connect through the PPP network.

### Steps to reproduce

1. use standard pppos_client example from esp idf.
2. I just add code to the beggining, which is restarting the SIM800L via RST pin. It will switch off for 5 seconds, and than start it on and waits for 60 seconds (just to be sure).

## Debug Logs

```
I (809) cpu_start: Starting scheduler on PRO CPU.
D (0) intr_alloc: Connected src 25 to int 2 (cpu 1)
I (0) cpu_start: Starting scheduler on APP CPU.
D (825) heap_init: New heap initialised at 0x3ffe0440
D (835) heap_init: New heap initialised at 0x3ffe4350
D (845) intr_alloc: Connected src 16 to int 12 (cpu 0)
I (845) gpio: GPIO[5]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
D (855) pppos_example: Resseting
D (5855) pppos_example: DOne resseting
D (65855) esp_netif_lwip: LwIP stack has been initialized
D (65855) esp_netif_lwip: esp-netif has been successfully initialized
D (65855) event: running task for loop 0x3ffb8b8c
D (65855) event: created task for loop 0x3ffb8b8c
D (65865) event: created event loop 0x3ffb8b8c
D (65865) esp_netif_objects: esp_netif_add_to_list 0x3ffb9b38
D (65875) esp_netif_objects: esp_netif_add_to_list netif added successfully (total netifs: 1)
D (65885) esp-netif_lwip-ppp: esp_netif_new_ppp: PPP connection created: 0x3ffb9e24
D (65885) esp-netif_lwip-ppp: Phase Dead
I (65895) uart: queue free spaces: 30
D (65895) intr_alloc: Connected src 35 to int 13 (cpu 0)
D (65905) event: created event loop 0x3ffbb608
D (65955) dce_service: sync ok
D (66055) dce_service: disable echo ok
D (66255) sim800: get module name ok
D (66455) sim800: get imei number ok
D (66655) sim800: get imsi number ok
D (66855) sim800: get network operator ok
D (66955) dce_service: set flow control ok
D (67055) dce_service: save settings ok
I (67055) pppos_example: Module: SIMCOM_SIM800L
I (67055) pppos_example: Operator: "EUROTEL PRAHA"
I (67055) pppos_example: IMEI: 865674032096113
I (67055) pppos_example: IMSI: 234500021973833
D (67255) sim800: inquire signal quality ok
I (67255) pppos_example: rssi: 18, ber: 0
D (67455) sim800: inquire battery status ok
I (67455) pppos_example: Battery voltage: 4158 mV
D (67555) dce_service: define pdp context ok
D (67655) sim800: enter ppp mode ok
D (67655) event: running post ESP_MODEM_EVENT:0 with handler 0x400d6358 on loop 0x3ffbb608
I (67655) pppos_example: Modem PPP Started
D (67655) event: running post ESP_MODEM_EVENT:0 with handler 0x400dadbc on loop 0x3ffbb608
D (67665) esp_netif_handlers: esp_netif action has started with netif0x3ffb9b38 from event_id=0
D (67675) esp-netif_lwip-ppp: esp_netif_start_ppp: Starting PPP connection: 0x3ffb9e24
D (67685) esp-netif_lwip-ppp: Phase Start
D (67685) esp-netif_lwip-ppp: Phase Establish
D (67795) esp-netif_lwip-ppp: Phase Authenticate
D (67845) esp-netif_lwip-ppp: Phase Network
D (68545) esp-netif_lwip-ppp: Phase Establish
D (74545) esp-netif_lwip-ppp: Phase Disconnect
D (74545) esp-netif_lwip-ppp: Phase Dead
I (74545) esp-netif_lwip-ppp: Connection lost
D (74545) event: running post IP_EVENT:6 with handler 0x400d63b4 on loop 0x3ffb8b8c
I (74555) pppos_example: IP event! 6
I (74555) pppos_example: Modem Disconnect from PPP Server
```

suda-morris, 2020-01-07:
There might be some regression in the uart driver, since 4.1 has introduced a HAL layer for drivers. 
A regular way to locate the issue is using `git bisect`, could you tell the commit id that works for you?

Arthedian, 2020-01-07:
Sadly I have no way how to find it, but probably I have some older version. I have tried example from v3.3.1, and it has the same problem. In my example, esp_modem doesnt write any DEBUG info about PPP phases.  I have SIM800L connected to GPIO1 and GPIO3, and outwards logging through UART on GPIO32 and GPIO35.

Arthedian, 2020-01-07:
I created a repo with working example (for me). https://github.com/Arthedian/esp32_gsm

amitcv, 2020-07-09:
I have exact same problem with same backtrace as above EDIT2. 
I am using esp-idf version v4.0 
@Arthedian did you have any new ideas?

Mluis31, 2020-07-22:
I have the same problem could it be that there's no place to input the provider APN?

edwin-oetelaar, 2020-07-31:
Same issues here. Latest idf. Verbose logging shows valid PPP connection. It seems callbacks are not fired. Will investigate further.

JonathanJol, 2020-10-11:
I had a similar problem using Nimbelink's BG96 modem, it would start the PPP connection, but I was never able to get IP data.

After quite some time reading and trying other things like just sending UART AT commands, I realized that the modem had flow control enabled by default, so I just turned on HW flow control in the example code and everything worked as intended after that, I hope this information can help anybody with similar problems.

devUsama, 2020-12-17:
Issue still exists with the espidf-v4.2 example

david-cermak, 2020-12-17:
I'm sorry, i've been neglecting this issue, as it's a bit difficult to say whether the connection issue is or is not on IDF side (SIM800L is largely obsolete and won't work in many countries, getting the IP address depends on the mobile operator, setting the APN, etc). We're going to introduce a new modem/console example in IDF which would just propagate the `AT` commands from the `esp-console` to the device. This should help us to tell apart the operator connection issues from PPPoS problems (if the connection succeeds via command line mode but fails in PPP mode is a clear indication of pppos/esp-modem issue).

As for the current status of the PPPoS example, the only outstanding issue is in clean exit of the data mode (https://github.com/espressif/esp-idf/issues/6024, also mentioned above I think) -- master.
* `v4.2` needs the same patch ([exit data mode](https://github.com/espressif/esp-idf/issues/6024#issuecomment-744414011)) plus backport of https://github.com/espressif/esp-idf/commit/e8834b66c583fdca31ef006d2a5cdb1e1af171de
* `v4.1` also the same patch and this [UART Tout fix](https://github.com/espressif/esp-idf/issues/6024#issuecomment-743316219)

I will revisit this issue when we have the console app ready.

svenehlers, 2021-06-15:
I have the same issue. Does anybody have a workaround for this?

david-cermak, 2021-06-25:
@svenehlers Please make sure you've correctly configured the `APN` (in the project configuration menu)
```
(Top) → Component config → ESP-MODEM
    Set Access Point Name (APN)
```
Also, please check with your operator that G2 connectivity is supported and enabled.

All issues mentioned [above](https://github.com/espressif/esp-idf/issues/4590#issuecomment-747649760) have been fixed and the `pppos-client` should work as expected. We'd still keep this open, until the `console` application mentioned above is ready.

For troubleshooting I'd suggest enabling PPP debug messages in lwip menu, enabling phase events and sharing your debug log. Although it might be hard to tell the connectivity and cellular issues apart, it might give us some pointers, at least.

david-cermak, 2021-09-14:
We're going to close this issue as [noted previously](https://github.com/espressif/esp-idf/issues/4590#issuecomment-747649760). There's no update from reporters on setting APN or checking for G2/GSM connectivity. Also the modem console application is now available as an example to the [esp_modem](https://components.espressif.com/component/espressif/esp_modem) component. Users can play with the device using the console app to check for connectivity issues and settings runtime.

Please feel free to reopen this if the console app could connect but pppos example won't get an IP.

closed 2021-09-14 by Alvin1Zhang

Arthedian, 2022-11-05:
@david-cermak I just resolved the problem I had at the beginning. Not sure if the problem is with the chip or the library. At the begging I am hardware resetting the chip (5s down, 5s up), then I cannot get the ip address. But if instead of 5 seconds waiting before connecting I will wait 15 seconds, than everything is working.