espressif/esp-idf #4591: idf_monitor does respect port setting when flashing

KloolK, 2020-01-06:
## Environment

- Development Kit: none
- Module or chip used: ESP32 (Heltec WiFi LoRa 32 V2)
- IDF version: v4.1-dev-1757-g826b9f638
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System: Linux
- Using an IDE?: CLion, but just for coding. Building and flashing is done from the terminal.
- Power Supply: USB

## Problem Description

I am using two Heltec WiFi LoRa 32 (V2) connected to the same PC (/dev/ttyUSB0 and /dev/ttyUSB1). I open the monitor with the following command:
`idf.py monitor -p /dev/ttyUSBx` with x = 0 and x = 1 in a second terminal.
I use Ctrl+T, Ctrl+A/Ctrl+F to build and flash.

### Expected Behavior
The code is rebuilt and flashed to the device connected to the port specified with the `-p` switch.

### Actual Behavior
The code is rebuilt and always flashed to the device on port `/dev/ttyUSB1`. The `-p` setting is ignored.

### Steps to reproduce

1. Connect two devices to the same PC. They should be assigned /dev/ttyUSB0 and /dev/ttyUSB1.
2. Open the monitor for both devices with the following two commands:
- `idf.py monitor -p /dev/ttyUSB0`
- `idf.py monitor -p /dev/ttyUSB1`
3. Press Ctrl+T, Ctrl+A in the monitor for `/dev/ttyUSB0`.
4. The code is built successfully, but flashed to the wrong device (`/dev/ttyUSB1`)

### Code to reproduce this issue

Any project.

## Debug Logs

No crashes.

## Other items if possible

N/A

KloolK, 2020-01-06:
I just noticed that when reversing the order of the arguments to `idf.py -p /dev/ttyUSBx monitor` it works as it should. However, I think this behavior is strange a maybe a consistent handling of the parameters  regardless of the order should be considered.

dobairoland, 2020-01-07:
Hi @KloolK. Yes you're right and flashing from IDF monitor should use the port specified for monitor. We already noticed this issue and the fix is under way. This GH issue will be closed when the fix gets pushed to GH.

> I just noticed that when reversing the order of the arguments to idf.py -p /dev/ttyUSBx monitor it works as it should. However, I think this behavior is strange and maybe a consistent handling of the parameters regardless of the order should be considered.

No, this is the intended behavior. Arguments after `idf.py` are global and local after a subcommand. At this moment using a global port bypasses the related bug.

For example, `port1` is a global port and `port2` is a locally overridden one for monitor in 
```
idf.py -p port1 monitor -p port2
```

Also, `--print-filter` is a local argument for `monitor`, i.e. `idf.py monitor --print-filter xy` can be used but `idf.py --print-filter xy monitor` makes no sense.

Alvin1Zhang, 2020-04-09:
@KloolK Thanks for reporting and sorry for the slow turnaround. Really appreciate your patience in waiting for the fix. The fix has been merged internally in our repo. and this issue will be automatically closed when the fix lands on Github. For your information. Thanks.

closed 2020-04-09 by espressif-bot