espressif/esp-idf #4592: wESP32 wont take IP address

avarsin, 2020-01-07:
## Environment

- Development Kit:      wESP32
- Module or chip used:   ESP32-WROOM-32 
- IDF version: v3.3.1
- Build System:         Make
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
- Operating System:     macOS
- Using an IDE?: No
- Power Supply:         USB

## Problem Description

Hiya,

I'm trying to troubleshoot a wESP32 and am having trouble getting it to get an IP Address on the wired ethernet interface. I'm using the Ethernet Basic example. 

I've tried with the wireless NIC and it connects fine.

I've configred the PHY pins etc and that all seemed to go fine.. though I'm not sure how to know if it didnt..

Any help would be much appreciated.

### Expected Behavior

Board should receive IP address via DHCP, it should show up in my DHCP server and also in the ESP's serial monitor.

### Actual Behavior

ESP doesnt get IP address, doesent show up in DHCP server and doesnt go past the serial monitor point below.

### Steps to reproduce

N/A

### Code to reproduce this issue

Code is the ethernet basic code.

https://github.com/espressif/esp-idf/tree/master/examples/ethernet/basic

## Debug Logs

I (0) cpu_start: App cpu up.
I (282) heap_init: Initializing. RAM available for dynamic allocation:
I (289) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (295) heap_init: At 3FFB4088 len 0002BF78 (175 KiB): DRAM
I (301) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (308) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (314) heap_init: At 40088968 len 00017698 (93 KiB): IRAM
I (320) cpu_start: Pro cpu start user code
I (3) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (6) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (16) emac: emac reset done
I (16) eth_example: Ethernet Started
I (4016) eth_example: Ethernet Link Up
I (84016) eth_example: Ethernet Link Down
I (92016) eth_example: Ethernet Link Up

## Other items if possible

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/4029547/sdkconfig.txt)

suda-morris, 2020-01-07:
Sorry, actually it's not a bug, the best place to get such help is the [forum](https://esp32.com/). And next time remember to post a picture of your board as we can't help you when we don't know what's your schematic design, especially for Ethernet...

And why you run an example on master branch with a v3.3 IDF? Ethernet driver has totally refactored since 4.0.

If this is your hardware: https://cdn.hackaday.io/files/853893653282976/wESP32-1.pdf
Then, the RMII clock is generated from ESP32 internal APLL, which should be output from GPIO0, so you need to select `CONFIG_PHY_CLOCK_GPIO0_OUT=y` instead of `CONFIG_PHY_CLOCK_GPIO0_IN=y`.

avarsin, 2020-01-07:
Sure thanks, sorry I didnt really know this wasnt the place.

Thanks though, I'll check out IDF 4.0

This is the board:
https://wesp32.com/files/wESP32-Product-Brief.pdf

AxelLin, 2020-08-04:
Looks like this is not a bug and should be closed. @Alvin1Zhang

Alvin1Zhang, 2020-08-05:
@AxelLin Thanks for the notes.

closed 2020-08-05 by Alvin1Zhang