espressif/esp-idf #4593: eth2ap - Slow traffic on RMII Clock GPIO 17

Jessenx94, 2020-01-07:
## Environment

- Development Kit:      [ESP32-POE-ISO (REV B), ESP32-EVB (REV F)]
 
- Module or chip used:  [ESP32-WROOM-32|ESP32-WROOM-32D]
- IDF version : v4
- Build System:         [Make|CMake|idf.py]
- Compiler version  xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
    
- Operating System:     [Windows]

- Using an IDE?: [NO]
- Power Supply:         [USB]

## Problem Description

Im trying to use the eth2ap example on the ESP32-POE-ISO board. I had run the example compiled and started it succesfully, but the connection is pretty slow. The homepage needs around 20sec to load. 
In comparison, using the ESP32-EVB board. Everything works fine without a big delay (homepage needs around 1-2sec). The difference between both boards is the RMII clock gpio. THE POE-ISO uses Output RMII clock from internal (GPIO17) and the EVB Input RMII clock from external on GPIO0. Changing the EVB to GPIO17 will result in the same loading issue as the POE-ISO board.

### Expected Behavior
Both boards should have the same transmission speed.
### Actual Behavior
GPIO17 on POE-ISO seems not to work.

### Code to reproduce this issue

```cpp

/* eth2ap (Ethernet to Wi-Fi AP packet forwarding) Example

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/
#include <string.h>
#include <stdlib.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/queue.h"
#include "esp_event_loop.h"
#include "esp_event.h"
#include "esp_log.h"
#include "esp_eth.h"
#include "esp_wifi.h"
#include "nvs_flash.h"
#include "esp_private/wifi.h"
#include "driver/gpio.h"
#include "sdkconfig.h"

static const char *TAG = "eth_example";
static esp_eth_handle_t s_eth_handle = NULL;
static xQueueHandle flow_control_queue = NULL;
static bool s_sta_is_connected = false;
static bool s_ethernet_is_connected = false;
static uint8_t s_eth_mac[6];
#define PIN_PHY_POWER 12
#define FLOW_CONTROL_QUEUE_TIMEOUT_MS (100)
#define FLOW_CONTROL_QUEUE_LENGTH (40)
#define FLOW_CONTROL_WIFI_SEND_TIMEOUT_MS (100)

typedef struct {
    void *packet;
    uint16_t length;
} flow_control_msg_t;

// Forward packets from Wi-Fi to Ethernet
static esp_err_t pkt_wifi2eth(void *buffer, uint16_t len, void *eb)
{
[... 869 lines cut ...]
CONFIG_TCP_OVERSIZE_MSS=y
# CONFIG_TCP_OVERSIZE_QUARTER_MSS is not set
# CONFIG_TCP_OVERSIZE_DISABLE is not set
CONFIG_UDP_RECVMBOX_SIZE=6
CONFIG_TCPIP_TASK_STACK_SIZE=3072
CONFIG_TCPIP_TASK_AFFINITY_NO_AFFINITY=y
# CONFIG_TCPIP_TASK_AFFINITY_CPU0 is not set
# CONFIG_TCPIP_TASK_AFFINITY_CPU1 is not set
CONFIG_TCPIP_TASK_AFFINITY=0x7FFFFFFF
# CONFIG_PPP_SUPPORT is not set
CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT=5
CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT=3072
CONFIG_ESP32_PTHREAD_STACK_MIN=768
CONFIG_ESP32_DEFAULT_PTHREAD_CORE_NO_AFFINITY=y
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_0 is not set
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_1 is not set
CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=-1
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS is not set
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is not set
CONFIG_IP_LOST_TIMER_INTERVAL=120
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y
# End of deprecated options

suda-morris, 2020-01-07:
Hi @Jessenx94 sounds weird, the clock configuration is very simple, only one or two register bit... I also have no idea for now. From my experience, RMII clock which sourced from external oscillator usually gives a better performance and stability. 
Anyway, thanks for reporting, we will look into this issue.

Jessenx94, 2020-01-07:
> Hi @Jessenx94 sounds weird, the clock configuration is very simple, only one or two register bit... I also have no idea for now. From my experience, RMII clock which sourced from external oscillator [...]

Im not 100% sure if the clock is the problem. But it works fine on the EVB, which uses the GPIO0 with an external clock. I cant see any other big differences.

Jessenx94, 2020-01-08:
Edit: the problem only exists when using a web server. When connecting the two boards to a "normal" network, there is no significant delay. 
I want to use the Poe-ISO to connect a webserver via Wifi. The webserver only has a Ethernet connector right now

closed 2021-06-23 by suda-morris