espressif/esp-idf #4594: E (19091) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x5180

Vbansal1996, 2020-01-07:
Hi 

I am trying to connect with the https over peap authenticated network and getting this issue.

I am trying to find out this error code 0x5180 but it is not listed in ssl.h file or not even get much info behind this.

https url and network are private so can't share the details.
But over same network I am able to connect with http url.

Any idea or help in this regard.
Thanks

AdityaHPatwardhan, 2020-01-07:
@Vbansal1996 , Thank you for the Issue , The error code you mentioned is linked to ` MBEDTLS_ERR_MD_ALLOC_FAILED ` as defined in `components/mbedtls/mbedtls/include/mbedtls/md.h`, You are having this error because your device is failing to allocate memory from heap at the time of handshake.(runtime)

We have some [Miscellaneous API](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/system/system.html#heap-memory) to check free heap size at  runtime.

for e.g. 
```
    uint32_t free_heap_size=0, min_free_heap_size=0;
    free_heap_size = esp_get_free_heap_size();
    min_free_heap_size = esp_get_minimum_free_heap_size(); 
    printf("\n free heap size = %d \t  min_free_heap_size = %d \n",free_heap_size,min_free_heap_size);                   
```
These functions are available in `esp_system.h`

AdityaHPatwardhan, 2020-01-07:
Please, verify that enough heap is available , and notify if the issue still persists. Thank you !

Vbansal1996, 2020-01-07:
What is the minimum heap requirement for the connection to get established?

AdityaHPatwardhan, 2020-01-07:
It is around 40KB - 45 KB !

Vbansal1996, 2020-01-07:
this is what I am getting 

[image]

AdityaHPatwardhan, 2020-01-07:
If that is your free heap size at the time of connecting to AP ( before any TLS part is started  ) I guess that is why you are facing the issue.

Vbansal1996, 2020-01-07:
40KB - 45KB free heap?

Vbansal1996, 2020-01-07:
yes this is the heap size after I am connected to AP, anyway to resolve this?

AdityaHPatwardhan, 2020-01-07:
I can list three options - 
1) (If you are not using `IDF-master` ) Please check that in `menuconfig -> Component config -> mbedTLS -> TLS max incoming frag. length` is set to 16384 and ` TLS maximum outgoing fragment length` is set to 4096 
2) you can perform idf.py size-components to get a detailed view of memory consumption by all libraries , with it you could try to free some heap 
3) If you are using `ESP32-WROVER`  module, then please enable SPIRAM in ` menuconfig -> Component config > ESP32-specific > Support for external, SPI-connected RAM `.

Vbansal1996, 2020-01-07:
1. checked menuconfig, values are set as expected.

My one more concern is, when I using open network other than PEAP and trying to access the test site https://www.howmyssl.com/a/check then I am able to establish the connection.

2. Can you please how to do that "idf.py size-components"

3. I am using NodeMCU 12E module

chegewara, 2020-01-07:
Really?
Node MCU 12E is esp8266, this is esp32 repository.

Vbansal1996, 2020-01-07:
Sorry for that! But both shared the same libraries and api for https request.

chegewara, 2020-01-07:
But are not sharing specs. Its really not the same, and problems you have with esp8266 can be completely different than with esp32 and for sure solutions are different.
You should fill this at beginning, which would save us 3 hours trying to help you:
```
- Development Kit:      [ESP32-Wrover-Kit|ESP32-DevKitC|ESP32-PICO-Kit|ESP32-LyraT|ESP32-LyraTD-MSC|none]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROOM-32|ESP32-WROOM-32D|ESP32-WROOM-32U|ESP32-WROVER|ESP32-WROVER-I|ESP32-WROVER-B|ESP32-WROVER-IB|ESP32-SOLO-1|ESP32-PICO-D4|ESP32]
- IDF version (run ``git describe --tags`` to find it):
    // v3.2-dev-1148-g96cd3b75c
- Build System:         [Make|CMake|idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     [Windows|Linux|macOS]
- (Windows only) environment type: [MSYS2 mingw32|ESP Command Prompt|Plain Command Prompt|PowerShell].
- Using an IDE?: [No|Yes (please give details)]
- Power Supply:         [USB|external 5V|external 3.3V|Battery]
```

AdityaHPatwardhan, 2020-01-07:
@Vbansal1996 Yes please consider what @chegewara has said for next time , but no issues anyways , When you will use esp32 you can remember this stuff ;) , Also according to me , reducing memory usage is a better option for you.

AdityaHPatwardhan, 2020-01-07:
@Alvin1Zhang Can you Please move this issue to ESP8266_RTOS_SDK as it is related to `esp8266` and not `esp32` , Thank You!

Vbansal1996, 2020-01-07:
OK Thanks for you help.

Alvin1Zhang, 2020-01-08:
@AdityaHPatwardhan @chegewara Thanks for the support and efforts. @Vbansal1996 Thanks for reporting, saw an issue https://github.com/espressif/ESP8266_RTOS_SDK/issues/806 had been created at ESP8266_RTOS_SDK. Will close this one. Thanks.

closed 2020-01-08 by Alvin1Zhang