espressif/esp-idf #4595: SPIFFS: failed to read addr 00006a00, size 00000100, err 257

GitOfLance, 2020-01-07:
### Environment
- Module or chip used: ESP32-WROOM-32
- IDF version :  v4.0-dev-1443-g39f090a4f 2nd stage bootloader
- Build System: Make
- Compiler version : xtensa-esp32-elf-
- Operating System: Ubuntu 18.04
- Power Supply: USB

### Problem Description:
Frequent errors occur when writing data to `SPIFFS` or reading data from `SPIFFS`. Error messages are as follows:
[image]

ErrorError 2:
[image]

Error 3:
[image]

I would like to ask, how to deal with this type of problem, and how to ensure that using SPIFFS to read and write data is almost 100% successful. If you do not use SPIFFS, but directly operate on Flash, will the stability be improved?

GitOfLance, 2020-02-24:
Is there progress on this issue?

Alvin1Zhang, 2020-02-24:
@GitOfLance Thanks for reporting and sorry for the late reply. We will look into. Thanks.

igrr, 2020-02-24:
Hi @GitOfLance, this is very likely the same issue as https://github.com/espressif/esp-idf/issues/4769. The fix for it should be merged shortly. In the meantime you can apply the change mentioned in https://github.com/espressif/esp-idf/issues/4769#issuecomment-586583027 locally.

GitOfLance, 2020-02-25:
Hi, @Alvin1Zhang @igrr 
It looks like this is a Flash read / write error caused by this bug. Thank you very much for your answers。

rebeliousconformist, 2020-04-29:
+1 
<pre>W (198897) wifi:alloc eb len=24 type=3 fail, heap:0

W (198897) wifi:m f null</pre>

Alvin1Zhang, 2021-01-13:
Fix mentioned in https://github.com/espressif/esp-idf/issues/4769 is available on GitHub, feel free to reopen.

closed 2021-01-13 by Alvin1Zhang