espressif/esp-idf #4596: Can the GATT CLIENT example used as HID Host?

mkardan, 2020-01-07:
Hi there,
I wonder if any of you used the GATT Client example below in order to implement HID Host please?
(https://github.com/espressif/esp-idf/blob/826b9f638/examples/bluetooth/bluedroid/ble/gatt_client/tutorial/Gatt_Client_Example_Walkthrough.md)

I wonder if this is easily achievable by modifying the GATT Client example as I am relatively new in this field and I have to implement a HID host in ESP32.

Many thanks,
FakeStar

chegewara, 2020-01-07:
Well, maybe its not you are looking for, because i did not use esp-idf, but recently i wrote HID keyboard host example with platformio and Kolban's library:
https://github.com/chegewara/esp32-hid-keyboard-client

It is easy to write host library for 1 single device, like keyboard or mouse, when you know HID report before you start and you dont have to parse it in real time. When you want to write universal HID host library it may be hard, especially if you dont know HID specification on expert level.

mkardan, 2020-01-07:
Aww.. Thanks a lot @chegewara for your prompt reply and the link to your HID keyboard host example. It sounds very promising as I only need to connect to one HID barcode scanner. I will give this a go and report accordingly- :)

chegewara, 2020-01-07:
Then it should be not so hard, but like i said, you have to know and understand HID report from that device.

mkardan, 2020-01-07:
Perfect, I will do.  I
f its OK I keep this thread open to ask my questions?

chegewara, 2020-01-07:
I believe that forum is better place. Maybe here:
https://esp32.com/viewtopic.php?f=18&t=13598&p=53566&hilit=hid#p53566

or new topic.

mkardan, 2020-01-07:
Perfect sure, so I close this now! :)

closed 2020-01-07 by mkardan