espressif/esp-idf #4601: post issue #4389 update error

rebeliousconformist, 2020-01-08:
## Environment

- Development Kit:  none
- Kit version (for WroverKit/PicoKit/DevKitC): custom
- Module or chip used:  ESP32-WROOM-32
- IDF version : v4.1-dev-1770-g71b4768df
- Build System:         Make
- Compiler version : crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0

- Operating System:     |Linux

- Using an IDE?:eclipse as per esp idf page
- Power Supply:         external 5V|

## Problem Description

//Detailed problem description goes here.

### Expected Behavior
compiles

### Actual Behavior
```cpp
```
xtensa-esp32-elf-gcc -std=gnu99 -Og -ggdb -Wno-frame-address -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -Wno-old-style-declaration -DMULTI_HEAP_FREERTOS -DESP_PLATFORM -D IDF_VER=\"v4.1-dev-1770-g71b4768df\" -MMD -MP   -D_GNU_SOURCE -DMBEDTLS_CONFIG_FILE='"mbedtls/esp_config.h"' -DHAVE_CONFIG_H -DUNITY_INCLUDE_CONFIG_H -I /home/pemubuntu/espressif/esp-idf/components/heap/include -I /home/pemubuntu/espressif/esp-idf/components/app_trace/include -I /home/pemubuntu/espressif/esp-idf/components/app_update/include -I /home/pemubuntu/espressif/esp-idf/components/asio/asio/asio/include -I /home/pemubuntu/espressif/esp-idf/components/asio/port/include -I /home/pemubuntu/espressif/esp-idf/components/bootloader_support/include -I /home/pemubuntu/espressif/esp-idf/components/bt/include -I /home/pemubuntu/espressif/esp-idf/components/cbor/port/include -I /home/pemubuntu/espressif/esp-idf/components/coap/port/include -I /home/pemubuntu/espressif/esp-idf/components/coap/port/include/coap -I /home/pemubuntu/espressif/esp-idf/components/coap/libcoap/include -I /home/pemubuntu/espressif/esp-idf/components/coap/libcoap/include/coap2 -I /home/pemubuntu/espressif/esp-idf/components/console -I /home/pemubuntu/espressif/esp-idf/components/driver/include -I /home/pemubuntu/espressif/esp-idf/components/driver/esp32/include -I /home/pemubuntu/espressif/esp-idf/components/efuse/esp32/include -I /home/pemubuntu/espressif/esp-idf/components/efuse/include -I /home/pemubuntu/espressif/esp-idf/components/esp-tls -I /home/pemubuntu/espressif/esp-idf/components/esp-tls/private_include -I /home/pemubuntu/espressif/esp-idf/components/esp32/include -I /home/pemubuntu/espressif/esp-idf/components/esp_adc_cal/include -I /home/pemubuntu/espressif/esp-idf/components/esp_common/include -I /home/pemubuntu/espressif/esp-idf/components/esp_eth/include -I /home/pemubuntu/espressif/esp-idf/components/esp_event/include -I /home/pemubuntu/espressif/esp-idf/components/esp_gdbstub/include -I /home/pemubuntu/espressif/esp-idf/components/esp_http_client/include -I /home/pemubuntu/espressif/esp-idf/components/esp_http_server/include -I /home/pemubuntu/espressif/esp-idf/components/esp_https_ota/include -I /home/pemubuntu/espressif/esp-idf/components/esp_local_ctrl/include -I /home/pemubuntu/espressif/esp-idf/components/esp_netif/include -I /home/pemubuntu/espressif/esp-idf/components/esp_ringbuf/include -I /home/pemubuntu/espressif/esp-idf/components/esp_rom/include -I /home/pemubuntu/espressif/esp-idf/components/esp_serial_slave_link/include -I /home/pemubuntu/espressif/esp-idf/components/esp_websocket_client/include -I /home/pemubuntu/espressif/esp-idf/components/esp_wifi/include -I /home/pemubuntu/espressif/esp-idf/components/esp_wifi/esp32/include -I /home/pemubuntu/espressif/esp-idf/components/espcoredump/include -I /home/pemubuntu/espressif/esp-idf/components/expat/expat/expat/lib -I /home/pemubuntu/espressif/esp-idf/components/expat/port/include -I /home/pemubuntu/espressif/esp-idf/components/fatfs/diskio -I /home/pemubuntu/espressif/esp-idf/components/fatfs/vfs -I /home/pemubuntu/espressif/esp-idf/components/fatfs/src -I /home/pemubuntu/espressif/esp-idf/components/freemodbus/common/include -I /home/pemubuntu/espressif/esp-idf/components/freertos/include -I /home/pemubuntu/espressif/esp-idf/components/idf_test/include -I /home/pemubuntu/espressif/esp-idf/components/jsmn/include -I /home/pemubuntu/espressif/esp-idf/components/json/cJSON -I /home/pemubuntu/espressif/esp-idf/components/libsodium/libsodium/src/libsodium/include -I /home/pemubuntu/espressif/esp-idf/components/libsodium/port_include -I /home/pemubuntu/espressif/esp-idf/components/log/include -I /home/pemubuntu/espressif/esp-idf/components/lwip/include/apps -I /home/pemubuntu/espressif/esp-idf/components/lwip/include/apps/sntp -I /home/pemubuntu/espressif/esp-idf/components/lwip/lwip/src/include -I /home/pemubuntu/espressif/esp-idf/components/lwip/port/esp32/include -I /home/pemubuntu/espressif/esp-idf/components/lwip/port/esp32/include/arch -I /home/pemubuntu/espressif/workspace/esp32-cpp/main/include -I /home/pemubuntu/espressif/esp-idf/components/mbedtls/port/include -I /home/pemubuntu/espressif/esp-idf/components/mbedtls/mbedtls/include -I /home/pemubuntu/espressif/esp-idf/components/mdns/include -I /home/pemubuntu/espressif/esp-idf/components/mqtt/esp-mqtt/include -I /home/pemubuntu/espressif/esp-idf/components/newlib/platform_include -I /home/pemubuntu/espressif/esp-idf/components/nghttp/port/include -I /home/pemubuntu/espressif/esp-idf/components/nghttp/nghttp2/lib/includes -I /home/pemubuntu/espressif/esp-idf/components/nvs_flash/include -I /home/pemubuntu/espressif/esp-idf/components/openssl/include -I /home/pemubuntu/espressif/esp-idf/components/perfmon/include -I /home/pemubuntu/espressif/esp-idf/components/protobuf-c/protobuf-c -I /home/pemubuntu/espressif/esp-idf/components/protocomm/include/common -I /home/pemubuntu/espressif/esp-idf/components/protocomm/include/security -I /home/pemubuntu/espressif/esp-idf/components/protocomm/include/transports -I /home/pemubuntu/espressif/esp-idf/components/pthread/include -I /home/pemubuntu/espressif/esp-idf/components/sdmmc/include -I /home/pemubuntu/espressif/esp-idf/components/soc/esp32/include -I /home/pemubuntu/espressif/esp-idf/components/soc/include -I /home/pemubuntu/espressif/esp-idf/components/spi_flash/include -I /home/pemubuntu/espressif/esp-idf/components/spiffs/include -I /home/pemubuntu/espressif/esp-idf/components/tcp_transport/include -I /home/pemubuntu/espressif/esp-idf/components/tcpip_adapter/include -I /home/pemubuntu/espressif/esp-idf/components/ulp/include -I /home/pemubuntu/espressif/esp-idf/components/unity/include -I /home/pemubuntu/espressif/esp-idf/components/unity/unity/src -I /home/pemubuntu/espressif/esp-idf/components/vfs/include -I /home/pemubuntu/espressif/esp-idf/components/wear_levelling/include -I /home/pemubuntu/espressif/esp-idf/components/wifi_provisioning/include -I /home/pemubuntu/espressif/esp-idf/components/wpa_supplicant/include -I /home/pemubuntu/espressif/esp-idf/components/wpa_supplicant/port/include -I /home/pemubuntu/espressif/esp-idf/components/wpa_supplicant/include/esp_supplicant -I /home/pemubuntu/espressif/esp-idf/components/xtensa/include -I /home/pemubuntu/espressif/esp-idf/components/xtensa/esp32/include -I /home/pemubuntu/espressif/workspace/esp32-cpp/build/include  -I . -c /home/pemubuntu/espressif/esp-idf/components/heap/multi_heap.c -o multi_heap.o
/home/pemubuntu/espressif/esp-idf/make/component_wrapper.mk:291: recipe for target 'heap_caps_init.o' failed
In file included from /home/pemubuntu/espressif/esp-idf/components/heap/heap_private.h:20:0,
xtensa-esp32-elf-gcc -DESP_PLATFORM -D IDF_VER=\"v4.1-dev-1770-g71b4768df\" -MMD -MP   -D_GNU_SOURCE -DMBEDTLS_CONFIG_FILE='"mbedtls/esp_config.h"' -DHAVE_CONFIG_H -DUNITY_INCLUDE_CONFIG_H -ggdb -I /home/pemubuntu/espressif/esp-idf/components/freertos/include/freertos -I /home/pemubuntu/espressif/esp-idf/components/freertos -I /home/pemubuntu/espressif/esp-idf/components/freertos/include -I /home/pemubuntu/espressif/esp-idf/components/app_trace/include -I /home/pemubuntu/espressif/esp-idf/components/app_update/include -I /home/pemubuntu/espressif/esp-idf/components/asio/asio/asio/include -I /home/pemubuntu/espressif/esp-idf/components/asio/port/include -I /home/pemubuntu/espressif/esp-idf/components/bootloader_support/include -I /home/pemubuntu/espressif/esp-idf/components/bt/include -I /home/pemubuntu/espressif/esp-idf/components/cbor/port/include -I /home/pemubuntu/espressif/esp-idf/components/coap/port/include -I /home/pemubuntu/espressif/esp-idf/components/coap/port/include/coap -I /home/pemubuntu/espressif/esp-idf/components [... cut ...]
     MULTI_HEAP_UNLOCK(&registered_heaps_write_lock);
     ^
/home/pemubuntu/espressif/esp-idf/components/heap/heap_caps_init.c: At top level:
cc1: warning: unrecognized command line option '-Wno-frame-address'
cc1: some warnings being treated as errors
make[1]: *** [heap_caps_init.o] Error 1
make[1]: *** Waiting for unfinished jobs....

```cpp
```

### Steps to reproduce

1. step1
2. ...

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp

```

igrr, 2020-01-08:
For IDF version v4.1-dev, the correct toolchain version is esp-2019r2 (based on GCC 8.2). Please follow the getting started guide to install the toolchain, and then try building the project again.

https://docs.espressif.com/projects/esp-idf/en/latest/get-started/index.html#get-started-set-up-tools

rebeliousconformist, 2020-01-08:
now i get the following errors

```cpp
/home/pemubuntu/espressif/esp-idf/components/log/include/esp_log.h:281:13: error: 'len' may be used uninitialized in this function [-Werror=maybe-uninitialized]
             ets_printf(LOG_FORMAT(log_tag_letter, format), esp_log_timestamp(), tag, ##__VA_ARGS__); \
             ^~~~~~~~~~
/home/pemubuntu/espressif/esp-idf/components/soc/src/hal/sdio_slave_hal.c:375:14: note: 'len' was declared here
     uint32_t len;
              ^~~
xtensa-esp32-elf-gcc -std=gnu99 -Og -ggdb -Wno-frame-address -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -Wno-old-style-declaration -DCONFIG_WPA3_SAE -DCONFIG_IEEE80211W -DESP_SUPPLICANT -DIEEE8021X_EAPOL -DEAP_PEER_METHOD -DEAP_TLS -DEAP_TTLS -DEAP_PEAP -DEAP_MSCHAPv2 -DUSE_WPA2_TASK -DCONFIG_WPS2 -DCONFIG_WPS_PIN -DUSE_WPS_TASK -DESPRESSIF_USE -DESP32_WORKAROUND -DCONFIG_ECC -D__ets__ -Wno-strict-aliasing -DESP_PLATFORM -D IDF_VER=\"v4.1-dev-1770-g71b4768df\" -MMD -MP   -D_GNU_SOURCE -DMBEDTLS_CONFIG_FILE='"mbedtls/esp_config.h"' -DHAVE_CONFIG_H -DUNITY_INCLUDE_CONFIG_H -I /home/pemubuntu/espressif/esp-idf/components/wpa_supplicant/src -I /home/pemubuntu/espressif/esp-idf/components/wpa_supplicant/include -I /home/pemubuntu/espressif/esp-idf/components/wpa_supplicant/port/include -I /home/pemubuntu/espressif/esp-idf/components/wpa_supplicant/include/esp_supplicant -I /home/pemubuntu/espressif/esp-idf/components/app_trace/include -I /home/pemubuntu/espressif/esp-idf/components/app_update/include -I /home/pemubuntu/espressif/esp-idf/components/asio/asio/asio/include -I /home/pemubuntu/espressif/esp-idf/components/asio/port/include -I /home/pemubuntu/espressif/esp-idf/components/bootloader_support/include -I /home/pemubuntu/espressif/esp-idf/components/bt/include -I /home/pemubuntu/espressif/esp-idf/components/cbor/port/include -I /home/pemubuntu/espressif/esp-idf/components/coap/port/include -I /home/pemubuntu/espressif/esp-idf/components/coap/port/include/coap -I /home/pemubuntu/espressif/esp-idf/components/coap/libcoap/include -I /home/pemubuntu/espressif/esp-idf/components/coap/libcoap/include/coap2 -I /home/pemubuntu/espressif/esp-idf/components/console -I /home/pemubuntu/espressif/esp-idf/components/driver/include -I /home/pemubuntu/espressif/esp-idf/components/driver/esp32/include -I /home/pemubuntu/espressif/esp-idf/components/efuse/esp32/include -I /home/pemubuntu/espressif/esp-idf/components/efuse/include -I /home/pemubuntu/espressif/esp-idf/components/esp-tls -I /home/pemubuntu/espressif/esp-idf/components/esp-tls/private_include -I /home/pemubuntu/espressif/esp-idf/components/esp32/include -I /home/pemubuntu/espressif/esp-idf/components/esp_adc_cal/include -I /home/pemubuntu/espressif/esp-idf/components/esp_common/include -I /home/pemubuntu/espressif/esp-idf/components/esp_eth/include -I /home/pemubuntu/espressif/esp-idf/components/esp_event/include -I /home/pemubuntu/espressif/esp-idf/components/esp_gdbstub/include -I /home/pemubuntu/espressif/esp-idf/components/esp_http_client/include -I /home/pemubuntu/espressif/esp-idf/components/esp_http_server/include -I /home/pemubuntu/espressif/esp-idf/components/esp_https_ota/include -I /home/pemubuntu/espressif/esp-idf/components/esp_local_ctrl/include -I /home/pemubuntu/espressif/esp-idf/components/esp_netif/include -I /home/pemubuntu/espressif/esp-idf/components/esp_ringbuf/include -I /home/pemubuntu/espressif/esp-idf/components/esp_rom/include -I /home/pemubuntu/espressif/esp-idf/components/esp_serial_slave_link/include -I /home/pemubuntu/espressif/esp-idf/components/esp_websocket_client/include -I /home/pemubuntu/espressif/esp-idf/components/esp_wifi/include -I /home/pemubuntu/espressif/esp-idf/components/esp_wifi/esp32/include -I /home/pemubuntu/espressif/esp-idf/components/espcoredump/include -I /home/pemubuntu/espressif/esp-idf/components/expat/expat/expat/lib -I /home/pemubuntu/espressif/esp-idf/components/expat/port/include -I /home/pemubuntu/espressif/esp-idf/components/fatfs/diskio -I /home/pemubuntu/espressif/esp-idf/components/fatfs/vfs -I /home/pemubuntu/espressif/esp-idf/components/fatfs/src -I /home/pemubuntu/espressif/esp-idf/components/freemodbus/common/include -I /home/pemubuntu/espressif/esp-idf/components/freertos/include -I /home/pemubuntu/espressif/esp-idf/components/heap/include -I /home/pemubuntu/espressif/esp-idf/components/idf_test/include -I /home/pemubuntu/espressif/esp-idf/components/jsmn/include -I /home/pemubuntu/espressif/esp-idf/components/json/cJSON -I /home/pemubuntu/espressif/esp-idf/components/libsodium/libsodium/src/libsodium/include -I /home/pemubuntu/espressif/esp-idf/components/libsodium/port_include -I /home/pemubuntu/espressif/esp-idf/components/log/include -I /home/pemubuntu/espressif/esp-idf/components/lwip/include/apps -I /home/pemubuntu/espressif/esp-idf/components/lwip/include/apps/sntp -I /home/pemubuntu/espressif/esp-idf/components/lwip/lwip/src/include -I /home/pemubuntu/espressif/esp-idf/components/lwip/port/esp32/include -I /home/pemubuntu/espressif/esp-idf/components/lwip/port/esp32/include/arch -I /home/pemubuntu/espressif/workspace/esp32-cpp/main/include -I /home/pemubuntu/espressif/esp-idf/components/mbedtls/port/include -I /home/pemubuntu/espressif/esp-idf/components/mbedtls/mbedtls/include [... cut ...]
cc1: some warnings being treated as errors
make[1]: *** [src/hal/sdio_slave_hal.o] Error 1
/home/pemubuntu/espressif/esp-idf/make/component_wrapper.mk:291: recipe for target 'src/hal/sdio_slave_hal.o' failed
make[1]: *** Waiting for unfinished jobs....
make[1]: Leaving directory '/home/pemubuntu/espressif/workspace/esp32-cpp/build/soc'
make: *** [component-soc-build] Error 2
/home/pemubuntu/espressif/esp-idf/make/project.mk:604: recipe for target 'component-soc-build' failed
make: *** Waiting for unfinished jobs....

```

igrr, 2020-01-08:
Yes, the issue you see now has already been reported in https://github.com/espressif/esp-idf/issues/4552. Until it is fixed, the workaround is to decrease log level from Verbose to Debug.

rebeliousconformist, 2020-01-09:
HI Igor thanks for the prompt reply,
i now have linker issues 
```
/home/pemubuntu/espressif/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/pemubuntu/espressif/workspace/esp32-cpp/build/main/libmain.a(wifiCore.o):(.literal._ZN8wifiCore27script_get_handler_functionEP9httpd_req[wifiCore::script_get_handler_function(httpd_req*)]+0x0): undefined reference to `_binary_jquery_min_js_end'
/home/pemubuntu/espressif/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld: /home/pemubuntu/espressif/workspace/esp32-cpp/build/main/libmain.a(wifiCore.o):(.literal._ZN8wifiCore27script_get_handler_functionEP9httpd_req[wifiCore::script_get_handler_function(httpd_req*)]+0x4): undefined reference to `_binary_jquery_min_js_start'
```

igrr, 2020-01-09:
These point to one of your source files (wifiCore in main component) and seem to be related to embedding a file. Please post your component makefile, perhaps there is an issue there.

rebeliousconformist, 2020-01-09:
solved it. added the embedded files to the Cmake. Thanks !

closed 2020-01-09 by rebeliousconformist