espressif/esp-idf #4602: Instalation of ESP-IDF tools v2.2 fails with exit code 1

lmagalhaes1337, 2020-01-08:
Hi there,

On Windows 10 64 bits, the instalation is failing with exit code 1.

Log:

2020-01-08 12:45:40.785   --------
2020-01-08 12:45:40.794   Installing Python environment:C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.7_3.7.1520.0_x64__qbz5n2kfra8p0\python.exe C:\esp_idf\tools\idf_tools.py install-python-env
2020-01-08 12:45:41.830   Done, exit code=1
2020-01-08 12:45:41.830   --------
2020-01-08 12:45:41.830   Running command: C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.7_3.7.1520.0_x64__qbz5n2kfra8p0\python.exe C:\esp_idf\tools\idf_tools.py install-python-env
                          Traceback (most recent call last):
                            File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.7_3.7.1520.0_x64__qbz5n2kfra8p0\lib\runpy.py", line 193, in _run_module_as_main
                              "__main__", mod_spec)
                            File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.7_3.7.1520.0_x64__qbz5n2kfra8p0\lib\runpy.py", line 85, in _run_code
                              exec(code, run_globals)
                            File "C:\Users\Sérgio Sobral\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.7_qbz5n2kfra8p0\LocalCache\local-packages\Python37\site-packages\virtualenv.py", line 2634, in <module>
                              main()
                            File "C:\Users\Sérgio Sobral\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.7_qbz5n2kfra8p0\LocalCache\local-packages\Python37\site-packages\virtualenv.py", line 870, in main
                              symlink=options.symlink,
                            File "C:\Users\Sérgio Sobral\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.7_qbz5n2kfra8p0\LocalCache\local-packages\Python37\site-packages\virtualenv.py", line 1162, in create_environment
                              install_python(home_dir, lib_dir, inc_dir, bin_dir, site_packages=site_packages, clear=clear, symlink=symlink)
                            File "C:\Users\Sérgio Sobral\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.7_qbz5n2kfra8p0\LocalCache\local-packages\Python37\site-packages\virtualenv.py", line 1532, in install_python
                              shutil.copyfile(executable, py_executable)
                            File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.7_3.7.1520.0_x64__qbz5n2kfra8p0\lib\shutil.py", line 120, in copyfile
                              with open(src, 'rb') as fsrc:
                          OSError: [Errno 22] Invalid argument: 'C:\\Users\\Sérgio Sobral\\AppData\\Local\\Microsoft\\WindowsApps\\PythonSoftwareFoundation.Python.3.7_qbz5n2kfra8p0\\python.exe'
                          Using base prefix 'C:\\Program Files\\WindowsApps\\PythonSoftwareFoundation.Python.3.7_3.7.1520.0_x64__qbz5n2kfra8p0'
                          New python executable in C:\Users\SRGIOS~1\ESPRES~1\PYTHON~1\IDF33_~1.7_E\Scripts\python.exe
                          Creating a new Python environment in C:\Users\Sérgio Sobral\.espressif\python_env\idf3.3_py3.7_env
                          Traceback (most recent call last):
                            File "C:\esp_idf\tools\idf_tools.py", line 1349, in <module>
                              main(sys.argv[1:])
                            File "C:\esp_idf\tools\idf_tools.py", line 1345, in main
                              action_func(args)
                            File "C:\esp_idf\tools\idf_tools.py", line 1156, in action_install_python_env
                              stdout=sys.stdout, stderr=sys.stderr)
                            File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.7_3.7.1520.0_x64__qbz5n2kfra8p0\lib\subprocess.py", line 363, in check_call
                              raise CalledProcessError(retcode, cmd)
                          subprocess.CalledProcessError: Command '['C:\\Users\\Sérgio Sobral\\AppData\\Local\\Microsoft\\WindowsApps\\PythonSoftwareFoundation.Python.3.7_qbz5n2kfra8p0\\python.exe', '-m', 'virtualenv', '--no-site-packages', 'C:\\Users\\Sérgio Sobral\\.espressif\\python_env\\idf3.3_py3.7_env']' returned non-zero exit status 1.
                          
2020-01-08 12:45:41.830   --------
2020-01-08 12:45:41.830   Message box (OK):
                          Installation has failed with exit code 1
2020-01-08 12:47:25.197   User chose OK.
2020-01-08 12:47:25.211   Message box (Yes/No):
                          Installation log has been created, it may contain more information about the problem.
                          Display the installation log now?
2020-01-08 12:47:26.081   User chose Yes.
2020-01-08 12:47:26.125   CurStepChanged raised an exception.

[Setup Log 2020-01-08 #003.txt](https://github.com/espressif/esp-idf/files/4035341/Setup.Log.2020-01-08.003.txt)

igrr, 2020-01-08:
Thanks for reporting. Looks like there is a compatibility issue when Python installed from Windows Store is used. Will investigate.

In the meantime, installing another copy of Python from python.org and letting the installer use it instead should work around the issue.

lmagalhaes1337, 2020-01-14:
Hi there,

I unnistalled the python from the windows store, downloaded the latest python (3.8.1) from python.org. Now it gives another error. First it started with a "ModuleNotFoundError: No module named 'runpy'" error, but followed your instructions here: [https://www.esp32.com/viewtopic.php?t=12481] and fixed it. But not, it gives me another error with the module "pkgutil" and I can't even install it with pip.

`2020-01-14 10:40:49.529   idf_tools.py command: C:\Users\Sérgio Sobral\AppData\Local\Programs\Python\Python38\python.exe C:\esp\tools\idf_tools.py
2020-01-14 10:40:49.529   Installing tools:C:\Users\Sérgio Sobral\AppData\Local\Programs\Python\Python38\python.exe C:\esp\tools\idf_tools.py install
2020-01-14 10:41:04.087   Done, exit code=0
2020-01-14 10:41:04.087   --------
2020-01-14 10:41:04.087   Running command: C:\Users\Sérgio Sobral\AppData\Local\Programs\Python\Python38\python.exe C:\esp\tools\idf_tools.py install
                          Installing tools: xtensa-esp32-elf, esp32ulp-elf, cmake, openocd-esp32, mconf, ninja, idf-exe, ccache
                          Installing xtensa-esp32-elf@1.22.0-80-g6c4433a5-5.2.0
                          file xtensa-esp32-elf-win32-1.22.0-80-g6c4433a-5.2.0.zip is already downloaded
                          Extracting C:\esp\.espressif\dist\xtensa-esp32-elf-win32-1.22.0-80-g6c4433a-5.2.0.zip to C:\esp\.espressif\tools\xtensa-esp32-elf\1.22.0-80-g6c4433a5-5.2.0
                          Installing esp32ulp-elf@2.28.51.20170517
                          file binutils-esp32ulp-win32-2.28.51-esp32ulp-20180809.zip is already downloaded
                          Extracting C:\esp\.espressif\dist\binutils-esp32ulp-win32-2.28.51-esp32ulp-20180809.zip to C:\esp\.espressif\tools\esp32ulp-elf\2.28.51.20170517
                          Installing cmake@3.13.4
                          file cmake-3.13.4-win64-x64.zip is already downloaded
                          Extracting C:\esp\.espressif\dist\cmake-3.13.4-win64-x64.zip to C:\esp\.espressif\tools\cmake\3.13.4
                          Installing openocd-esp32@v0.10.0-esp32-20190313
                          file openocd-esp32-win32-0.10.0-esp32-20190313.zip is already downloaded
                          Extracting C:\esp\.espressif\dist\openocd-esp32-win32-0.10.0-esp32-20190313.zip to C:\esp\.espressif\tools\openocd-esp32\v0.10.0-esp32-20190313
                          Installing mconf@v4.6.0.0-idf-20190628
                          file mconf-v4.6.0.0-idf-20190628-win32.zip is already downloaded
                          Extracting C:\esp\.espressif\dist\mconf-v4.6.0.0-idf-20190628-win32.zip to C:\esp\.espressif\tools\mconf\v4.6.0.0-idf-20190628
                          Installing ninja@1.9.0
                          file ninja-1.9.0-win64.zip is already downloaded
                          Extracting C:\esp\.espressif\dist\ninja-1.9.0-win64.zip to C:\esp\.espressif\tools\ninja\1.9.0
                          Installing idf-exe@1.0.1
                          file idf-exe-v1.0.1.zip is already downloaded
                          Extracting C:\esp\.espressif\dist\idf-exe-v1.0.1.zip to C:\esp\.espressif\tools\idf-exe\1.0.1
                          Installing ccache@3.7
                          file ccache-3.7-w64.zip is already downloaded
                          Extracting C:\esp\.espressif\dist\ccache-3.7-w64.zip to C:\esp\.espressif\tools\ccache\3.7
                          
2020-01-14 10:41:04.087   --------
2020-01-14 10:41:04.090   Installing Python environment:C:\Users\Sérgio Sobral\AppData\Local\Programs\Python\Python38\python.exe C:\esp\tools\idf_tools.py install-python-env
2020-01-14 10:41:17.069   Done, exit code=1
2020-01-14 10:41:17.069   --------
2020-01-14 10:41:17.069   Running command: C:\Users\Sérgio Sobral\AppData\Local\Programs\Python\Python38\python.exe C:\esp\tools\idf_tools.py install-python-env
                          Using base prefix 'C:\\Users\\Sérgio Sobral\\AppData\\Local\\Programs\\Python\\Python38'
                          New python executable in C:\esp\.espressif\python_env\idf3.3_py3.8_env\Scripts\python.exe
                          Command C:\esp\.espressif\py...v\Scripts\python.exe -m pip config list had error code 1
                          Installing setuptools, pip, wheel...
                          
                            Complete output from command C:\esp\.espressif\py...v\Scripts\python.exe - setuptools pip wheel:
                            Traceback (most recent call last):
                            File "<stdin>", line 3, in <module>
                          ModuleNotFoundError: No module named 'pkgutil'
                          ----------------------------------------
                          ...Installing setuptools, pip, wheel...done.
                          Traceback (most recent call last):
                            File "C:\Users\Sérgio Sobral\AppData\Local\Programs\Python\Python38\lib\runpy.py", line 193, in _run_module_as_main
                              return _run_code(code, main_globals, None,
                            File "C:\Users\Sérgio Sobral\AppData\Local\Programs\Python\Python38\lib\runpy.py", line 86, in _run_code
                              exec(code, run_globals)
                            File "C:\Users\Sérgio Sobral\AppData\Roaming\Python\Python38\site-packages\virtualenv.py", line 2634, in <module>
                              main()
[... cut ...]
2020-01-14 10:41:17.070   --------
2020-01-14 10:41:17.070   Message box (OK):
                          Installation has failed with exit code 1
2020-01-14 10:41:23.606   User chose OK.
2020-01-14 10:41:23.606   Message box (Yes/No):
                          Installation log has been created, it may contain more information about the problem.
                          Display the installation log now?
2020-01-14 10:41:24.690   User chose Yes.
2020-01-14 10:41:24.721   CurStepChanged raised an exception.`

igrr, 2020-01-14:
Thanks for trying that. Can you please check if virtualenv can be created without IDF involved? E.g. open cmd.exe, go to some directory like C:\tmp, and run `C:\Users\Sérgio Sobral\AppData\Local\Programs\Python\Python38\python.exe -m virtualenv myenv`.

If that doesn't work, it means that there is something wrong with the Python installation. You may try to select "Install Python 3.7" option in the IDF Tools installer. This option has been verified to work. (Uninstalling Python 3.8 is not necessary.)

lmagalhaes1337, 2020-01-14:
Running that command to create the virtual environment did fail. It was the same error of not finding the module "pkgutil". So I ran the idf-tools installer again and chose to install python 3.7 but after a while I got the same error. Here's the full log:

```
2020-01-14 11:03:20.789   Log opened. (Time zone: UTC+00:00)
2020-01-14 11:03:20.789   Setup version: Inno Setup version 6.0.2 (u)
2020-01-14 11:03:20.790   Original Setup EXE: C:\Users\Sérgio Sobral\Downloads\esp-idf-tools-setup-2.2.exe
2020-01-14 11:03:20.790   Setup command line: /SL5="$A07EE,408206375,721408,C:\Users\Sérgio Sobral\Downloads\esp-idf-tools-setup-2.2.exe" 
2020-01-14 11:03:20.790   Windows version: 10.0.18362  (NT platform: Yes)
2020-01-14 11:03:20.790   64-bit Windows: Yes
2020-01-14 11:03:20.790   Processor architecture: x64
2020-01-14 11:03:20.790   User privileges: None
2020-01-14 11:03:20.792   Administrative install mode: No
2020-01-14 11:03:20.792   Install mode root key: HKEY_CURRENT_USER
2020-01-14 11:03:20.792   64-bit install mode: Yes
2020-01-14 11:03:20.796   Created temporary directory: C:\Users\SRGIOS~1\AppData\Local\Temp\is-96MI5.tmp
2020-01-14 11:03:20.804   -- DLL function import --
2020-01-14 11:03:20.804   Function name: idpAddFile
2020-01-14 11:03:20.804   DLL name: files:idp.dll
2020-01-14 11:03:20.804   Extracting temporary file: C:\Users\SRGIOS~1\AppData\Local\Temp\is-96MI5.tmp\idp.dll
2020-01-14 11:03:20.824   Dest DLL name: C:\Users\SRGIOS~1\AppData\Local\Temp\is-96MI5.tmp\idp.dll
2020-01-14 11:03:20.824   Importing the DLL function.
2020-01-14 11:03:20.880   Successfully imported the DLL function. Delay loaded? No
2020-01-14 11:03:20.881   -- DLL function import --
2020-01-14 11:03:20.881   Function name: idpAddFileComp
2020-01-14 11:03:20.881   DLL name: files:idp.dll
2020-01-14 11:03:20.881   Dest DLL name: C:\Users\SRGIOS~1\AppData\Local\Temp\is-96MI5.tmp\idp.dll
2020-01-14 11:03:20.881   Importing the DLL function.
2020-01-14 11:03:20.881   Successfully imported the DLL function. Delay loaded? No
2020-01-14 11:03:20.881   -- DLL function import --
2020-01-14 11:03:20.881   Function name: idpAddMirror
2020-01-14 11:03:20.881   DLL name: files:idp.dll
2020-01-14 11:03:20.881   Dest DLL name: C:\Users\SRGIOS~1\AppData\Local\Temp\is-96MI5.tmp\idp.dll
2020-01-14 11:03:20.881   Importing the DLL function.
2020-01-14 11:03:20.881   Successfully imported the DLL function. Delay loaded? No
2020-01-14 11:03:20.881   -- DLL function import --
2020-01-14 11:03:20.881   Function name: idpAddFtpDir
2020-01-14 11:03:20.881   DLL name: files:idp.dll
2020-01-14 11:03:20.881   Dest DLL name: C:\Users\SRGIOS~1\AppData\Local\Temp\is-96MI5.tmp\idp.dll
2020-01-14 11:03:20.881   Importing the DLL function.
2020-01-14 11:03:20.881   Successfully imported the DLL function. Delay loaded? No
2020-01-14 11:03:20.881   -- DLL function import --
2020-01-14 11:03:20.881   Function name: idpAddFtpDirComp
2020-01-14 11:03:20.881   DLL name: files:idp.dll
2020-01-14 11:03:20.881   Dest DLL name: C:\Users\SRGIOS~1\AppData\Local\Temp\is-96MI5.tmp\idp.dll
2020-01-14 11:03:20.881   Importing the DLL function.
2020-01-14 11:03:20.881   Successfully imported the DLL function. Delay loaded? No
2020-01-14 11:03:20.881   -- DLL function import --
2020-01-14 11:03:20.881   Function name: idpClearFiles
2020-01-14 11:03:20.881   DLL name: files:idp.dll
2020-01-14 11:03:20.882   Dest DLL name: C:\Users\SRGIOS~1\AppData\Local\Temp\is-96MI5.tmp\idp.dll
2020-01-14 11:03:20.882   Importing the DLL function.
2020-01-14 11:03:20.882   Successfully imported the DLL function. Delay loaded? No
2020-01-14 11:03:20.882   -- DLL function import --
2020-01-14 11:03:20.882   Function name: idpFilesCount
2020-01-14 11:03:20.882   DLL name: files:idp.dll
2020-01-14 11:03:20.882   Dest DLL name: C:\Users\SRGIOS~1\AppData\Local\Temp\is-96MI5.tmp\idp.dll
2020-01-14 11:03:20.882   Importing the DLL function.
2020-01-14 11:03:20.882   Successfully imported the DLL function. Delay loaded? No
2020-01-14 11:03:20.882   -- DLL function import --
2020-01-14 11:03:20.882   Function name: idpFtpDirsCount
2020-01-14 11:03:20.882   DLL name: files:idp.dll
2020-01-14 11:03:20.882   Dest DLL name: C:\Users\SRGIOS~1\AppData\Local\Temp\is-96MI5.tmp\idp.dll
2020-01-14 11:03:20.882   Importing the DLL function.
2020-01-14 11:03:20.882   Successfully imported the DLL function. Delay loaded? No
2020-01-14 11:03:20.882   -- DLL function import --
2020-01-14 11:03:20.882   Function name: idpFileDownloaded
2020-01-14 11:03:20.882   DLL name: files:idp.dll
2020-01-14 11:03:20.882   Dest DLL name: C:\Users\SRGIOS~1\AppData\Local\Temp\is-96MI5.tmp\idp.dll
2020-01-14 11:03:20.882   Importing the DLL function.
2020-01-14 11:03:20.882   Successfully imported the DLL function. Delay loaded? No
[... 464 lines cut ...]
                            File "C:\Users\Sérgio Sobral\AppData\Roaming\Python\Python37\site-packages\virtualenv.py", line 963, in call_subprocess
                              raise OSError("Command {} failed with error code {}".format(cmd_desc, proc.returncode))
                          OSError: Command C:\esp\.espressif\py...v\Scripts\python.exe - setuptools pip wheel failed with error code 1
                          Creating a new Python environment in C:\esp\.espressif\python_env\idf3.3_py3.7_env
                          Traceback (most recent call last):
                            File "C:\esp\tools\idf_tools.py", line 1349, in <module>
[... cut ...]
2020-01-14 11:04:50.034   --------
2020-01-14 11:04:50.034   Message box (OK):
                          Installation has failed with exit code 1
2020-01-14 11:05:12.760   User chose OK.
2020-01-14 11:05:12.772   Message box (Yes/No):
                          Installation log has been created, it may contain more information about the problem.
                          Display the installation log now?
2020-01-14 11:05:14.066   User chose Yes.
2020-01-14 11:05:14.105   CurStepChanged raised an exception.
```

ymhsss, 2020-03-10:
Hello Dear supporter, I got the same issue with it.
C:\Win64app\Python37>python.exe -m virtualenv myenv
created virtual environment CPython3.7.3.final.0-64 in 4124ms
  creator CPython3Windows(dest=C:\Win64app\Python37\myenv, clear=False, global=False)
  seeder FromAppData(download=False, pip=latest, setuptools=latest, wheel=latest, via=copy, app_data_dir=C:\Users\Administrator\AppData\Local\Temp\tmpppp1kjd7\seed-app-data\v1)
  activators BashActivator,BatchActivator,FishActivator,PowerShellActivator,PythonActivator,XonshActivator
C:\Win64app\Python37>cd ..

C:\Win64app>cd esp-idf

C:\Win64app\esp-idf>install.bat
Installing ESP-IDF tools
Installing tools: xtensa-esp32-elf, esp32ulp-elf, cmake, openocd-esp32, mconf, ninja, idf-exe, ccache
Skipping xtensa-esp32-elf@esp-2019r2-8.2.0 (already installed)
Skipping esp32ulp-elf@2.28.51.20170517 (already installed)
Skipping cmake@3.13.4 (already installed)
Skipping openocd-esp32@v0.10.0-esp32-20190313 (already installed)
Skipping mconf@v4.6.0.0-idf-20190628 (already installed)
Skipping ninja@1.9.0 (already installed)
Skipping idf-exe@1.0.1 (already installed)
Skipping ccache@3.7 (already installed)
Setting up Python environment
Creating a new Python environment in C:\Win64app\.espressif\python_env\idf4.0_py3.7_env
usage: virtualenv [--version] [--with-traceback] [-v | -q] [--app-data APP_DATA] [--clear-app-data] [--discovery {builtin}] [-p py] [--creator {builtin,cpython3-win,venv}] [--seeder {app-data,pip}] [--no-seed] [--activators comma_sep_list]
                  [--clear] [--system-site-packages] [--symlinks | --copies] [--download | --no-download] [--extra-search-dir d [d ...]] [--pip version] [--setuptools version] [--wheel version] [--no-pip] [--no-setuptools] [--no-wheel]
                  [--symlink-app-data] [--prompt prompt] [-h]
                  dest
virtualenv: error: unrecognized arguments: --no-site-packages
Traceback (most recent call last):
  File "C:\Win64app\esp-idf\tools\idf_tools.py", line 1372, in <module>
    main(sys.argv[1:])
  File "C:\Win64app\esp-idf\tools\idf_tools.py", line 1368, in main
    action_func(args)
  File "C:\Win64app\esp-idf\tools\idf_tools.py", line 1184, in action_install_python_env
    stdout=sys.stdout, stderr=sys.stderr)
  File "C:\Win64app\Python37\lib\subprocess.py", line 347, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['C:\\Win64app\\Python37\\python.exe', '-m', 'virtualenv', '--no-site-packages', 'C:\\Win64app\\.espressif\\python_env\\idf4.0_py3.7_env']' returned non-zero exit status 2.

C:\Win64app\esp-idf>

igrr, 2020-03-10:
@ymhsss this has been fixed in commit 7ccf6eb on release/v4.0 branch. The fix will be part of the next bugfix release (v4.0.1).

In the meantime, IDF Tools installer v2.3 (https://dl.espressif.com/dl/esp-idf-tools-setup-2.3.exe) will work around this issue even when installing IDF v4.0, by using its own bundled version of idf_tools.py.

wnoliveira, 2020-03-18:
> @ymhsss this has been fixed in commit [7ccf6eb](https://github.com/espressif/esp-idf/commit/7ccf6ebadae432e24d933a5d4090b6f2373cf2d0) on release/v4.0 branch. The fix will be part of the next bugfix [...]

Unfortunately even IDF Tools installer v2.3 still generating the same error:

2020-03-18 09:51:02.048   Log opened. (Time zone: UTC-03:00)
2020-03-18 09:51:02.048   Setup version: Inno Setup version 6.0.2 (u)
2020-03-18 09:51:02.049   Original Setup EXE: C:\Users\wende\Downloads\esp-idf-tools-setup-2.3.exe
2020-03-18 09:51:02.049   Setup command line: /SL5="$280C9E,350580993,721408,C:\Users\wende\Downloads\esp-idf-tools-setup-2.3.exe" 
2020-03-18 09:51:02.049   Windows version: 10.0.18362  (NT platform: Yes)
2020-03-18 09:51:02.049   64-bit Windows: Yes
2020-03-18 09:51:02.049   Processor architecture: x64
2020-03-18 09:51:02.049   User privileges: None
2020-03-18 09:51:02.050   Administrative install mode: No
2020-03-18 09:51:02.050   Install mode root key: HKEY_CURRENT_USER
2020-03-18 09:51:02.053   64-bit install mode: Yes
2020-03-18 09:51:02.061   Created temporary directory: C:\Users\wende\AppData\Local\Temp\is-H7URN.tmp
2020-03-18 09:51:02.071   -- DLL function import --
2020-03-18 09:51:02.071   Function name: idpAddFile
2020-03-18 09:51:02.071   DLL name: files:idp.dll
2020-03-18 09:51:02.071   Extracting temporary file: C:\Users\wende\AppData\Local\Temp\is-H7URN.tmp\idp.dll
2020-03-18 09:51:02.083   Dest DLL name: C:\Users\wende\AppData\Local\Temp\is-H7URN.tmp\idp.dll
2020-03-18 09:51:02.083   Importing the DLL function.
2020-03-18 09:51:02.098   Successfully imported the DLL function. Delay loaded? No
2020-03-18 09:51:02.098   -- DLL function import --
2020-03-18 09:51:02.098   Function name: idpAddFileComp
2020-03-18 09:51:02.098   DLL name: files:idp.dll
2020-03-18 09:51:02.099   Dest DLL name: C:\Users\wende\AppData\Local\Temp\is-H7URN.tmp\idp.dll
2020-03-18 09:51:02.099   Importing the DLL function.
2020-03-18 09:51:02.099   Successfully imported the DLL function. Delay loaded? No
2020-03-18 09:51:02.099   -- DLL function import --
2020-03-18 09:51:02.099   Function name: idpAddMirror
2020-03-18 09:51:02.099   DLL name: files:idp.dll
2020-03-18 09:51:02.099   Dest DLL name: C:\Users\wende\AppData\Local\Temp\is-H7URN.tmp\idp.dll
2020-03-18 09:51:02.099   Importing the DLL function.
2020-03-18 09:51:02.099   Successfully imported the DLL function. Delay loaded? No
2020-03-18 09:51:02.099   -- DLL function import --
2020-03-18 09:51:02.099   Function name: idpAddFtpDir
2020-03-18 09:51:02.099   DLL name: files:idp.dll
2020-03-18 09:51:02.099   Dest DLL name: C:\Users\wende\AppData\Local\Temp\is-H7URN.tmp\idp.dll
2020-03-18 09:51:02.099   Importing the DLL function.
2020-03-18 09:51:02.099   Successfully imported the DLL function. Delay loaded? No
2020-03-18 09:51:02.099   -- DLL function import --
2020-03-18 09:51:02.099   Function name: idpAddFtpDirComp
2020-03-18 09:51:02.099   DLL name: files:idp.dll
2020-03-18 09:51:02.099   Dest DLL name: C:\Users\wende\AppData\Local\Temp\is-H7URN.tmp\idp.dll
2020-03-18 09:51:02.099   Importing the DLL function.
2020-03-18 09:51:02.099   Successfully imported the DLL function. Delay loaded? No
2020-03-18 09:51:02.099   -- DLL function import --
2020-03-18 09:51:02.099   Function name: idpClearFiles
2020-03-18 09:51:02.099   DLL name: files:idp.dll
2020-03-18 09:51:02.099   Dest DLL name: C:\Users\wende\AppData\Local\Temp\is-H7URN.tmp\idp.dll
2020-03-18 09:51:02.100   Importing the DLL function.
2020-03-18 09:51:02.100   Successfully imported the DLL function. Delay loaded? No
2020-03-18 09:51:02.100   -- DLL function import --
2020-03-18 09:51:02.100   Function name: idpFilesCount
2020-03-18 09:51:02.100   DLL name: files:idp.dll
2020-03-18 09:51:02.100   Dest DLL name: C:\Users\wende\AppData\Local\Temp\is-H7URN.tmp\idp.dll
2020-03-18 09:51:02.100   Importing the DLL function.
2020-03-18 09:51:02.100   Successfully imported the DLL function. Delay loaded? No
2020-03-18 09:51:02.100   -- DLL function import --
2020-03-18 09:51:02.100   Function name: idpFtpDirsCount
2020-03-18 09:51:02.100   DLL name: files:idp.dll
2020-03-18 09:51:02.100   Dest DLL name: C:\Users\wende\AppData\Local\Temp\is-H7URN.tmp\idp.dll
2020-03-18 09:51:02.100   Importing the DLL function.
2020-03-18 09:51:02.100   Successfully imported the DLL function. Delay loaded? No
2020-03-18 09:51:02.100   -- DLL function import --
2020-03-18 09:51:02.100   Function name: idpFileDownloaded
2020-03-18 09:51:02.100   DLL name: files:idp.dll
2020-03-18 09:51:02.100   Dest DLL name: C:\Users\wende\AppData\Local\Temp\is-H7URN.tmp\idp.dll
2020-03-18 09:51:02.100   Importing the DLL function.
[... 358 lines cut ...]
                          usage: virtualenv [--version] [--with-traceback] [-v | -q] [--app-data APP_DATA] [--clear-app-data] [--discovery {builtin}] [-p py] [--creator {builtin,cpython3-win,venv}] [--seeder {app-data,pip}] [--no-seed] [--activators comma_sep_list]
                                            [--clear] [--system-site-packages] [--copies] [--download | --no-download] [--extra-search-dir d [d ...]] [--pip version] [--setuptools version] [--wheel version] [--no-pip] [--no-setuptools] [--no-wheel]
                                            [--symlink-app-data] [--prompt prompt] [-h]
                                            dest
                          virtualenv: error: unrecognized arguments: --no-site-packages
[... cut ...]
2020-03-18 09:53:34.211   --------
2020-03-18 09:53:34.211   Message box (OK):
                          Installation has failed with exit code 1
2020-03-18 09:53:37.271   User chose OK.
2020-03-18 09:53:37.280   Message box (Yes/No):
                          Installation log has been created, it may contain more information about the problem.
                          Display the installation log now?
2020-03-18 09:53:38.559   User chose Yes.
2020-03-18 09:53:38.609   CurStepChanged raised an exception.

igrr, 2020-03-18:
@wnoliveira I think the copy of IDF which you are using doesn't include the fix, yet. There are 2 options:

1) go to C:\Projetos\ESP32\esp-idf\v4.0-release and run `git fetch origin` and `git checkout release/v4.0` to update to latest release/v4.0 branch. Then run the installer again.

2) Let the installer download a new copy of IDF. In this case the installer will detect if the version doesn't include this fix, and will work around the issue.

wnoliveira, 2020-03-18:
> @wnoliveira I think the copy of IDF which you are using doesn't include the fix, yet. There are 2 options:
>
> 1. go to C:\Projetos\ESP32\esp-idf\v4.0-release and run `git fetch origin` and `git [...]

Hi @igrr  ! I've tryed to let the installer download a new copy of IDF. Ant the LOG of erro (the same) is:

2020-03-18 12:37:29.783   Log opened. (Time zone: UTC-03:00)
2020-03-18 12:37:29.784   Setup version: Inno Setup version 6.0.2 (u)
2020-03-18 12:37:29.784   Original Setup EXE: C:\Users\wende\Downloads\esp-idf-tools-setup-2.3.exe
2020-03-18 12:37:29.784   Setup command line: /SL5="$2F0E26,350580993,721408,C:\Users\wende\Downloads\esp-idf-tools-setup-2.3.exe" 
2020-03-18 12:37:29.784   Windows version: 10.0.18362  (NT platform: Yes)
2020-03-18 12:37:29.784   64-bit Windows: Yes
2020-03-18 12:37:29.784   Processor architecture: x64
2020-03-18 12:37:29.784   User privileges: None
2020-03-18 12:37:29.786   Administrative install mode: No
2020-03-18 12:37:29.786   Install mode root key: HKEY_CURRENT_USER
2020-03-18 12:37:29.786   64-bit install mode: Yes
2020-03-18 12:37:29.796   Created temporary directory: C:\Users\wende\AppData\Local\Temp\is-9H7UK.tmp
2020-03-18 12:37:29.808   -- DLL function import --
2020-03-18 12:37:29.808   Function name: idpAddFile
2020-03-18 12:37:29.808   DLL name: files:idp.dll
2020-03-18 12:37:29.808   Extracting temporary file: C:\Users\wende\AppData\Local\Temp\is-9H7UK.tmp\idp.dll
2020-03-18 12:37:29.823   Dest DLL name: C:\Users\wende\AppData\Local\Temp\is-9H7UK.tmp\idp.dll
2020-03-18 12:37:29.823   Importing the DLL function.
2020-03-18 12:37:29.839   Successfully imported the DLL function. Delay loaded? No
2020-03-18 12:37:29.839   -- DLL function import --
2020-03-18 12:37:29.839   Function name: idpAddFileComp
2020-03-18 12:37:29.839   DLL name: files:idp.dll
2020-03-18 12:37:29.839   Dest DLL name: C:\Users\wende\AppData\Local\Temp\is-9H7UK.tmp\idp.dll
2020-03-18 12:37:29.839   Importing the DLL function.
2020-03-18 12:37:29.839   Successfully imported the DLL function. Delay loaded? No
2020-03-18 12:37:29.839   -- DLL function import --
2020-03-18 12:37:29.839   Function name: idpAddMirror
2020-03-18 12:37:29.839   DLL name: files:idp.dll
2020-03-18 12:37:29.839   Dest DLL name: C:\Users\wende\AppData\Local\Temp\is-9H7UK.tmp\idp.dll
2020-03-18 12:37:29.839   Importing the DLL function.
2020-03-18 12:37:29.839   Successfully imported the DLL function. Delay loaded? No
2020-03-18 12:37:29.839   -- DLL function import --
2020-03-18 12:37:29.839   Function name: idpAddFtpDir
2020-03-18 12:37:29.839   DLL name: files:idp.dll
2020-03-18 12:37:29.839   Dest DLL name: C:\Users\wende\AppData\Local\Temp\is-9H7UK.tmp\idp.dll
2020-03-18 12:37:29.839   Importing the DLL function.
2020-03-18 12:37:29.839   Successfully imported the DLL function. Delay loaded? No
2020-03-18 12:37:29.839   -- DLL function import --
2020-03-18 12:37:29.840   Function name: idpAddFtpDirComp
2020-03-18 12:37:29.840   DLL name: files:idp.dll
2020-03-18 12:37:29.840   Dest DLL name: C:\Users\wende\AppData\Local\Temp\is-9H7UK.tmp\idp.dll
2020-03-18 12:37:29.840   Importing the DLL function.
2020-03-18 12:37:29.840   Successfully imported the DLL function. Delay loaded? No
2020-03-18 12:37:29.840   -- DLL function import --
2020-03-18 12:37:29.840   Function name: idpClearFiles
2020-03-18 12:37:29.840   DLL name: files:idp.dll
2020-03-18 12:37:29.840   Dest DLL name: C:\Users\wende\AppData\Local\Temp\is-9H7UK.tmp\idp.dll
2020-03-18 12:37:29.840   Importing the DLL function.
2020-03-18 12:37:29.840   Successfully imported the DLL function. Delay loaded? No
2020-03-18 12:37:29.840   -- DLL function import --
2020-03-18 12:37:29.840   Function name: idpFilesCount
2020-03-18 12:37:29.840   DLL name: files:idp.dll
2020-03-18 12:37:29.840   Dest DLL name: C:\Users\wende\AppData\Local\Temp\is-9H7UK.tmp\idp.dll
2020-03-18 12:37:29.840   Importing the DLL function.
2020-03-18 12:37:29.840   Successfully imported the DLL function. Delay loaded? No
2020-03-18 12:37:29.840   -- DLL function import --
2020-03-18 12:37:29.840   Function name: idpFtpDirsCount
2020-03-18 12:37:29.840   DLL name: files:idp.dll
2020-03-18 12:37:29.840   Dest DLL name: C:\Users\wende\AppData\Local\Temp\is-9H7UK.tmp\idp.dll
2020-03-18 12:37:29.840   Importing the DLL function.
2020-03-18 12:37:29.840   Successfully imported the DLL function. Delay loaded? No
2020-03-18 12:37:29.841   -- DLL function import --
2020-03-18 12:37:29.841   Function name: idpFileDownloaded
2020-03-18 12:37:29.841   DLL name: files:idp.dll
[... 486 lines cut ...]
                            Created wheel for future: filename=future-0.18.2-py3-none-any.whl size=491062 sha256=10757a0c6f6714cc1b03e7ec3e9b3944d4366e7d69b81f6535a68971bb89576f
                            Stored in directory: c:\users\wende\appdata\local\pip\cache\wheels\56\b0\fe\4410d17b32f1f0c3cf54cdfb2bc04d7b4b8f4ae377e2229ba0
                          Successfully built future
                          Installing collected packages: click, pyserial, future, six, pycparser, cffi, cryptography, pyparsing, pyelftools
                          Successfully installed cffi-1.14.0 click-7.1.1 cryptography-2.8 future-0.18.2 pycparser-2.20 pyelftools-0.26 pyparsing-2.3.1 pyserial-3.4 six-1.14.0
                          
2020-03-18 12:50:03.346   --------
[... cut ...]
2020-03-18 12:52:14.124   --------
2020-03-18 12:52:14.124   Message box (OK):
                          Installation has failed with exit code 1
2020-03-18 12:56:57.258   User chose OK.
2020-03-18 12:56:57.270   Message box (Yes/No):
                          Installation log has been created, it may contain more information about the problem.
                          Display the installation log now?
2020-03-18 12:56:58.582   User chose Yes.
2020-03-18 12:56:58.658   CurStepChanged raised an exception.

Any idea?

lmagalhaes1337, 2020-03-18:
Hi there,

I formatted my PC and now can't get the install working again ... this time, there's no special characters on the name and the error points to a virtual env error ... I did try your previous recommendation of trying the virtualenv command by itself and see if it failed and it doesn't.

Here's the full log:

`2020-03-18 16:03:00.598   Log opened. (Time zone: UTC+00:00)
2020-03-18 16:03:00.598   Setup version: Inno Setup version 6.0.2 (u)
2020-03-18 16:03:00.598   Original Setup EXE: C:\Users\Luis\Downloads\esp-idf-tools-setup-2.3.exe
2020-03-18 16:03:00.598   Setup command line: /SL5="$1B0B24,350580993,721408,C:\Users\Luis\Downloads\esp-idf-tools-setup-2.3.exe" 
2020-03-18 16:03:00.598   Windows version: 10.0.18363  (NT platform: Yes)
2020-03-18 16:03:00.598   64-bit Windows: Yes
2020-03-18 16:03:00.598   Processor architecture: x64
2020-03-18 16:03:00.598   User privileges: None
2020-03-18 16:03:00.600   Administrative install mode: No
2020-03-18 16:03:00.600   Install mode root key: HKEY_CURRENT_USER
2020-03-18 16:03:00.600   64-bit install mode: Yes
2020-03-18 16:03:00.601   Created temporary directory: C:\Users\Luis\AppData\Local\Temp\is-FPL7I.tmp
2020-03-18 16:03:00.604   -- DLL function import --
2020-03-18 16:03:00.604   Function name: idpAddFile
2020-03-18 16:03:00.604   DLL name: files:idp.dll
2020-03-18 16:03:00.604   Extracting temporary file: C:\Users\Luis\AppData\Local\Temp\is-FPL7I.tmp\idp.dll
2020-03-18 16:03:00.611   Dest DLL name: C:\Users\Luis\AppData\Local\Temp\is-FPL7I.tmp\idp.dll
2020-03-18 16:03:00.611   Importing the DLL function.
2020-03-18 16:03:00.640   Successfully imported the DLL function. Delay loaded? No
2020-03-18 16:03:00.640   -- DLL function import --
2020-03-18 16:03:00.640   Function name: idpAddFileComp
2020-03-18 16:03:00.640   DLL name: files:idp.dll
2020-03-18 16:03:00.640   Dest DLL name: C:\Users\Luis\AppData\Local\Temp\is-FPL7I.tmp\idp.dll
2020-03-18 16:03:00.640   Importing the DLL function.
2020-03-18 16:03:00.640   Successfully imported the DLL function. Delay loaded? No
2020-03-18 16:03:00.640   -- DLL function import --
2020-03-18 16:03:00.640   Function name: idpAddMirror
2020-03-18 16:03:00.640   DLL name: files:idp.dll
2020-03-18 16:03:00.640   Dest DLL name: C:\Users\Luis\AppData\Local\Temp\is-FPL7I.tmp\idp.dll
2020-03-18 16:03:00.640   Importing the DLL function.
2020-03-18 16:03:00.640   Successfully imported the DLL function. Delay loaded? No
2020-03-18 16:03:00.640   -- DLL function import --
2020-03-18 16:03:00.640   Function name: idpAddFtpDir
2020-03-18 16:03:00.640   DLL name: files:idp.dll
2020-03-18 16:03:00.640   Dest DLL name: C:\Users\Luis\AppData\Local\Temp\is-FPL7I.tmp\idp.dll
2020-03-18 16:03:00.640   Importing the DLL function.
2020-03-18 16:03:00.640   Successfully imported the DLL function. Delay loaded? No
2020-03-18 16:03:00.640   -- DLL function import --
2020-03-18 16:03:00.641   Function name: idpAddFtpDirComp
2020-03-18 16:03:00.641   DLL name: files:idp.dll
2020-03-18 16:03:00.641   Dest DLL name: C:\Users\Luis\AppData\Local\Temp\is-FPL7I.tmp\idp.dll
2020-03-18 16:03:00.641   Importing the DLL function.
2020-03-18 16:03:00.641   Successfully imported the DLL function. Delay loaded? No
2020-03-18 16:03:00.641   -- DLL function import --
2020-03-18 16:03:00.641   Function name: idpClearFiles
2020-03-18 16:03:00.641   DLL name: files:idp.dll
2020-03-18 16:03:00.641   Dest DLL name: C:\Users\Luis\AppData\Local\Temp\is-FPL7I.tmp\idp.dll
2020-03-18 16:03:00.641   Importing the DLL function.
2020-03-18 16:03:00.641   Successfully imported the DLL function. Delay loaded? No
2020-03-18 16:03:00.641   -- DLL function import --
2020-03-18 16:03:00.641   Function name: idpFilesCount
2020-03-18 16:03:00.641   DLL name: files:idp.dll
2020-03-18 16:03:00.641   Dest DLL name: C:\Users\Luis\AppData\Local\Temp\is-FPL7I.tmp\idp.dll
2020-03-18 16:03:00.641   Importing the DLL function.
2020-03-18 16:03:00.641   Successfully imported the DLL function. Delay loaded? No
2020-03-18 16:03:00.641   -- DLL function import --
2020-03-18 16:03:00.641   Function name: idpFtpDirsCount
2020-03-18 16:03:00.641   DLL name: files:idp.dll
2020-03-18 16:03:00.641   Dest DLL name: C:\Users\Luis\AppData\Local\Temp\is-FPL7I.tmp\idp.dll
2020-03-18 16:03:00.641   Importing the DLL function.
2020-03-18 16:03:00.641   Successfully imported the DLL function. Delay loaded? No
2020-03-18 16:03:00.641   -- DLL function import --
2020-03-18 16:03:00.641   Function name: idpFileDownloaded
2020-03-18 16:03:00.641   DLL name: files:idp.dll
[... 414 lines cut ...]
                          usage: virtualenv [--version] [--with-traceback] [-v | -q] [--app-data APP_DATA] [--clear-app-data] [--discovery {builtin}] [-p py] [--creator {builtin,cpython3-win,venv}] [--seeder {app-data,pip}] [--no-seed] [--activators comma_sep_list]
                                            [--clear] [--system-site-packages] [--copies] [--download | --no-download] [--extra-search-dir d [d ...]] [--pip version] [--setuptools version] [--wheel version] [--no-pip] [--no-setuptools] [--no-wheel]
                                            [--symlink-app-data] [--prompt prompt] [-h]
                                            dest
                          virtualenv: error: unrecognized arguments: --no-site-packages
                          Traceback (most recent call last):
                            File "C:\Users\Luis\Desktop\esp-idf\tools\idf_tools.py", line 1372, in <module>
[... cut ...]
2020-03-18 16:03:42.897   --------
2020-03-18 16:03:42.897   Message box (OK):
                          Installation has failed with exit code 1
2020-03-18 16:03:47.427   User chose OK.
2020-03-18 16:03:47.433   Message box (Yes/No):
                          Installation log has been created, it may contain more information about the problem.
                          Display the installation log now?
2020-03-18 16:03:48.348   User chose Yes.
2020-03-18 16:03:48.365   CurStepChanged raised an exception.`

igrr, 2020-03-19:
@wnoliveira It seems that this time the installation failed when installing Windows Defender exceptions. Can you try running the installer again, and this time un-check the checkbox for the Windows Defender?

igrr, 2020-03-19:
@lmagalhaes1337 Same as suggested to @wnoliveira above (https://github.com/espressif/esp-idf/issues/4602#issuecomment-600621002), there are 2 options:

1. Go to C:\Users\Luis\Desktop\esp-idf and run git fetch origin and git checkout release/v4.0 to update to latest release/v4.0 branch. Then run the installer again.

2. Let the installer download a new copy of IDF. In this case the installer will detect if the version doesn't include this fix, and will work around the issue.

wnoliveira, 2020-03-19:
> @wnoliveira It seems that this time the installation failed when installing Windows Defender exceptions. Can you try running the installer again, and this time un-check the checkbox for the Windows Defender?

Hi @igrr . Yes, and still generating error:

2020-03-19 14:43:30.893   Log opened. (Time zone: UTC-03:00)
2020-03-19 14:43:30.893   Setup version: Inno Setup version 6.0.2 (u)
2020-03-19 14:43:30.893   Original Setup EXE: C:\Users\wende\Downloads\esp-idf-tools-setup-2.3.exe
2020-03-19 14:43:30.894   Setup command line: /SL5="$50DC6,350580993,721408,C:\Users\wende\Downloads\esp-idf-tools-setup-2.3.exe" 
2020-03-19 14:43:30.894   Windows version: 10.0.18363  (NT platform: Yes)
2020-03-19 14:43:30.894   64-bit Windows: Yes
2020-03-19 14:43:30.894   Processor architecture: x64
2020-03-19 14:43:30.894   User privileges: None
2020-03-19 14:43:30.895   Administrative install mode: No
2020-03-19 14:43:30.895   Install mode root key: HKEY_CURRENT_USER
2020-03-19 14:43:30.896   64-bit install mode: Yes
2020-03-19 14:43:30.904   Created temporary directory: C:\Users\wende\AppData\Local\Temp\is-AHII5.tmp
2020-03-19 14:43:30.917   -- DLL function import --
2020-03-19 14:43:30.917   Function name: idpAddFile
2020-03-19 14:43:30.917   DLL name: files:idp.dll
2020-03-19 14:43:30.917   Extracting temporary file: C:\Users\wende\AppData\Local\Temp\is-AHII5.tmp\idp.dll
2020-03-19 14:43:30.931   Dest DLL name: C:\Users\wende\AppData\Local\Temp\is-AHII5.tmp\idp.dll
2020-03-19 14:43:30.931   Importing the DLL function.
2020-03-19 14:43:30.971   Successfully imported the DLL function. Delay loaded? No
2020-03-19 14:43:30.971   -- DLL function import --
2020-03-19 14:43:30.971   Function name: idpAddFileComp
2020-03-19 14:43:30.971   DLL name: files:idp.dll
2020-03-19 14:43:30.971   Dest DLL name: C:\Users\wende\AppData\Local\Temp\is-AHII5.tmp\idp.dll
2020-03-19 14:43:30.971   Importing the DLL function.
2020-03-19 14:43:30.971   Successfully imported the DLL function. Delay loaded? No
2020-03-19 14:43:30.971   -- DLL function import --
2020-03-19 14:43:30.971   Function name: idpAddMirror
2020-03-19 14:43:30.971   DLL name: files:idp.dll
2020-03-19 14:43:30.971   Dest DLL name: C:\Users\wende\AppData\Local\Temp\is-AHII5.tmp\idp.dll
2020-03-19 14:43:30.971   Importing the DLL function.
2020-03-19 14:43:30.971   Successfully imported the DLL function. Delay loaded? No
2020-03-19 14:43:30.971   -- DLL function import --
2020-03-19 14:43:30.971   Function name: idpAddFtpDir
2020-03-19 14:43:30.971   DLL name: files:idp.dll
2020-03-19 14:43:30.972   Dest DLL name: C:\Users\wende\AppData\Local\Temp\is-AHII5.tmp\idp.dll
2020-03-19 14:43:30.972   Importing the DLL function.
2020-03-19 14:43:30.972   Successfully imported the DLL function. Delay loaded? No
2020-03-19 14:43:30.972   -- DLL function import --
2020-03-19 14:43:30.972   Function name: idpAddFtpDirComp
2020-03-19 14:43:30.972   DLL name: files:idp.dll
2020-03-19 14:43:30.972   Dest DLL name: C:\Users\wende\AppData\Local\Temp\is-AHII5.tmp\idp.dll
2020-03-19 14:43:30.972   Importing the DLL function.
2020-03-19 14:43:30.972   Successfully imported the DLL function. Delay loaded? No
2020-03-19 14:43:30.972   -- DLL function import --
2020-03-19 14:43:30.972   Function name: idpClearFiles
2020-03-19 14:43:30.972   DLL name: files:idp.dll
2020-03-19 14:43:30.972   Dest DLL name: C:\Users\wende\AppData\Local\Temp\is-AHII5.tmp\idp.dll
2020-03-19 14:43:30.972   Importing the DLL function.
2020-03-19 14:43:30.972   Successfully imported the DLL function. Delay loaded? No
2020-03-19 14:43:30.972   -- DLL function import --
2020-03-19 14:43:30.972   Function name: idpFilesCount
2020-03-19 14:43:30.972   DLL name: files:idp.dll
2020-03-19 14:43:30.972   Dest DLL name: C:\Users\wende\AppData\Local\Temp\is-AHII5.tmp\idp.dll
2020-03-19 14:43:30.972   Importing the DLL function.
2020-03-19 14:43:30.972   Successfully imported the DLL function. Delay loaded? No
2020-03-19 14:43:30.972   -- DLL function import --
2020-03-19 14:43:30.972   Function name: idpFtpDirsCount
2020-03-19 14:43:30.972   DLL name: files:idp.dll
2020-03-19 14:43:30.973   Dest DLL name: C:\Users\wende\AppData\Local\Temp\is-AHII5.tmp\idp.dll
2020-03-19 14:43:30.973   Importing the DLL function.
2020-03-19 14:43:30.973   Successfully imported the DLL function. Delay loaded? No
2020-03-19 14:43:30.973   -- DLL function import --
2020-03-19 14:43:30.973   Function name: idpFileDownloaded
2020-03-19 14:43:30.973   DLL name: files:idp.dll
2020-03-19 14:43:30.973   Dest DLL name: C:\Users\wende\AppData\Local\Temp\is-AHII5.tmp\idp.dll
2020-03-19 14:43:30.973   Importing the DLL function.
[... 359 lines cut ...]
                          usage: virtualenv [--version] [--with-traceback] [-v | -q] [--app-data APP_DATA] [--clear-app-data] [--discovery {builtin}] [-p py] [--creator {builtin,cpython3-win,venv}] [--seeder {app-data,pip}] [--no-seed] [--activators comma_sep_list]
                                            [--clear] [--system-site-packages] [--copies] [--download | --no-download] [--extra-search-dir d [d ...]] [--pip version] [--setuptools version] [--wheel version] [--no-pip] [--no-setuptools] [--no-wheel]
                                            [--symlink-app-data] [--prompt prompt] [-h]
                                            dest
                          virtualenv: error: unrecognized arguments: --no-site-packages
                          Traceback (most recent call last):
[... cut ...]
2020-03-19 14:45:14.154   --------
2020-03-19 14:45:14.154   Message box (OK):
                          Installation has failed with exit code 1
2020-03-19 14:45:17.569   User chose OK.
2020-03-19 14:45:17.581   Message box (Yes/No):
                          Installation log has been created, it may contain more information about the problem.
                          Display the installation log now?
2020-03-19 14:45:18.388   User chose Yes.
2020-03-19 14:45:18.479   CurStepChanged raised an exception.

igrr, 2020-03-19:
@wnoliveira Did you end up updating IDF to the latest release/v4.0 branch? If you haven't done this, the only way to install v4.0 tag is to let the installer download IDF from scratch. If you are using an existing copy of IDF, the installer is not able to work around the idf_tools.py bug present in v4.0 release.

wnoliveira, 2020-03-19:
> @wnoliveira Did you end up updating IDF to the latest release/v4.0 branch? If you haven't done this, the only way to install v4.0 tag is to let the installer download IDF from scratch. If you are using an existing copy of IDF, the installer is not able to work around the idf_tools.py bug present in v4.0 release.
> 
> Alternatively, you can choose to download "release/v4.0" version in the installer instead of "v4.0".

Hi. Yes, I have deleted my copy of idf and downloaded a new one trhough the idf tools installer. I did it at the first time that I run the idf-toolsv2.3. This last time I've used this copy and just unchecked the windows defender option.

igrr, 2020-03-19:
> This last time I've used this copy and just unchecked the windows defender option.

Okay, if you want to keep using this copy, you'll need to pull the latest release/v4.0 branch, which includes a fix to idf_tools.py.

wnoliveira, 2020-03-21:
This works by updating to the latest realease v4.0!

igrr, 2020-03-23:
@lmagalhaes1337 Have you been able to resolve the issue?

lmagalhaes1337, 2020-03-23:
Hi @igrr ,

Yes. I had to do your fix: use git to checkout the release4.0 branch (even though that was the one selected when installing the ESP-IDF) and then run the 2.3 installer. The 2.2 still didn't work.

igrr, 2020-03-23:
Thanks for confirming. We have fixed the link in the documentation (v4.0 release branch): https://docs.espressif.com/projects/esp-idf/en/release-v4.0/get-started/windows-setup.html#esp-idf-tools-installer

closed 2020-03-23 by igrr