espressif/esp-idf #4603: how to include dht22 library (import third party library) error in CMakeLists.txt file

ravishree19, 2020-01-08:
I have to interface dht22 sensor in ESP32 using esp-idf.I refered https://github.com/UncleRus/esp-idf-lib for dht22 and i was unable to include the dht lib.Can you please help me with the CmakeList.txt file and the Make file to include path.The issue after following the procedure in above link is that esp32 is restarting.I didnt know how CmakeList file works.

renzbagaporo, 2020-01-09:
Hi @ravishree19, so you are able to build a project using the library you mentioned successfully?

ravishree19, 2020-01-09:
> Hi @ravishree19, so you are able to build a project using the library you mentioned successfully?

yes bt after flashing esp32 is restarting below are the screenshots.
[image]
[image]
[image]

renzbagaporo, 2020-01-09:
@ravishree19 Since the build succeeds, I would guess that the library you were trying to use got 'imported' correctly. The error at runtime is a totally different issue, and from what I can see in the logs, is with the third-party library.

Unfortunately, you would have to file this issue with the author of the library you were trying to use.

Alvin1Zhang, 2020-01-09:
@renzbagaporo Thanks for the support. @ravishree19 Thanks for reporting, for "how" question or "third party issues", suggest you could create a post on our forum at https://esp32.com or file the issue with the author of the library you were trying to use. Will close the ticket for now. Thanks.

closed 2020-01-09 by Alvin1Zhang