espressif/esp-idf #4604: Instrument Relay feature of BLE Mesh Nodes

linkineo, 2020-01-08:
## Goal
**Document how to see in the console of an ESP32 device that it is actually relaying BLE Mesh messages from within the same provisioned network**

## Steps
Using the SDK 4.0 beta 2 and three ESP-WROOM-32 devices
Device 1: Using the esp-idf ble_mesh_node example
Device 2: Using the esp-idf ble_mesh_client_model example
Device 3: Using the esp-idf ble_mesh_node example again, with Relay feature enabled in menuconfig and .relay set to enabled in the config server (line 37 of example main)
```
static esp_ble_mesh_cfg_srv_t config_server = {
    .relay = ESP_BLE_MESH_RELAY_ENABLED,
```

Steps:
1) Provision Devices 1,2,3 using nrfMesh app as detailed in the tutorial so that 1 and 2 share both network key and app keys where needed, and that 3 has the same network key.
2) Confirm I can actually turn the LED ON/OFF through gatt bearer->ble mesh as detailed in the tutorial (also setting the proper destination using UART).
3) Put just enough distance between devices 1 and 2 so they cannot communicate anymore (processing iteratively)
4) Add device 3 in the middle as configured. 

## Results
Communication is still not happening between 1 and 2. 
I tried hooking some debug info on 3 but see nothing happening. 

## Request
**Can you confirm relay functionality has been tested and is working ?
Could you explain/document how I can instrument a provisioned BLE Mesg node to see whether it gets relays messages ?**

Alvin1Zhang, 2020-01-09:
@linkineo Thanks for reporting.

WCCWCC, 2020-01-09:
Hi @linkineo ，
1. we confirm relay functionality has been tested.
2. The conditions of the packet relay are: 
      (1)The destination address of the packet is not its own address.
      (2)The TTL value of the packet is greater than 1.
3. The packet relay is implemented by calling bt_mesh_net_relay in the bt_mesh_net_recv function. 
4. You can increase the print level of bt_mesh_net_relay (BT_DBG-> BT_WARN), and then you can see if the packet is relayed.

linkineo, 2020-01-15:
Thanks a lot. Works exactly as you mentioned above. Increasing the log level as you told me help me find the issue in my code.

closed 2020-01-15 by linkineo

linkineo, 2020-01-20:
hello @mayowu0101,

The individual node addresses are first set when you provision both ble_mesh_node and ble_mesh_client using the nrf mobile app (see at the end of the [tutorial](https://github.com/espressif/esp-idf/blob/ble_mesh_release/esp-ble-mesh-v0.6.1/examples/bluetooth/ble_mesh/ble_mesh_client_model/tutorial/ble_mesh_client_model.md)). 
NB: Provisioning somehow works only on the Android version of the app for me (Huawei P30 / Android 9) and not on iOS (iPhone 11 Pro / ios13).

Only then can the UART be used to configure the destination address of the BLE Mesh Message. It's the UART of the ble_mesh_client_model on which you write the address of the ble_mesh_node to which you will be sending the packets (see [readme](https://github.com/espressif/esp-idf/blob/ble_mesh_release/esp-ble-mesh-v0.6.1/examples/bluetooth/ble_mesh/ble_mesh_node/tutorial/Ble_Mesh_Node_Example_Walkthrough.md))

The picture in the tutorial sums it all up:
[image]

mayowu0101, 2020-02-11:
hello @linkineo ,

I apologize for not replying you earlier. 
Thank your replying. I can open the subscribe and publish channel through the setting of the phone.

I have test relay node setting. 
The setting relay function ( `.relay = ESP_BLE_MESH_RELAY_ENABLED` ) on server node, and my state is:

> Device 1(cliant node) ,Device 2(server relay enabled) ,Device 3(server node).
> There is not see between "Device 1" and "Device 3",and in the middle is "Device 2". 

But when "Device 2" and "Device 3" is open and my phone stand on the "Device 1" place can see "Device 2" ,but can't see the "Device 3".

The "Device 2" relay node will enabled if my phone stand on the "Device 1" can see "Device 3"? 
or 
If "Device 3" so farway, "Device 2" relay node function have effected? (my state is "Device 2"  can see "Device 3".)

Thank and Best regards,
Mayo

heidecrouss, 2020-08-19:
Could you explain/document how I can how can I configure my node exactly to relay messages? I use a vendor model

xuanchibkd14, 2022-08-15:
> 

hi, I can't check relay node state in nrf mesh app, always is disabled

Shane555, 2022-10-11:
@xuanchibkd14  have you 

1. checked that you enabled relay support on sdkconfig? you can do so via menuconfig command
2. refreshed the node's configuration page on the mesh app by "pulling down the screen" ?