espressif/esp-idf #4605: Ridiclulus Exception when i declare new variable (Panic)

drony, 2020-01-08:
ESP IDF v3.3 , v3.3.1 and v4.0beta2 and toolchain 5.2, 8.2 r1 and r2 tested..... same result
just declare **int bc_counter;**
and increment in thread **bc_counter++;**
then panic handler!!! comment out **bc_counter++**. everythings fine.

Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x40159d37  PS      : 0x00060c30  A0      : 0x80159d78  A1      : 0x3ffb5580
0x40159d37: corereactor::drivers::corem1::CoreM1Driver::uart_event_task() at c:\msys32\opt\xtens
a-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\bits/shared_ptr_base.h:127 (discriminator 1)

A2      : 0x00000000  A3      : 0x00000000  A4      : 0x3ffb57ac  A5      : 0x3ffce740
A6      : 0x3ffce71c  A7      : 0x3ffcc6b4  A8      : 0x80159d31  A9      : 0x3ffb5540
A10     : 0x00000001  A11     : 0x00000000  A12     : 0x3ffb27e4  A13     : 0x00000000
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001c
EXCVADDR: 0x0000000c  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffff1

ELF file SHA256: 0cd3fbd12f094c3af0c1bfebf2baec0e243b4fffb89df371a35e55381a45d1d3

Backtrace: 0x40159d37:0x3ffb5580 0x40159d75:0x3ffb55b0
0x40159d37: corereactor::drivers::corem1::CoreM1Driver::uart_event_task() at c:\msys32\opt\xtens
a-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\bits/shared_ptr_base.h:127 (discriminator 1)

0x40159d75: corereactor::drivers::corem1::CoreM1Driver::uart_event_task_impl(void*) at c:\msys32
\opt\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\bits/shared_ptr_base.h:127 (discriminat
or 1)

chegewara, 2020-01-08:
` int bc_counter;` is declared without value, how do you want to increase nothing?
` int bc_counter = 0;` will fix it.

drony, 2020-01-08:
> 
> 
> ` int bc_counter;` is declared without value, how do you want to increase nothing?
> ` int bc_counter = 0;` will fix it.

thank i did it before. nothing changed

chegewara, 2020-01-08:
Then problem is somewhere else.
Backtrace is suggesting uart driver, so you should start in uart code in you app.

koobest, 2020-01-10:
What is the stack size?

Alvin1Zhang, 2020-03-30:
@drony Thanks for reporting, will close due to short of feedback, feel free to reopen with more feedback or details. Thanks.

closed 2020-03-30 by Alvin1Zhang