espressif/esp-idf #4606: Unexpected Output From cJSON Library

physiii, 2020-01-08:
I am starting with latest commit from master branch with updated submodules.

When printing from cJSON library (e.g. cJSON_Print, cJSON_PrintUnformatted) number values are replaced by the character 'g'

Test code:
```
char *string = "{\"key1\":123,\"key2\":321,\"key3\":\"value\"}";
cJSON *json = cJSON_Parse(string);
char *printed_string = cJSON_PrintUnformatted(json);

printf("String:\t%s\n",string);
printf("cJSON Printed String:\t%s\n",printed_string);
```
Output:

> String:    {"key1":123,"key2":321,"key3":"value"}
> cJSON Printed String:    {"key1":g,"key2":g,"key3":"value"}

igrr, 2020-01-09:
Sorry, can not reproduce here with the code provided:

```
String:	{"key1":123,"key2":321,"key3":"value"}
cJSON Printed String:	{"key1":123,"key2":321,"key3":"value"}
```

Please verify that the toolchain version (`xtensa-esp32-elf-gcc --version`) is `crosstool-NG esp-2019r2 8.2.0`. If not, please see https://github.com/espressif/esp-idf/issues/4113#issuecomment-535217053.

physiii, 2020-01-09:
Was going to open a new issue but maybe we can solve this here. For the last few years I just added this to my ~/.bashrc

```
export PATH="$PATH:~/esp/xtensa-esp32-elf/bin"
export IDF_PATH="~/esp/esp-idf"
```
However the new method of setting up the tool chain has failed on all three computers I have tried on, like this:

```
andy@themachine:~/esp/esp-idf$ ./install.sh
Installing ESP-IDF tools
Installing tools: xtensa-esp32-elf, xtensa-esp32s2-elf, esp32ulp-elf, esp32s2ulp-elf, openocd-esp32
Skipping xtensa-esp32-elf@esp-2019r2-8.2.0 (already installed)
Skipping xtensa-esp32s2-elf@esp-2019r2-8.2.0 (already installed)
Skipping esp32ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping esp32s2ulp-elf@2.28.51-esp-20191205 (already installed)
Skipping openocd-esp32@v0.10.0-esp32-20191114 (already installed)
Installing Python environment and packages
Installing Python packages from /home/andy/esp/esp-idf/requirements.txt
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
Ignoring None: markers 'sys_platform == "win32"' don't match your environment
Requirement already satisfied: setuptools>=21 in /home/andy/.espressif/python_env/idf4.1_py2.7_env/lib/python2.7/site-packages (from -r /home/andy/esp/esp-idf/requirements.txt (line 4)) (42.0.2)
Requirement already satisfied: click>=5.0 in /home/andy/.espressif/python_env/idf4.1_py2.7_env/lib/python2.7/site-packages (from -r /home/andy/esp/esp-idf/requirements.txt (line 8)) (7.0)
Requirement already satisfied: pyserial>=3.0 in /home/andy/.espressif/python_env/idf4.1_py2.7_env/lib/python2.7/site-packages (from -r /home/andy/esp/esp-idf/requirements.txt (line 9)) (3.4)
Requirement already satisfied: future>=0.15.2 in /home/andy/.espressif/python_env/idf4.1_py2.7_env/lib/python2.7/site-packages (from -r /home/andy/esp/esp-idf/requirements.txt (line 10)) (0.18.2)
Requirement already satisfied: cryptography>=2.1.4 in /home/andy/.espressif/python_env/idf4.1_py2.7_env/lib/python2.7/site-packages (from -r /home/andy/esp/esp-idf/requirements.txt (line 11)) (2.8)
Requirement already satisfied: pyparsing<2.4.0,>=2.0.3 in /home/andy/.espressif/python_env/idf4.1_py2.7_env/lib/python2.7/site-packages (from -r /home/andy/esp/esp-idf/requirements.txt (line 12)) (2.3.1)
Requirement already satisfied: pyelftools>=0.22 in /home/andy/.espressif/python_env/idf4.1_py2.7_env/lib/python2.7/site-packages (from -r /home/andy/esp/esp-idf/requirements.txt (line 13)) (0.26)
Requirement already satisfied: enum34; python_version < "3" in /home/andy/.espressif/python_env/idf4.1_py2.7_env/lib/python2.7/site-packages (from cryptography>=2.1.4->-r /home/andy/esp/esp-idf/requirements.txt (line 11)) (1.1.6)
Requirement already satisfied: ipaddress; python_version < "3" in /home/andy/.espressif/python_env/idf4.1_py2.7_env/lib/python2.7/site-packages (from cryptography>=2.1.4->-r /home/andy/esp/esp-idf/requirements.txt (line 11)) (1.0.23)
Requirement already satisfied: six>=1.4.1 in /home/andy/.espressif/python_env/idf4.1_py2.7_env/lib/python2.7/site-packages (from cryptography>=2.1.4->-r /home/andy/esp/esp-idf/requirements.txt (line 11)) (1.13.0)
Requirement already satisfied: cffi!=1.11.3,>=1.8 in /home/andy/.espressif/python_env/idf4.1_py2.7_env/lib/python2.7/site-packages (from cryptography>=2.1.4->-r /home/andy/esp/esp-idf/requirements.txt (line 11)) (1.13.2)
Requirement already satisfied: pycparser in /home/andy/.espressif/python_env/idf4.1_py2.7_env/lib/python2.7/site-packages (from cffi!=1.11.3,>=1.8->cryptography>=2.1.4->-r /home/andy/esp/esp-idf/requirements.txt (line 11)) (2.19)
All done! You can now run:

  . ./export.sh
```
Then I export but for some reason the script can't find files that are definitely there.
 
```
andy@themachine:~/esp/esp-idf$ . ./export.sh
Adding ESP-IDF tools to PATH...
bash: ~/esp/esp-idf/tools/idf_tools.py: No such file or directory
```
```
andy@themachine:~/esp/esp-idf$ ls ~/esp/esp-idf/tools/idf_tools.py
/home/andy/esp/esp-idf/tools/idf_tools.py
```
Any idea why I am not able to set up the tool path using the new method? Maybe python2.7 being deprecated?

igrr, 2020-01-09:
Please try modifying the tilde to `$HOME`. I think `~` is not getting expanded.

physiii, 2020-01-09:
Changing `export IDF_PATH="~/esp/esp-idf"` to `export IDF_PATH="/home/user/esp/esp-idf"` worked, thank you!

closed 2020-01-09 by physiii