espressif/esp-idf #4607: add_definitions / add_compile_definitions halt build

malachib, 2020-01-09:
## Environment

- Development Kit:      [none]
- Module or chip used:  [ESP32]
- IDF version (run ``git describe --tags`` to find it): v4.0-beta2
- Build System:         [idf.py]
- Compiler version: 1.22.0-80-g6c4433a
- Operating System:     [Linux]
- Using an IDE?: [No]
- Power Supply:         [USB]

## Problem Description

Utilizing `add_definitions` or `add_compile_definitions` crashes the build.  

### Expected Behavior

Include preprocessor definitions across main project and all custom modules

### Actual Behavior

Apparently the build system adds them twice - once with the `-D` then again without.  The without part it interprets as files to compile, and fails

Unfortunately the `add_compile_definitions` as suggested in Issue #3545 did not work out

### Steps to reproduce

1. Open your main project CMakeLists.txt
2. Add the line `add_compile_definitions(FAIL)` or `add_definitions(-DFAIL)`
3. Execute `idf.py build` and observe failure

### Code to reproduce this issue

No code required

renzbagaporo, 2020-01-13:
Hi @malachib, is this the same error that you get?

```
[4/817] Building C object esp-idf/driver/CMakeFiles/__idf_driver.dir/i2c.c.obj
FAILED: esp-idf/driver/CMakeFiles/__idf_driver.dir/i2c.c.obj 
/home/renz/.espressif/tools/xtensa-esp32-elf/esp-2019r2-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc  -Iconfig -I../../../../components/driver/include -I../../../../components/driver/include/driver -I../../../../components/newlib/platform_include -I../../../../components/freertos/include -I../../../../components/heap/include -I../../../../components/log/include -I../../../../components/soc/esp32/include -I../../../../components/soc/include -I../../../../components/esp_rom/include -I../../../../components/esp_common/include -I../../../../components/xtensa/include -I../../../../components/xtensa/esp32/include -I../../../../components/esp32/include -I../../../../components/esp_ringbuf/include -I../../../../components/esp_event/include -I../../../../components/tcpip_adapter/include -I../../../../components/lwip/include/apps -I../../../../components/lwip/include/apps/sntp -I../../../../components/lwip/lwip/src/include -I../../../../components/lwip/port/esp32/include -I../../../../components/lwip/port/esp32/include/arch -I../../../../components/vfs/include -I../../../../components/esp_wifi/include -I../../../../components/esp_wifi/esp32/include -I../../../../components/esp_eth/include -I../../../../components/efuse/include -I../../../../components/efuse/esp32/include -I../../../../components/app_trace/include -mlongcalls -Wno-frame-address   -DFAIL -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -nostdlib -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Og -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v4.0-beta2-368-g463a9d8b7-dirty\" -DGCC_NOT_5_2_0 -DESP_PLATFORM -MD -MT esp-idf/driver/CMakeFiles/__idf_driver.dir/i2c.c.obj -MF esp-idf/driver/CMakeFiles/__idf_driver.dir/i2c.c.obj.d -o esp-idf/driver/CMakeFiles/__idf_driver.dir/i2c.c.obj   -c /home/renz/Work/esp-idf/components/driver/i2c.c
<command-line>: error: expected identifier before numeric constant
../../../../components/esp_rom/include/esp32/rom/ets_sys.h:636:5: note: in expansion of macro 'FAIL'
     FAIL,
     ^~~~
```

Which doesn't seem like a build system error. It just so happens that the passed compile definition `FAIL`  overrides the existing meaning of `FAIL` in `ets_sys.h`.

malachib, 2020-01-13:
@renzbagaporo nope.  Anything you put in to `add_definitions` causes an issue

Refer to https://bitbucket.org/malachib/playground.esp/src/master/src/PGESP-25/

renzbagaporo, 2020-01-13:
Yeah, I've tried your example, I don't think CMake knows to add a `-D` in front to make it a compile definition, so you have to help it out (since it doesn't fall into compile option territory). See documentation for `add_definitions`: https://cmake.org/cmake/help/v3.5/command/add_definitions.html. 

Using `add_compile_definitions` seem to work as well (without -D since adding it is redundant). You just need newer CMake versions.

malachib, 2020-01-13:
It's true you found a flaw in my example regarding `-D`.  `3.3.1` works OK with that addressed.

On my machine, as per the original post, `4.0-beta2` continues to not function regardless.  Please define "just need newer CMake versions".  I've tried up to 3.13.

renzbagaporo, 2020-01-14:
You are right, @malachib. Using git bisect I've found that 72ddc940e01a290c32c7618fdab9d4fff667431e somehow fixes the issue... I'm not sure what's happening here yet.

0xjakob, 2022-10-10:
@malachib I tested again with your application and the corrected definition (`add_definitions(UNAMIBIGIOUS_ISSUE_4607_FAILURE)` -> `add_definitions(-DUNAMIBIGIOUS_ISSUE_4607_FAILURE)`). There isn't any build error anymore, neither on version 4.1, nor on the latest master. Do you still have this problem?

0xjakob, 2022-11-10:
@malachib We're closing the issue for now. If you still need any assistance related to this issue, please let us know or re-open the issue.

closed 2022-11-10 by 0xjakob

malachib, 2022-11-13:
I haven't encountered this one in a long time.  I think it's resolved.  I'll have to give it a shot and give real feedback