espressif/esp-idf #4608: RMT with same pin for RX and TX does no longer work

astrid2resist, 2020-01-09:
## Environment

- Development Kit:      none
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it):   v4.1-dev-1795-gca8fac876
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
   xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System:     Windows
- (Windows only) environment type: Plain Command Prompt.
- Using an IDE?:  No
- Power Supply:       external power supply

## Problem Description

In our project we use two Microchip UNI/O serial EEPROMs. I read/write them using the RMT module with RX channel and TX channel on the same I/O-Pin (GPIO5 and GPIO14). Everything worked fine until my last "git pull". Now I'm no longer able to read data from the I/O pins.

//Detailed problem description goes here.

### Expected Behavior

read data from RMT

### Actual Behavior

no data read

### Steps to reproduce

### Code to reproduce this issue

Initialization:
```cpp
bool unio_init_rmt(struct unio_t *unio)
{
	ESP_LOGD(TAG, "unio_init_rmt");
    rmt_config_t rmt_tx_cfg =
	{
        .rmt_mode = RMT_MODE_TX,
        .channel = unio->rmt_tx_channel,
        .clk_div = UNIO_RMT_CLK_DIV,
        .gpio_num = unio->gpio,
        .mem_block_num = 1,
        .tx_config = 
		{
            .loop_en = false,
            .carrier_freq_hz = 100, // Not used, but has to be set to avoid divide by 0 err
            .carrier_duty_percent = 50,
            .carrier_level = RMT_CARRIER_LEVEL_LOW,
            .carrier_en = false,
            .idle_level = RMT_IDLE_LEVEL_HIGH,
            .idle_output_en = true,
        }
    };
   rmt_config_t rmt_rx_cfg =
	{
        .rmt_mode = RMT_MODE_RX,
        .channel = unio->rmt_rx_channel,
        .clk_div = UNIO_RMT_CLK_DIV,
        .gpio_num = unio->gpio,
        .mem_block_num = unio->NrOfRXChannels,
        .rx_config = 
		{
            .filter_en = true,
            .filter_ticks_thresh = UNIO_RMT_TICKS_HALF_BIT/2 ,  
            .idle_threshold = UNIO_RMT_MAX_TICK       
        }
    };

    esp_err_t cfg_ok = rmt_config(&rmt_tx_cfg);
    if (cfg_ok != ESP_OK) 
	{
		ESP_LOGE(TAG, "failed to config TX %d", cfg_ok);
 		unio->ErrorCode = UNIO_ERROR_INIT3;
        return false;
    }
    esp_err_t install_ok = rmt_driver_install(rmt_tx_cfg.channel, 0, 0);//ESP_INTR_FLAG_LOWMED | ESP_INTR_FLAG_IRAM | ESP_INTR_FLAG_SHARED); //owb_mt.c: rmt_driver_install(rmt_tx.channel, 0, ESP_INTR_FLAG_LOWMED | ESP_INTR_FLAG_IRAM | ESP_INTR_FLAG_SHARED)
    if (install_ok != ESP_OK) 
	{
		ESP_LOGE(TAG, "failed to install TX %d", install_ok);
 		unio->ErrorCode = UNIO_ERROR_INIT3;
        return false;
    }
   
    cfg_ok = rmt_config(&rmt_rx_cfg);
    if (cfg_ok != ESP_OK) 
	{
		ESP_LOGE(TAG, "failed to config RX %d", cfg_ok);
 		unio->ErrorCode = UNIO_ERROR_INIT3;
        return false;
    }
    install_ok = rmt_driver_install(rmt_rx_cfg.channel, 64*4*sizeof(rmt_item32_t)*unio->NrOfRXChannels, 0);//ESP_INTR_FLAG_LOWMED | ESP_INTR_FLAG_IRAM | ESP_INTR_FLAG_SHARED); //owb_mt.c: rmt_driver_install(rmt_tx.channel, 0, ESP_INTR_FLAG_LOWMED | ESP_INTR_FLAG_IRAM | ESP_INTR_FLAG_SHARED)
    if (install_ok != ESP_OK) 
	{
		ESP_LOGE(TAG, "failed to install RX %d", install_ok);
  		unio->ErrorCode = UNIO_ERROR_INIT3;
       return false;
    }
    rmt_get_ringbuf_handle(unio->rmt_rx_channel, &unio->rb);

>>> from here until end I have tried a lot of combinations, none worked <<<
    // attach GPIO to previous pin
    if (unio->gpio < 32)
    {
        GPIO.enable_w1ts = (0x1 << unio->gpio);
    }
    else
    {
        GPIO.enable1_w1ts.data = (0x1 << (unio->gpio - 32));
    }

    // ATTENTION: set pin for rx first since gpio_output_disable() will
    //            remove rmt output signal in matrix!
//    rmt_set_pin(unio->rmt_rx_channel, RMT_MODE_RX, unio->gpio);
 //   rmt_set_pin(unio->rmt_tx_channel, RMT_MODE_TX, unio->gpio);
		//rx
		gpio_set_direction(unio->gpio, GPIO_MODE_INPUT_OUTPUT_OD);//GPIO_MODE_INPUT);
        gpio_matrix_in(unio->gpio, RMT_SIG_IN0_IDX + unio->rmt_rx_channel, 0);
        //tx
		//gpio_set_direction(unio->gpio, GPIO_MODE_OUTPUT);
        gpio_matrix_out(unio->gpio, RMT_SIG_OUT0_IDX + unio->rmt_tx_channel, 0, 0);

    // force pin direction to input to enable path to RX channel
    PIN_INPUT_ENABLE(GPIO_PIN_MUX_REG[unio->gpio]);

    // enable open drain
    GPIO.pin[unio->gpio].pad_driver = 1;

    return true;
}

```

## Debug Logs

```
D (216615) _UNIO: Read Page 0 lower half
D (216615) rmt: RMT[1] ERR
D (216615) rmt: status: 0x0c040040
D (216615) rmt: RMT[1] ERR
D (216615) rmt: status: 0x0c040040
D (216615) rmt: RMT[1] ERR
D (216615) rmt: status: 0x0c040040
D (216615) rmt: RMT[1] ERR
D (216615) rmt: status: 0x0c040040
D (216615) rmt: RMT[1] ERR
D (216615) rmt: status: 0x0c040040
D (216615) rmt: RMT[1] ERR
D (216615) rmt: status: 0x00040040
D (216645) _UNIO: unio_decode_item_end count 2
W (216645) _UNIO: nByteCounter wrong: 0 (should be 13)

D (222245) _UNIO: Read Page 0 lower half
D (222245) rmt: RMT[5] ERR
D (222245) rmt: status: 0x0c140140
D (222245) rmt: RMT[5] ERR
D (222245) rmt: status: 0x0c140140
D (222245) rmt: RMT[5] ERR
D (222245) rmt: status: 0x0c140140
D (222245) rmt: RMT[5] ERR
D (222245) rmt: status: 0x0c140140
D (222245) rmt: RMT[5] ERR
D (222245) rmt: status: 0x0c140140
D (222245) rmt: RMT[5] ERR
D (222245) rmt: status: 0x00140140
D (222245) _UNIO: unio_decode_item_end count 2
```

Thanks in advance
astrid

astrid2resist, 2020-01-09:
forgotten: working IDF version was v4.1-dev-1126-g95b9b4125

negativekelvin, 2020-01-09:
https://github.com/espressif/esp-idf/commit/a3b79e9202c1990de7d4f5d5032474ca24f49b1a?diff=split#diff-b4297a2fb1ed4bee77a801d825b3a676L269-L271

koobest, 2020-01-10:
Hi, @astrid2resist 
Can you try the following configurations?
```
    gpio_set_direction(unio->gpio, GPIO_MODE_INPUT_OUTPUT);
    gpio_matrix_in(unio->gpio, RMT_SIG_IN0_IDX + unio->rmt_rx_channel, 0);
    gpio_matrix_out(unio->gpio, RMT_SIG_OUT0_IDX + unio->rmt_tx_channel, 0, 0);
    //PIN_INPUT_ENABLE(GPIO_PIN_MUX_REG[unio->gpio]);
```
thanks !!

astrid2resist, 2020-01-13:
Hi koobest,

thank you for your answer. I have tried it, but it doesn't work either.  In addition the serial EEPROM is now not able to pull the data line to GND, because the IO pin is no longer open drain.

regards,
astrid

koobest, 2020-05-18:
Hi, @astrid2resist 
Have you solved your issue?  A similar application (https://github.com/espressif/esp-idf/issues/5237), using RMT to achieve a one-wire bus, without any issues. can you check you configurations?

thanks !!

astrid2resist, 2020-05-18:
Hallo koobest,

thank you for the hint, I will test it and share my results as soon as possible.

Thanks, astrid

astrid2resist, 2020-05-19:
Hallo again,

I have tried a lot but it still doesn't work.

I have tried to:

- add .flags = 0
- added = { 0 } to structure rmt_config_ t creation
- memset the whole structure to 0 
- making  rmt_config_t structure global instead of local
- I tried the gpio setup from koobest: https://github.com/espressif/esp-idf/issues/5237#issuecomment-627215206
- I tried the gpio setup from owb-rmt.c
- I tried to only use one of my two RMT modules.

nothing works. I still can't read my EEPROMs.

My initialization code is:
`bool unio_init_rmt(struct unio_t *unio)
{
	ESP_LOGD(TAG, "unio_init_rmt");
//    rmt_config_t rmt_tx_cfg = {0};
	memset((void *)&unio->rmt_tx_cfg, 0, sizeof(rmt_config_t));
        unio->rmt_tx_cfg.rmt_mode = RMT_MODE_TX;
        unio->rmt_tx_cfg.channel = unio->rmt_tx_channel;
        unio->rmt_tx_cfg.gpio_num = unio->gpio;
        unio->rmt_tx_cfg.clk_div = UNIO_RMT_CLK_DIV;
        unio->rmt_tx_cfg.mem_block_num = 1;
		unio->rmt_tx_cfg.flags = 0;
		unio->rmt_tx_cfg.tx_config.carrier_freq_hz = 100; // Not used, but has to be set to avoid divide by 0 err
		unio->rmt_tx_cfg.tx_config.carrier_level = RMT_CARRIER_LEVEL_LOW;
		unio->rmt_tx_cfg.tx_config.idle_level = RMT_IDLE_LEVEL_HIGH;
		unio->rmt_tx_cfg.tx_config.carrier_duty_percent = 50;
		unio->rmt_tx_cfg.tx_config.carrier_en = false;
		unio->rmt_tx_cfg.tx_config.loop_en = false;
		unio->rmt_tx_cfg.tx_config.idle_output_en = true;

//   rmt_config_t rmt_rx_cfg = {0};
	memset((void *)&unio->rmt_rx_cfg, 0, sizeof(rmt_config_t));
        unio->rmt_rx_cfg.rmt_mode = RMT_MODE_RX;
        unio->rmt_rx_cfg.channel = unio->rmt_rx_channel;
        unio->rmt_rx_cfg.gpio_num = unio->gpio;
        unio->rmt_rx_cfg.clk_div = UNIO_RMT_CLK_DIV;
        unio->rmt_rx_cfg.mem_block_num = unio->NrOfRXChannels;
		unio->rmt_rx_cfg.flags = 0;
		unio->rmt_rx_cfg.rx_config.idle_threshold = UNIO_RMT_MAX_TICK;
		unio->rmt_rx_cfg.rx_config.filter_ticks_thresh = UNIO_RMT_TICKS_HALF_BIT/2;
		unio->rmt_rx_cfg.rx_config.filter_en = true;

    esp_err_t cfg_ok = rmt_config(&unio->rmt_tx_cfg);
    if (cfg_ok != ESP_OK) 
	{
		ESP_LOGE(TAG, "failed to config TX %d", cfg_ok);
 		unio->ErrorCode = UNIO_ERROR_INIT3;
        return false;
    }
    esp_err_t install_ok = rmt_driver_install(unio->rmt_tx_cfg.channel, 0, /* 0);*/ ESP_INTR_FLAG_LOWMED | ESP_INTR_FLAG_IRAM | ESP_INTR_FLAG_SHARED); //owb_mt.c: rmt_driver_install(rmt_tx.channel, 0, ESP_INTR_FLAG_LOWMED | ESP_INTR_FLAG_IRAM | ESP_INTR_FLAG_SHARED);
    if (install_ok != ESP_OK) 
	{
		ESP_LOGE(TAG, "failed to install TX %d", install_ok);
 		unio->ErrorCode = UNIO_ERROR_INIT3;
        return false;
    }
   
    cfg_ok = rmt_config(&unio->rmt_rx_cfg);
    if (cfg_ok != ESP_OK) 
	{
		ESP_LOGE(TAG, "failed to config RX %d", cfg_ok);
 		unio->ErrorCode = UNIO_ERROR_INIT3;
        return false;
    }
    install_ok = rmt_driver_install(unio->rmt_rx_cfg.channel, 64*4*sizeof(rmt_item32_t)*unio->NrOfRXChannels, /*0);*/ESP_INTR_FLAG_LOWMED | ESP_INTR_FLAG_IRAM | ESP_INTR_FLAG_SHARED); //owb_mt.c: rmt_driver_install(rmt_tx.channel, 0, ESP_INTR_FLAG_LOWMED | ESP_INTR_FLAG_IRAM | ESP_INTR_FLAG_SHARED)
    if (install_ok != ESP_OK) 
	{
		ESP_LOGE(TAG, "failed to install RX %d", install_ok);
  		unio->ErrorCode = UNIO_ERROR_INIT3;
       return false;
    }
    rmt_get_ringbuf_handle(unio->rmt_rx_channel, &unio->rb);

    // attach GPIO to previous pin
    if (unio->gpio < 32)
    {
		GPIO.enable_w1ts = (0x1 << unio->gpio);
    }
    else
    {
        GPIO.enable1_w1ts.data = (0x1 << (unio->gpio - 32));
    }
    // attach RMT channels to new gpio pin	  
    // ATTENTION: set pin for rx first since gpio_output_disable() will
    //            remove rmt output signal in matrix!	    
    rmt_set_pin(unio->rmt_rx_channel, RMT_MODE_RX, unio->gpio);
    rmt_set_pin(unio->rmt_tx_channel, RMT_MODE_TX, unio->gpio);
    // force pin direction to input to enable path to RX channel
    PIN_INPUT_ENABLE(GPIO_PIN_MUX_REG[unio->gpio]);
    // enable open drain	    
    GPIO.pin[unio->gpio].pad_driver = 1;

    return true;
}
`
Any more ideas? Any help appreciated!

thanks in advance
astrid

koobest, 2020-05-22:
Hi, Can you get any data from receiver?  How many bytes do you read from UNI/O serial EEPROMs each time?

astrid2resist, 2020-05-25:
Hi again,

after a lot of test-and-error I finally managed to read out my EEPROMs again!
It seems that there were 3 problems:
1) the timing was incorrect, setting flags to 0 fixed this problem
2) the I/O pins were not open collector. I have no idea which of the changes fixed it, but with the source in the last posting, I/O pins work.
3) I did not read anything from my EEPROM. Although I could see a correct data transmission on the oscilloscope,  xRingBufferReceive didn't return data.
I finally found out, that I have to move rmt_rx_stop to the end of the routine, AFTER xRingBufferReceive. I do not know, why, but now it works.

Thanks four the help,
astrid

Alvin1Zhang, 2020-05-26:
@astrid2resist Thanks for reporting and glad to hear that you have resolved the issue. Feel free to reopen if any more updates. Thanks.

closed 2020-05-26 by Alvin1Zhang