espressif/esp-idf #4611: Make fucntions from esp_netif_lwip.c public

no1seman, 2020-01-09:
**Environment:**
- Development Kit:  ESP32-Wrover-Kit
- Kit version (for WroverKit/PicoKit/DevKitC): v4.1
- Module or chip used: ESP32-D0WD
- IDF version: v4.1-dev-1795-gca8fac876
- Build System: CMake
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System: Linux
- Power Supply:  USB|external 5V&Battery

**Problem description:**
There was an ability to set default IF (that needed if using more than 1 IF, for example WiFi STA & Ethernet) in depricated in v4.x TCPIP_ADAPTER API t. I used smth like this:

#include <esp_types.h>
#include <tcpip_adapter.h>
#include <tcpip_adapter_internal.h>
#include <dhcpserver/dhcpserver.h>
#include <lwip/netif.h>

esp_err_t setDefaultIf(tcpip_adapter_if_t tcpip_if)
{
    struct netif *_netif;
    esp_err_t ret = tcpip_adapter_get_netif(tcpip_if, (void **)&_netif);
    if (ret != ESP_OK)
    {
        ESP_LOGE(TAG, "Error setting default netif: %s", esp_err_to_name(ret));
        return ret;
    }
    netif_set_default(_netif);
    return ESP_OK;
}

But in new ESP_NETIF API functions 'esp_netif_set_default_netif', 'esp_netif_get_netif', 'esp_netif_update_default_netif' and so on from esp_netif_lwip.c has no apropriate header file.

**Suggestion:**
Add to ESP IDF a header file for functions and structures from esp_netif_lwip.c

david-cermak, 2020-01-12:
Hi @no1seman 

The refactor of `tcpip_adapter` into `esp-netif` aims to  abstract the functions of the underlying TCP/IP stack so we are trying to limit the number of public APIs. It is still perfectly okay to use the lwip directly and use its API directly (as some IDF examples do). There's a public API `esp_netif_get_netif_impl()` that returns pointer to lwip netif structure, it is however hidden in the `esp_netif_net_stack.h` which makes it a bit awkward to use, so should be probably moved to more appropriate public header.

In case of setting the default netif, there's the option `route_prio` from esp-netif-config struct which could be used to set priority of interfaces and thus if multiple interfaces are `up` the one with the highest `route_prio` is set to be default. There's no way to update this value runtime though. Would an API to configure this value solve your problem? Something like: `erp_err_t esp_netif_set_route_prio(esp_netif_t*, int)`

no1seman, 2020-01-13:
Hi, @david-cermak ,

yes, you are right, it's possible to use lwip API but seems that it's not a best way, beacuse of PPP or custom netif's and route_prio logic, i.e. need  to repeat all the logic.

erp_err_t esp_netif_set_route_prio(esp_netif_t*, int) - seems to be a solution, but IMHO better way to make it possible to set prio during netif creation. For now netif's prio doesn't set at all:
https://github.com/espressif/esp-idf/blob/7a7c4a5486d85242a6b515e300c89197f1c2f62d/components/esp_netif/include/esp_netif_defaults.h#L29
and according to esp_netif_update_default_netif() default IF will be the last started.

Anyway API need to have an ability to init esp_netif->route_prio during IF creation and also API to change esp_netif->route_prio for existing netif's. 

For example to make esp_netif_config definable from user app:
```
esp_netif_t* esp_netif_create_default_wifi_sta(esp_netif_config_t *esp_netif_config)
{
    esp_netif_t *netif = NULL;
    if (!esp_netif_config) {
       esp_netif_config_t cfg = ESP_NETIF_DEFAULT_WIFI_STA();
       netif = esp_netif_new(&cfg);
   } 
   else {
       netif = esp_netif_new(esp_netif_config);
   }

    assert(netif);
    esp_netif_attach_wifi_station(netif);
    esp_wifi_set_default_wifi_sta_handlers();
    return netif;
}
```

david-cermak, 2020-01-14:
@no1seman 

> it's possible to use lwip API but seems that it's not a best way

100 % agree, the preferred way is to use the esp-idf abstraction

> erp_err_t esp_netif_set_route_prio(esp_netif_t*, int) - seems to be a solution

Do you have to update the route priority runtime or is the creation time configuration enough?

> but IMHO better way to make it possible to set prio during netif creation

This should already be possible with the custom configuration style commonly used in esp-idf, i.e. assigning a local variable to the `XXX_DEFAULT_CONFIG()` macro and updating only the fields of interest when passing the configuration to the init/create API. In this case it's not that straight-forward since the `ESP_NETIF_BASE_DEFAULT_ETH()` is a pointer to a const config, but something like the below should work:
```c
    esp_netif_inherent_config_t cfg;
    memcpy(&cfg, ESP_NETIF_BASE_DEFAULT_ETH, sizeof(cfg));
    cfg.route_prio = ...;
    ... // other customised updates
    esp_netif_config_t netif_cfg = { .base = &cfg, .stack = ESP_NETIF_NETSTACK_DEFAULT_ETH };
    netif = esp_netif_new(&netif_cfg);
```
Might be a good idea to provide a macro enabling `esp_netif_inherent_config_t cfg = ESP_NETIF_BASE_CONFIG();` directly.

no1seman, 2020-01-20:
Hi, @david-cermak ,

it will be quite comfortably if can use esp_netif_create_default_wifi_sta(), esp_netif_create_default_wifi_ap() and so on with default esp_netif_inherent_config_t  and also customised one. 

As for updating the route priority in runtime just imagine, that you have multiple network interfaces (in my case it's a home bridge that agregates all data from diffrerrent sensors) and you have an api and web-server to configure it. If API will not have an ability to change prio in runtime you will have to reboot the device if config of network interfaces will changed (for example after insertion of SIM-card PPP IF must me on and have to set priority)...

Well, changing prio in runtime is rare case, but there will be API to get prio and for now it's unavaible (esp_netif_t structure is private in last avaible API). Getting prio needed for routing customisation.

no1seman, 2020-01-20:
@david-cermak ,

also, there is no function to get MAC in ESP_NETIF API.

david-cermak, 2020-01-28:
@no1seman Hi,

Sorry for the late update.
* There's no get the mac of the interface, but still in line with the tcpip_adapter (https://github.com/espressif/esp-idf/blob/release/v4.0/components/tcpip_adapter/tcpip_adapter_lwip.c#L557), though I find it useful to get the netif mac as public interface.
* The APIs like ` esp_netif_create_default_wifi_sta()` and `esp_netif_create_default_wifi_ap()` should be kept for compatibility, but we could introduce another function(s), where the configuration would be supplied.
* I think what you describe should be covered by the internal netif logic, as if a certain interface goes down it gets excluded from choosing the default one.

david-cermak, 2020-01-31:
@no1seman 

so here's my suggestion:

1) custom init of wifi -- Add new API `esp_netif_create_wifi()` to enable such netif creation:
```c
    esp_netif_inherent_config_t esp_netif_config = ESP_NETIF_INHERENT_DEFAULT_WIFI_STA();
    esp_netif_config.if_desc = "simple wifi station";
    esp_netif_config.route_prio = 1;
    esp_netif_create_wifi(WIFI_IF_STA, &esp_netif_config);
    esp_wifi_set_default_wifi_sta_handlers();
```

2) Add new APIs to get mac address, and route prio:
```c
esp_err_t esp_netif_get_mac(esp_netif_t *esp_netif, uint8_t mac[]);

int esp_netif_get_route_prio(esp_netif_t *esp_netif);
```

3) Not introduce any `set_route_prio()` for now

Is this okay with you? Would such API update address your application use?

no1seman, 2020-01-31:
@david-cermak ,

sorry for no answer, was busy with other tasks.

1. OK. Also it will help to solve 3rd case, for example by destroying netif and creating new one with other prio.
2. That will be pretty good, because it's extremely uncomfortable to get netif mac in different handlers.
3. I think that in future in some tasks it will be useful to have such a functionality, but now it's not a problem.

Thank you for attention to my problems, will wait for commit.

closed 2020-02-11 by espressif-bot

AxelLin, 2020-02-16:
Hi @david-cermak 
If you can provide an API to get current default netif, it would be much
easier to debug/verify routing issue when multiple interfaces are used
and network interface can be up/down at run-time.

gtec2, 2020-11-16:
> Hi @no1seman
>
> The refactor of `tcpip_adapter` into `esp-netif` aims to abstract the functions of the underlying TCP/IP stack so we are trying to limit the number of public APIs. It is still [...]

Hi @david-cermak. I'm using NETIF library and using WiFi with priority 128 and PPP with priority 64. Both are (should) ap together. I have an active mqtt connection and I was able to verify that if the WiFi loses connection (WIFI DISCONNETED) it automatically reconnects through PPP (after MQTT_DISCONNECT_EVENT). But when WiFi reconnects, MQTT connection stays indefinitely via PPP, it doesn't automatically switch to WiFi, which is what I hope will happen since WiFi has higher priority. How can I force that change?

AxelLin, 2020-11-18:
@gtec2  Please create a new issue since this was closed and the subject line does not match your issue.