espressif/esp-idf #4612: Add support of start/stop marker and KEEP keyword in linkerscript fragments

chapipo, 2020-01-09:
**Is your feature request related to a problem? Please describe.**

I would like to be able to put some struct in a special section in the linker script and put start and stop marker to be able to loop on them. When trying to do this with linkerfragments, I cannot put marker and I cannot force some section to be kept (with the KEEP keyword).

**Describe the solution you'd like**

A modification to be able to put start / stop marker in linkerscript fragments.

**Describe alternatives you've considered**

At the moment, the only way to do this, is to patch est-idf linkerscript.

Linkerscript extract example:
```
 _s_custom_section = ABSOLUTE(.);
KEEP(*(.custom.section))
_e_custom_section = ABSOLUTE(.);
```

I found on esp32 forum a similar request: https://esp32.com/viewtopic.php?t=13102

Alvin1Zhang, 2020-01-10:
@chapipo Thanks for raising the feature request.

renzbagaporo, 2020-04-28:
Hi @chapipo, this is also a desirable feature for us internally, so we definitely take this into consideration. However, this is a non-trivial feature so I may not be able to give an exact timeline when this feature will be implemented.

renzbagaporo, 2021-03-09:
Hi @chapipo, this feature should be merged to master in a couple of weeks. The commit will automatically close this issue.

closed 2021-03-12 by espressif-bot