espressif/esp-idf #4616: pppos_client does not allow setting PPPAUTHTYPE_NONE

AxelLin, 2020-01-10:
In current pppos_client example, it does not allow setting PPPAUTHTYPE_NONE.

1. I got compile error "Unsupported AUTH Negotiation" if disable both PAP and CHAP.
   So it is not possible to set PPPAUTHTYPE_NONE (which is supported by lwIP).

2. The pppos_client example uses CONFIG_LWIP_PPP_PAP_SUPPORT and CONFIG_LWIP_PPP_CHAP_SUPPORT
   to decide auth_type, but this is obviously wrong.
   The CONFIG_LWIP_PPP_PAP_SUPPORT and CONFIG_LWIP_PPP_CHAP_SUPPORT is supposed
   to enable the capability to support PAP/CHAP.
   The application(pppos_client) should be able to decide using which auth_type.

3. Looking at esp_netif_ppp.h, it even does not support setting PPPAUTHTYPE_NONE.
   I suggest adding "NETIF_PPP_AUTHTYPE_NONE = 0x00" to esp_netif_auth_type_t.
   So at least people who developer their own pppos application can set it.

4. Suggest to enable both PRINTPKT_SUPPORT and PPP_PROTOCOLNAME if PPP_DEBUG_ON is set.
   This would be helpful to identify issue and also helpful for report issues.

AxelLin, 2020-01-10:
In lwIP, the ppp_set_auth() is guarded by #if PPP_AUTH_SUPPORT.

However, In components/lwip/lwip/src/netif/ppp/pppapi.c
pppapi_set_auth()/pppapi_do_ppp_set_auth() are guarded by #if ESP_PPP.
This is wrong because even ESP_PPP=1, PPP_AUTH_SUPPORT can be not set.
Should change to #if PPP_AUTH_SUPPORT guard.

In additional, I think the esp_netif_ppp_set_auth should be guard by #if PPP_AUTH_SUPPORT as well.

BTW, I just checked the lwip version used in ESP-IDF-v4.1-dev, it's quite old compare to the same
(2.0.3-esp) branch in esp-lwip tree. I think the lwip submodule needs update.

AxelLin, 2020-01-17:
I have sent some fixes below, please help review it.

1)
In esp-lwip: https://github.com/espressif/esp-lwip/pull/10
pppapi_set_auth() needs #if PPP_AUTH_SUPPORT guard

Otherwise lwIP build fails if both PAP and CHAP are disabled.

The ppp_set_auth() is guarded by PPP_AUTH_SUPPORT, so the pppapi_set_auth
and pppapi_do_ppp_set_auth should also use the same #if guard.

2)
In esp-idf: https://github.com/espressif/esp-idf/pull/4639
Allow setting NETIF_PPP_AUTHTYPE_NONE

Add NETIF_PPP_AUTHTYPE_NONE to esp_netif_auth_type_t and allow
esp_netif_ppp_set_auth to set auth_type with NETIF_PPP_AUTHTYPE_NONE.

Alvin1Zhang, 2020-04-01:
@AxelLin Thanks for reporting and sharing the PR, we have a fix internally now under reviewing/taken for internal reviewing, will keep you posted. Thanks.

closed 2020-04-24 by espressif-bot