espressif/esp-idf #4617: Potential buffer overflow in  http_header_generate_string() in file http_header.c

umer-ilyas, 2020-01-10:
## Environment

- Development Kit:      custom
- Kit version 
- Module or chip used:   ESP32-WROOM-32D 
- IDF version  v3.3.1
- Build System:         Make
- Compiler version 1.22.0-80-g6c4433a5
- Operating System:     Windows
- (Windows only) environment type: MSYS2 mingw32
- Using an IDE?:     No
- Power Supply:         Battery

## Problem Description

Potential buffer overflow in  http_header_generate_string() in file http_header.c at line 190.

### Expected Behavior
loop should break here and also should print error here that buffer_size is smaller than required 
### Actual Behavior
loop is not breaking here and buffer gets overflow and write data in undesirable location of memory 
### Steps to reproduce

init http client
  esp_http_client_handle_t client = esp_http_client_init(&config);
  config.buffer_size = 100;  (default size is 512)
set multiple header here so that size of headers (strings) gets bigger than  config.buffer_size 
  esp_http_client_set_header(client, "authorization", accessToken);

code will not issue any warning and http request gets failed 

I think issue is fixed in v4.x and  that patch needs to be added here too in next release of v3.3.x

shubhamkulkarni97, 2020-01-13:
@umer-ilyas The issue has been fixed in our internal repository for v3.3.x as well. When the changes are synced with Github, your issue will be fixed.

mahavirj, 2020-01-14:
@umer-ilyas Fix is available on `release/v3.3` branch as well (https://github.com/espressif/esp-idf/commit/b7a9f5e115ca8d58186baad061251365c9dededd#diff-c7c711c57aa3482655dd39165cf29087). Closing this issue.

closed 2020-01-14 by mahavirj