espressif/esp-idf #4619: esp-idf/examples/bluetooth/bluedroid/classic_bt/hfp_hf   example

ACJ-kor, 2020-01-10:
I execute the 'hfp_hf.' example without any modification but i can't see the log outcome of 'Audio Connection and Disconnection'. Please refer my attachment file!!!
[image]

AbnerFederer, 2020-01-10:
[image]

ACJ-kor, 2020-01-14:
[image]

When i modified 'BR/EDR Sync(SCO/eSCO)' in menuconfig 0 into 1 then the following picture appears!!!

But 'handle hex' value didn't print value 181 but  value 180.

What do i add more configuration? and What does the diffierence between two values mean? 

also when smartphone is connected to 'ESP32', can i see two devices are disconnected? can you let me know how to keep connecting between smartphone and ESP32?

Specification :
Board: nodemcu (esp-wroom-32)
version: ESP-IDF master
power: usb

[image]

AbnerFederer, 2020-01-14:
Hi @ACJ-kor ,

1. BR/EDR Max Connection means the number of SCO links at your case.
2. Handle is just a number allocated by HCI from 180, it has no other meaning.
3. We have specific disconnected log with disconnecting reason.

Thanks
Abner

Alvin1Zhang, 2020-02-19:
@ACJ-kor Thanks for reporting. Feel free to reopen if the issue still happens, or help open another issue for new issues. Thanks.

closed 2020-02-19 by Alvin1Zhang