espressif/esp-idf #4620: Reading and writing the same virtual address multiple times will cause the system to crash

GitOfLance, 2020-01-10:
### Environment
- Module or chip used: ESP32-WROOM-32
- IDF version :  v4.0-dev-1443-g39f090a4f 2nd stage bootloader
- Build System: Make
- Compiler version : xtensa-esp32-elf-
- Operating System: Ubuntu 18.04
- Power Supply: USB

### Problem Description
In the file copy function, I defined a structure variable (not a pointer). Read out a structure-sized data in a loop, and then write this structure data into a backup file until all data is backed up.

When I call this function multiple times in succession, the system crashes. I changed this variable to a pointer and allocated space through the `malloc ()` function before each operation. The problem of system crash still exists.
But I found that the value of this structure pointer variable is the same every time it crashes. That is, each time the crash, the same address is operated. Is this address a virtual address mapped by the `MMU`? Why does reading and writing the same address multiple times cause the system to crash?

### Debug record
#### use structure variable
- Struct：
```
typedef struct {
    int unumber;                
    char userid[USER_ID_LEN];         
    int type;      
    unsigned long long int time;     
    char ways;         
}UserPermission __attribute__ ((packed));

```
- Code：not pointer
```
/**
 * FunctionName	: record_file_backup
 * Description	: 用于操作文件前拷贝文件做备份
    源文件是有数据的，目标文件内容全部清除
 每次更改文件前要先备份，一方面保证数据的完整性，
 一方面尝试读写文件，防止文件操作失败
 * Input	: srcfile -- 源文件
                  destfile -- 目标文件
 * Output	: 文件
 * Return	: -1,出错；0，成功
 * Others	: 仅限于内部使用，不供外部使用
 */
static int record_file_backup(char *srcfile, char *destfile)
{
    FILE *fpsrc = NULL;
    FILE *fpdest = NULL;
    int ret = -1;
    UserPermission user; //暂定为用户类型，不用缓冲池
    struct stat st;
    
    //申请锁
    xSemaphoreTake(Mutex_openrecord, portMAX_DELAY);
    
    do {
        //获取原文件大小
        if (stat(srcfile, &st) < 0) //文件不存在
        {
            ESP_LOGE(TAG, "failed to stat, file not exist");
            ret = 0;
            break;
        }
        ESP_LOGI(TAG, "source file size is %ld", st.st_size);
        //vTaskDelay(5000 / portTICK_PERIOD_MS); //未联网刷卡多次崩溃，确认问题点

        printf("user addr. is %p\n", &user);
        printf("sizeof is %d\n", sizeof(UserPermission));

        //打开文件
        fpsrc = fopen(srcfile, "r");
        fpdest = fopen(destfile, "w");
        if (NULL == fpsrc || NULL == fpdest)
[... 250 lines cut ...]

0x4016f961: fwrite at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/fwrite.c:211

0x40107baa: record_file_backup at /home/lance/Workspace/esp/glock/components/device/door.c:542

0x40107d69: door_record_wirte at /home/lance/Workspace/esp/glock/components/device/door.c:542

0x40107f73: door_wait_open at /home/lance/Workspace/esp/glock/components/device/door.c:542

0x40095045: vPortTaskWrapper at /home/lance/Workspace/esp/esp-idf/components/freertos/port.c:435
```

This is the last message. The value of the pointer user is **`0x3ffd0a38`**.
[image]

This is the penultimate message:
[image]

This is the third last message:
[image]

- Extra error logs：
[image]

[image]

negativekelvin, 2020-01-10:
Looks like you may have an interrupt that is marked as iram but calling code in flash

GitOfLance, 2020-01-13:
Yes，I used smart configuration（EspTouch） to configure wif. 
How can I solve this problem?

GitOfLance, 2020-01-13:
Hi, @negativekelvin 
I remember, I will start smartconfig on the ESP32, but I don't use EspTouch to configure the network. Then test the above code and this problem will occur.
If I start the smartconfig function and use EspTouch to connect the ESP32 to the network, there is no problem in testing the above code.