espressif/esp-idf #4622: esp_wifi_disconnect() generates an exception

JB-DX, 2020-01-11:
Environment
- IDF version v4.1-dev-1572-g30372f5a4-dirty
- Build System:        idf.py

System runs fine with Wifi.
But calling esp_wifi_disconnect() results in this exception:

> I (642074) wifi: state: run -> init (0)
> I (642074) wifi: pm stop, total sleep time: 607620249 us / 632054767 us
> 
> I (642074) wifi: new:<6,0>, old:<6,0>, ap:<255,255>, sta:<6,0>, prof:1
> I (642084) tag: start wGuru Meditation Error: Core  1 panic'ed (LoadProhibited).
>  Exception was unhandled.
> Core 1 register dump:
> PC      : 0x40155a91  PS      : 0x00060630  A0      : 0x800e799c  A1      : 0x3ffbdb20
> 0x40155a91: esp_netif_is_netif_up at C:/mysys32/home/admin/esp-idf/components/es
> p_netif/lwip/esp_netif_lwip.c:1107 (discriminator 2)
> 
> A2      : 0xf0de1e1e  A3      : 0x0000000a  A4      : 0x00000000  A5      : 0x3ffb46c0
> A6      : 0x00000005  A7      : 0x00000000  A8      : 0x3ffd1a80  A9      : 0x3ffd18e8
> A10     : 0x00000001  A11     : 0x00000000  A12     : 0x3ffb9938  A13     : 0x00000000
> A14     : 0x00000000  A15     : 0x3ffd2bb0  SAR     : 0x00000010  EXCCAUSE: 0x0000001c
> EXCVADDR: 0xf0de1f09  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff
> 
> ELF file SHA256: b7099c666c9adb801188dc14a21dde943bebefc6c21e02e93b0aabc4cfc7f389
> 
> Backtrace: 0x40155a8e:0x3ffbdb20 0x400e7999:0x3ffbdb40 0x400e7c58:0x3ffbdb60 0x4
> 00e71f2:0x3ffbdb80 0x400ebb3d:0x3ffbdba0 0x400ebbe4:0x3ffbdbc0 0x40093689:0x3ffbdbf0
> 0x40155a8e: esp_netif_is_netif_up at C:/mysys32/home/admin/esp-idf/components/es
> p_netif/lwip/esp_netif_lwip.c:1107 (discriminator 1)
> 
> 0x400e7999: esp_netif_update_default_netif at C:/mysys32/home/admin/esp-idf/comp
> onents/esp_netif/lwip/esp_netif_lwip.c:184
> 
> 0x400e7c58: esp_netif_down_api at C:/mysys32/home/admin/esp-idf/components/esp_n
> etif/lwip/esp_netif_lwip.c:1096
> 
> 0x400e71f2: esp_netif_api_cb at C:/mysys32/home/admin/esp-idf/components/esp_net
> if/lwip/esp_netif_lwip.c:94
> 
> 0x400ebb3d: tcpip_thread_handle_msg at C:/mysys32/home/admin/esp-idf/components/
> lwip/lwip/src/api/tcpip.c:168
> 
> 0x400ebbe4: tcpip_thread at C:/mysys32/home/admin/esp-idf/components/lwip/lwip/s
> rc/api/tcpip.c:154
> 
> 0x40093689: vPortTaskWrapper at C:/mysys32/home/admin/esp-idf/components/freerto
> s/port.c:143
> 
> 
> Rebooting...
> 

This worked fine in an earlier IDF version.

Any idea what goes wrong?
The reason to call esp_wifi_disconnect() is the subsequent esp_wifi_wps_start(0)

freakyxue, 2020-01-20:
hi @JB-DX  Can you give me the engineering code to repeat the question

freakyxue, 2020-01-20:
esp_err_t ret = nvs_flash_init();
    if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND) {
      ESP_ERROR_CHECK(nvs_flash_erase());
      ret = nvs_flash_init();
    }
    ESP_ERROR_CHECK(ret);

    ESP_LOGI(TAG, "ESP_WIFI_MODE_STA");
    wifi_init_sta();
    vTaskDelay(5000 / portTICK_PERIOD_MS);
    printf("wifi is ok\n");
    esp_wifi_disconnect();

freakyxue, 2020-01-20:
@JB-DX  I called this function, I can't reproduce the problem

JB-DX, 2020-01-20:
The module is also connected to LAN8270 for ethernet.
During powerup when no ethernet is connected the module switches to WiFi and shuts down ethernet and LAN8270.
I suspected the problem in this situation. After having fixed other problems with ethernet I undid all changes in order to reproduce the esp_wifi_disconnect problem. To my surprise it never occured again. Maybe it got "solved" by re-compiling the entire project when I changed the log-level. I tried the original log-level also, but that now also works.

It seems the problem is solved now. I only hope it remains that way.
Thank you for your help.

freakyxue, 2020-01-21:
@JB-DX   May I close this question?

JB-DX, 2020-01-21:
Yes, you may do so.

Alvin1Zhang, 2020-01-21:
@JB-DX Thanks for the updates. Feel free to reopen if the issue still happens. Thanks.

closed 2020-01-21 by Alvin1Zhang