espressif/esp-idf #4624: Not able to set cookie in HttpClient

animeshjaipurkar, 2020-01-12:
I am trying to use esp_http_client_set_header() to set **cookie** name / value pair as follows:

esp_http_client_set_header(client, "token", "1234")

However, this does not set the cookie. Is this the correct way to set the cookie? I could not find a esp_http_client_set_cookie function.

Thank you for your help.

shubhamkulkarni97, 2020-01-13:
@animeshjaipurkar I tried using the same API to set cookie,  the header is set successfully and is visible in Wireshark capture.

Can you please try adding `esp_http_client_set_header` after `esp_http_client_init` and capture the data in Wireshark?

In case this doesn't work for you, can you share sample code and Wireshark capture?

animeshjaipurkar, 2020-01-16:
@shubhamkulkarni97  Thank you for the quick help. I appreciate it. 

I modified the parameters as follows to the set the cookie:
esp_http_client_set_header(client, "Cookie", "token=1234")

This seems to have set the cookie.

mahavirj, 2020-01-17:
Closing per above comment.

closed 2020-01-17 by mahavirj