espressif/esp-idf #4626: wifi 断开重连

hnhkj, 2020-01-13:
MQTT发送数据失败之后，wifi进入循环重连状态，不能正常连接网络。

<pre><font color="#C4A000">commit e28c5127c239504091adf6d353efb2fb0252fe92 (</font><font color="#34E2E2"><b>HEAD -&gt; </b></font><font color="#8AE234"><b>release/v3.3</b></font><font color="#C4A000">, </font><font color="#EF2929"><b>origin/release/v3.3</b></font><font color="#C4A000">, </font><font color="#8AE234"><b>v3.3</b></font><font color="#C4A000">)</font>
</pre>

```
I (1090228) BED_MQTT: tape data: 21 03 00 78 9C
I (1093788) wifi: bcn_timout,ap_probe_send_start
I (1096288) wifi: ap_probe_send over, resett wifi status to disassoc
I (1096288) wifi: state: run -> init (c800)
I (1096288) wifi: pm stop, total sleep time: 430548257 us / 852641359 us

I (1096298) wifi: new:<1,0>, old:<1,1>, ap:<255,255>, sta:<1,1>, prof:1
I (1096308) BED_MQTT: mqtt_event (0)
I (1096308) WIFI_STATION: wifi_event_handler:[5]
E (1096308) MQTT_CLIENT: Error write data or timeout, written len = -1, errno=113
W (1096318) WIFI_STATION: retry to connect to the AP 1
I (1096318) BED_MQTT: mqtt_event (2)
W (1096328) MQTT_CLIENT: Publish: Losing qos0 data when client not connected
I (1096338) BED_MQTT: esp mqtt client public msg_id=0
I (1096348) BED_MQTT: tape data: 21 01 03 20 45
I (1096348) BED_MQTT: tape data: 21 03 00 6E 92
I (1098368) WIFI_STATION: wifi_event_handler:[5]
W (1098368) WIFI_STATION: retry to connect to the AP 2
I (1098758) BED_MQTT: tape data: 21 01 02 F8 1C
I (1100198) BED_MQTT: tape data: 21 03 00 6E 92
I (1100428) WIFI_STATION: wifi_event_handler:[5]
W (1100428) WIFI_STATION: retry to connect to the AP 3
I (1102478) WIFI_STATION: wifi_event_handler:[5]
W (1102478) WIFI_STATION: retry to connect to the AP 4
I (1103798) BED_MQTT: tape data: 21 01 02 DA FE
I (1104538) WIFI_STATION: wifi_event_handler:[5]
W (1104538) WIFI_STATION: retry to connect to the AP 5
I (1105238) BED_MQTT: tape data: 21 03 00 64 88
I (1106338) BED_MQTT: mqtt_event (7)
I (1106588) WIFI_STATION: wifi_event_handler:[5]
W (1106588) WIFI_STATION: retry to connect to the AP 6
```

liuzfesp, 2020-01-20:
Hi @hnhkj, need you help to provide following info:

- When the problem happens, can the mobile phone connect to the router?  I want to know whether the router is in good status or bad status when problem happens.
- Does this issue easy to reproduce? How frequency does this issue happen?

liuzfesp, 2020-02-11:
HI @hnhkj , any feedback of this issue?

hnhkj, 2020-02-13:
stack overflow. I use a funciton cJSON_PrintUnformatted(). this function eat all of RAM. I disable this function. MQTT continue running.

Alvin1Zhang, 2020-02-17:
@hnhkj Thanks for the updates, and good to know that it continues working. Feel free to reopen if the issue still happens or help to open another ticket for new issue. Thanks.

closed 2020-02-17 by Alvin1Zhang