espressif/esp-idf #4627: Using select() with stdin

eduardsui, 2020-01-13:
Hello,

I'm facing a issue when running select() with stdin file descriptor. abort() gets called in locks.c, here:
```
    if (!xPortCanYield()) {
        /* In ISR Context */
        if (mutex_type == queueQUEUE_TYPE_RECURSIVE_MUTEX) {
            abort(); /* recursive mutexes make no sense in ISR context */
        }
```

The offending code is:
```
void app_main() {
    while (1) {
        int s;
        fd_set rfds;
        struct timeval tv = {
            .tv_sec = 100,
            .tv_usec = 0,
        };

        FD_ZERO(&rfds);
        FD_SET(fileno(stdin), &rfds);

        s = select(2, &rfds, NULL, NULL, &tv);

        if (s < 0) {
            printf("Select failed: errno %d", errno);
        } else if (s == 0) {
            printf("Timeout has been reached and nothing has been received");
        }
    }
}
```

igrr, 2020-01-13:
Please make sure you are not calling select from an interrupt or a critical section.

dobairoland, 2020-01-13:
The crash is caused by an assertion which tries to print (from a critical section) that the UART driver is not installed.

@eduardsui Please call `uart_driver_install()` and `esp_vfs_dev_uart_use_driver()` in the beginning. 

Please take a look at driver initialization in the following examples:
https://github.com/espressif/esp-idf/tree/master/examples/peripherals/uart/uart_select
https://github.com/espressif/esp-idf/tree/master/examples/system/select

eduardsui, 2020-01-13:
Thank you, it works perfectly after UART driver is installed.

closed 2020-01-13 by eduardsui