espressif/esp-idf #4629: macOS - install.sh fails to download binutils

evhv, 2020-01-13:
## Environment
- IDF version: v4.1-dev-1895-g7a7c4a548
- Build System:         [CMake]
- Operating System:     [macOS 10.12.6 Sierra]
- Using an IDE?: [No]

## Problem Description
Good afternoon, I am trying to install esp-idf tools on a new machine using install.sh script.
idf_tools.py fails at downloading binutils-esp32ulp-macos-2.28.51-esp-20191205.tar.gz 
<pre><code>Evgs-MacBook-Pro-2:esp-idf ev$ ./install.sh
Installing ESP-IDF tools
Installing tools: xtensa-esp32-elf, xtensa-esp32s2-elf, esp32ulp-elf, esp32s2ulp-elf, openocd-esp32
Skipping xtensa-esp32-elf@esp-2019r2-8.2.0 (already installed)
Skipping xtensa-esp32s2-elf@esp-2019r2-8.2.0 (already installed)
Installing esp32ulp-elf@2.28.51-esp-20191205
Downloading binutils-esp32ulp-macos-2.28.51-esp-20191205.tar.gz to /Users/ev/.espressif/dist/binutils-esp32ulp-macos-2.28.51-esp-20191205.tar.gz.tmp
WARNING: Download failure [Errno socket error] [SSL: TLSV1_ALERT_PROTOCOL_VERSION] tlsv1 alert protocol version (_ssl.c:590)
WARNING: Failed to download https://github.com/espressif/binutils-esp32ulp/releases/download/v2.28.51-esp-20191205/binutils-esp32ulp-macos-2.28.51-esp-20191205.tar.gz to /Users/ev/.espressif/dist/binutils-esp32ulp-macos-2.28.51-esp-20191205.tar.gz.tmp
Downloading binutils-esp32ulp-macos-2.28.51-esp-20191205.tar.gz to /Users/ev/.espressif/dist/binutils-esp32ulp-macos-2.28.51-esp-20191205.tar.gz.tmp
WARNING: Download failure [Errno socket error] [SSL: TLSV1_ALERT_PROTOCOL_VERSION] tlsv1 alert protocol version (_ssl.c:590)
WARNING: Failed to download https://github.com/espressif/binutils-esp32ulp/releases/download/v2.28.51-esp-20191205/binutils-esp32ulp-macos-2.28.51-esp-20191205.tar.gz to /Users/ev/.espressif/dist/binutils-esp32ulp-macos-2.28.51-esp-20191205.tar.gz.tmp
Downloading binutils-esp32ulp-macos-2.28.51-esp-20191205.tar.gz to /Users/ev/.espressif/dist/binutils-esp32ulp-macos-2.28.51-esp-20191205.tar.gz.tmp
WARNING: Download failure [Errno socket error] [SSL: TLSV1_ALERT_PROTOCOL_VERSION] tlsv1 alert protocol version (_ssl.c:590)
WARNING: Failed to download https://github.com/espressif/binutils-esp32ulp/releases/download/v2.28.51-esp-20191205/binutils-esp32ulp-macos-2.28.51-esp-20191205.tar.gz to /Users/ev/.espressif/dist/binutils-esp32ulp-macos-2.28.51-esp-20191205.tar.gz.tmp
ERROR: Failed to download, and retry count has expired
Traceback (most recent call last):
  File "/Users/ev/esp/esp-idf/tools/idf_tools.py", line 1387, in <module>
    main(sys.argv[1:])
  File "/Users/ev/esp/esp-idf/tools/idf_tools.py", line 1383, in main
    action_func(args)
  File "/Users/ev/esp/esp-idf/tools/idf_tools.py", line 1167, in action_install
    tool_obj.download(tool_version)
  File "/Users/ev/esp/esp-idf/tools/idf_tools.py", line 584, in download
    raise DownloadError()
__main__.DownloadError</code></pre>
however it downloaded and extracted xtensa-esp32-elf and xtensa-esp32s2-elf with no issues.
<pre><code>Evgs-MacBook-Pro-2:.espressif ev$ tree -L 2
.
|-- dist
|   |-- xtensa-esp32-elf-gcc8_2_0-esp-2019r2-macos.tar.gz
|   `-- xtensa-esp32s2-elf-gcc8_2_0-esp-2019r2-macos.tar.gz
`-- tools
    |-- xtensa-esp32-elf
    `-- xtensa-esp32s2-elf</code></pre>

Manually downloading and placing it into .espressif/dist leads to file size mismatch.
<pre><code>Evgs-MacBook-Pro-2:esp-idf ev$ ./install.sh
Installing ESP-IDF tools
Installing tools: xtensa-esp32-elf, xtensa-esp32s2-elf, esp32ulp-elf, esp32s2ulp-elf, openocd-esp32
Skipping xtensa-esp32-elf@esp-2019r2-8.2.0 (already installed)
Skipping xtensa-esp32s2-elf@esp-2019r2-8.2.0 (already installed)
Installing esp32ulp-elf@2.28.51-esp-20191205
WARNING: file size mismatch for /Users/ev/.espressif/dist/binutils-esp32ulp-macos-2.28.51-esp-20191205.tar.gz, expected 8872874, got 22804480
WARNING: removing downloaded file binutils-esp32ulp-macos-2.28.51-esp-20191205.tar.gz and downloading again
Downloading binutils-esp32ulp-macos-2.28.51-esp-20191205.tar.gz to /Users/ev/.espressif/dist/binutils-esp32ulp-macos-2.28.51-esp-20191205.tar.gz.tmp
WARNING: Download failure [Errno socket error] [SSL: TLSV1_ALERT_PROTOCOL_VERSION] tlsv1 alert protocol version (_ssl.c:590)
WARNING: Failed to download https://github.com/espressif/binutils-esp32ulp/releases/download/v2.28.51-esp-20191205/binutils-esp32ulp-macos-2.28.51-esp-20191205.tar.gz to /Users/ev/.espressif/dist/binutils-esp32ulp-macos-2.28.51-esp-20191205.tar.gz.tmp
Downloading binutils-esp32ulp-macos-2.28.51-esp-20191205.tar.gz to /Users/ev/.espressif/dist/binutils-esp32ulp-macos-2.28.51-esp-20191205.tar.gz.tmp
WARNING: Download failure [Errno socket error] [SSL: TLSV1_ALERT_PROTOCOL_VERSION] tlsv1 alert protocol version (_ssl.c:590)
WARNING: Failed to download https://github.com/espressif/binutils-esp32ulp/releases/download/v2.28.51-esp-20191205/binutils-esp32ulp-macos-2.28.51-esp-20191205.tar.gz to /Users/ev/.espressif/dist/binutils-esp32ulp-macos-2.28.51-esp-20191205.tar.gz.tmp
Downloading binutils-esp32ulp-macos-2.28.51-esp-20191205.tar.gz to /Users/ev/.espressif/dist/binutils-esp32ulp-macos-2.28.51-esp-20191205.tar.gz.tmp
WARNING: Download failure [Errno socket error] [SSL: TLSV1_ALERT_PROTOCOL_VERSION] tlsv1 alert protocol version (_ssl.c:590)
WARNING: Failed to download https://github.com/espressif/binutils-esp32ulp/releases/download/v2.28.51-esp-20191205/binutils-esp32ulp-macos-2.28.51-esp-20191205.tar.gz to /Users/ev/.espressif/dist/binutils-esp32ulp-macos-2.28.51-esp-20191205.tar.gz.tmp
ERROR: Failed to download, and retry count has expired
Traceback (most recent call last):
  File "/Users/ev/esp/esp-idf/tools/idf_tools.py", line 1387, in <module>
    main(sys.argv[1:])
  File "/Users/ev/esp/esp-idf/tools/idf_tools.py", line 1383, in main
    action_func(args)
  File "/Users/ev/esp/esp-idf/tools/idf_tools.py", line 1167, in action_install
    tool_obj.download(tool_version)
  File "/Users/ev/esp/esp-idf/tools/idf_tools.py", line 584, in download
    raise DownloadError()
__main__.DownloadError</code></pre>

Thank you in advance for your help,
Ev

evhv, 2020-01-13:
Sorry, I oversaw that by default Mac uses the old version of python, by explicitly calling
<pre><code>python3.7 ~/esp/esp-idf/tools/idf_tools.py install</code></pre>
all works as expected.

Thank you and all the best,
Ev

closed 2020-01-13 by evhv

exrector, 2020-10-01:
I have the same error. This option does not help me. There are other options?

Craigzyc, 2021-02-07:
python3 ~/esp/esp-idf/tools/idf_tools.py install
worked for me