espressif/esp-idf #4630: ESP32 only work on USB power

JDchew, 2020-01-14:
Hi all,

    I had try to built the sample code from esp-mdf -> example -> function demo -> mconfig and flash to ESP32-WROOM-32D. 
    
    I face issue when I connect using DC power supply, my power supply are able to draw up to 2A.
What I expected is the behavior should same as USB power supply, but it show me the RED LED keep blinking and not able to run the program.

    
[image]

    Can anyone inform me what is the different between I power up using USB versus power up using DC power supply ? I notice when I power up using USB I monitor using terminal it will had something running.

projectgus, 2020-01-14:
Hi @JDchew ,

Is the board you're using the "DOIT" board shown in the picture? You may need to ask the development board manufacturer about this, this is related to the hardware board you have not the ESP32 itself.

How do you determine that the problem is not running? Are you looking at the serial output or something else?

Angus

JDchew, 2020-01-14:
Sorry it my mistake here. I attached the ESP32 Dev board to Breadboard, and somehow inside my breadboard I believe got some short circuit, I take out standalone power up using power supply it was running fine.

closed 2020-01-14 by JDchew