espressif/esp-idf #4633: How to use esp-mesh (no root) with smartconfig ?

Gesiz, 2020-01-15:
## Environment

- Development Kit:     [ESP32-DevKitC]
- Kit version  :             [v4]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version               [v3.3]
- Build System:            [Make]
- Compiler version  [xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0]
- Operating System:     [Windows]
- (Windows only) environment type: [WSL(Debian)]
- Using an IDE?:          [vscode]
- Power Supply:          [USB]

## Problem Description

When I tried to use (mesh no route), I wanted to add (smart_config) function, but the reason why (mesh route) was not applicable was that there might not be wifi in the expected environment, so I simply put (mesh no route) together with (smart_config) and many error logs occurred

### Code to reproduce this issue

https://gist.github.com/Gesiz/8fd27a1133931ee6cf0153cf2a43ea34
// If your code is longer than 30 lines, 
## Debug Logs

```
I (0) cpu_start: App cpu up.
I (412) heap_init: Initializing. RAM available for dynamic allocation:
I (419) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (425) heap_init: At 3FFB9D80 len 00026280 (152 KiB): DRAM
I (431) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (438) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (444) heap_init: At 40090034 len 0000FFCC (63 KiB): IRAM
I (450) cpu_start: Pro cpu start user code
I (133) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (234) wifi: wifi driver task: 3ffc2c80, prio:23, stack:3584, core=0
I (234) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (234) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (264) wifi: wifi firmware version: 0c3a77a
I (264) wifi: config NVS flash: enabled
I (264) wifi: config nano formating: disabled
I (264) wifi: Init dynamic tx buffer num: 32
I (274) wifi: Init data frame dynamic rx buffer num: 32
I (274) wifi: Init management frame dynamic rx buffer num: 32
I (284) wifi: Init management short buffer num: 32
I (284) wifi: Init static rx buffer size: 1600
I (294) wifi: Init static rx buffer num: 10
I (294) wifi: Init dynamic rx buffer num: 32
I (304) wifi: Set ps type: 0

I (364) phy: phy_version: 4008, 544f89f, Jan 24 2019, 14:54:06, 0, 0
I (364) wifi: mode : sta (24:6f:28:95:00:4c)
I (364) [mwifi, 261]: esp-mdf version: v1.0-beta1-63-g4d9e0f9
W (374) wifi: [beacon]interval:100ms
I (374) wifi: mode : sta (24:6f:28:95:00:4c) + softAP (24:6f:28:95:00:4d)
I (384) wifi: Total power save buffer number: 16
I (384) wifi: Init max length of beacon: 752/752
I (394) wifi: Init max length of beacon: 752/752
I (394) mesh: <nvs>read layer:0
I (394) mesh: <nvs>read assoc:0, err:0x1102
W (404) wifi: [beacon]interval:100ms
I (404) mesh: [CONFIG]invalid router settings, ssid_len:0, ssid:, bssid:00:00:00:00:00:00
I (1404) wifi: Set ps type: 0

D (1404) [mwifi, 131]: esp_mesh_event_cb event.id: 0
I (1414) [mwifi, 194]: MESH is started
I (1414) [sc, 60]: event_loop_cb, event: 0
I (1414) [sc, 64]: MESH is started
I (1714) mesh: [SCAN][ch:13]AP:3, other(ID:0, RD:0), MAP:1, idle:0, candidate:1, root:1, topMAP:0[c:0,i:0][00:00:00:00:00:00][FIXED-ROOT]<>
I (1714) mesh: 6563[selection]try rssi_threshold:-78, backoff times:0, max:5<-78,-82,-85>
I (1724) mesh: [DONE]connect to parent:ESPM_C11F78, channel:13, rssi:-22, 24:0a:c4:c1:1f:79[layer:1, assoc:0], my_vote_num:0/voter_num:0, rc[00:00:00:00:00:00/-120/0]
I (3234) wifi: n:13 2, o:13 0, ap:13 2, sta:13 2, prof:13
I (3234) wifi: state: init -> auth (b0)
I (4244) wifi: state: auth -> init (200)
I (4244) wifi: n:13 0, o:13 2, ap:13 2, sta:13 2, prof:13
D (4244) [mwifi, 131]: esp_mesh_event_cb event.id: 8
I (4244) [mwifi, 172]: Parent is disconnected, reason: 2
I (4244) wifi: n:13 2, o:13 0, ap:13 2, sta:13 2, prof:13
I (4244) mesh: [wifi]disconnected reason:2(auth expire), continuous:1/max:12, non-root, vote(,)<><>
I (4254) wifi: state: init -> auth (b0)
I (4264) wifi: state: auth -> init (0)
I (4274) wifi: n:13 0, o:13 2, ap:13 2, sta:13 2, prof:13
D (4274) [mwifi, 131]: esp_mesh_event_cb event.id: 8
I (4274) wifi: n:13 2, o:13 0, ap:13 2, sta:13 2, prof:13
I (4284) wifi: state: init -> auth (b0)
I (4284) [mwifi, 172]: Parent is disconnected, reason: 202
I (4254) [sc, 60]: event_loop_cb, event: 8
I (4294) mesh: [wifi]disconnected reason:202(), continuous:2/max:12, non-root, vote(,)<><>
I (4304) [sc, 72]: Parent is disconnected on station interface
I (4314) wifi: state: auth -> init (0)
I (4314) wifi: n:13 0, o:13 2, ap:13 2, sta:13 2, prof:13
D (4324) [mwifi, 131]: esp_mesh_event_cb event.id: 8
I (4324) wifi: n:13 2, o:13 0, ap:13 2, sta:13 2, prof:13
I (4334) wifi: state: init -> auth (b0)
I (4324) [mwifi, 172]: Parent is disconnected, reason: 202
I (4324) [sc, 60]: event_loop_cb, event: 8
I (4344) mesh: [wifi]disconnected reason:202(), continuous:3/max:12, non-root, vote(,)<><>
I (4354) [sc, 72]: Parent is disconnected on station interface
I (4364) wifi: state: auth -> init (0)
I (4364) wifi: n:13 0, o:13 2, ap:13 2, sta:13 2, prof:13
D (4374) [mwifi, 131]: esp_mesh_event_cb event.id: 8
I (4374) wifi: n:13 2, o:13 0, ap:13 2, sta:13 2, prof:13
I (4384) wifi: state: init -> auth (b0)
I (4374) [mwifi, 172]: Parent is disconnected, reason: 202
I (4394) mesh: [wifi]disconnected reason:202(), continuous:4/max:12, non-root, vote(,)<><>
I (4374) [sc, 60]: event_loop_cb, event: 8
I (4394) wifi: state: auth -> init (0)
I (4404) wifi: n:13 0, o:13 2, ap:13 2, sta:13 2, prof:13
D (4414) [mwifi, 131]: esp_mesh_event_cb event.id: 8
I (4414) wifi: n:13 2, o:13 0, ap:13 2, sta:13 2, prof:13
I (4424) wifi: state: init -> auth (b0)
I (4414) [mwifi, 172]: Parent is disconnected, reason: 202
I (4414) [sc, 72]: Parent is disconnected on station interface
I (4434) mesh: [wifi]disconnected reason:202(), continuous:5/max:12, non-root, vote(,)<><>
I (4444) [sc, 60]: event_loop_cb, event: 8
I (4454) wifi: state: auth -> init (0)
I (4454) wifi: n:13 0, o:13 2, ap:13 2, sta:13 2, prof:13
D (4464) [mwifi, 131]: esp_mesh_event_cb event.id: 8
I (4464) wifi: n:13 2, o:13 0, ap:13 2, sta:13 2, prof:13
I (4474) wifi: state: init -> auth (b0)
I (4464) [mwifi, 172]: Parent is disconnected, reason: 202
I (4464) [sc, 72]: Parent is disconnected on station interface
I (4474) mesh: [wifi]disconnected reason:202(), continuous:6/max:12, non-root, vote(,)<><>
I (4494) [sc, 60]: event_loop_cb, event: 8
I (4494) [sc, 72]: Parent is disconnected on station interface
I (4504) wifi: state: auth -> init (0)
I (4504) wifi: n:13 0, o:13 2, ap:13 2, sta:13 2, prof:13
D (4514) [mwifi, 131]: esp_mesh_event_cb event.id: 8
I (4514) wifi: n:13 2, o:13 0, ap:13 2, sta:13 2, prof:13
I (4524) wifi: state: init -> auth (b0)
I (4514) [mwifi, 172]: Parent is disconnected, reason: 202
I (4514) [sc, 60]: event_loop_cb, event: 8
I (4534) mesh: [wifi]disconnected reason:202(), continuous:0/max:12, non-root, vote(,)<><>
I (4544) [sc, 72]: Parent is disconnected on station interface
I (4554) wifi: state: auth -> init (0)
I (4554) wifi: n:13 0, o:13 2, ap:13 2, sta:13 2, prof:13
D (4564) [mwifi, 131]: esp_mesh_event_cb event.id: 8
I (4564) wifi: n:13 2, o:13 0, ap:13 2, sta:13 2, prof:13
I (4574) wifi: state: init -> auth (b0)
I (4564) [mwifi, 172]: Parent is disconnected, reason: 202
I (4564) [sc, 60]: event_loop_cb, event: 8
I (4584) mesh: [wifi]disconnected reason:202(), continuous:1/max:12, non-root, vote(,)<><>
I (4594) [sc, 72]: Parent is disconnected on station interface
I (4604) wifi: state: auth -> init (0)
I (4604) wifi: n:13 0, o:13 2, ap:13 2, sta:13 2, prof:13
D (4614) [mwifi, 131]: esp_mesh_event_cb event.id: 8
I (4614) wifi: n:13 2, o:13 0, ap:13 2, sta:13 2, prof:13
I (4624) wifi: state: init -> auth (b0)
I (4614) [mwifi, 172]: Parent is disconnected, reason: 202
I (4614) [sc, 60]: event_loop_cb, event: 8
...(loop)
```

dongdong004, 2020-01-15:
> When self organized networking is enabled, the ESP-MESH stack will internally make calls to Wi-Fi APIs. Therefore, the application layer should not make any calls to Wi-Fi APIs whilst self organized networking is enabled as doing so would risk interfering with ESP-MESH.

[The document](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/network/esp_mesh.html#self-organized-networking) said application should not call Wi-Fi APIs with self organized mesh.

I think smartconfig uses Wi-Fi APIs such as `esp_wifi_connect`.  Maybe you can disable self organized mesh when you use smartconfig.

Gesiz, 2020-01-19:
got it . thanks bro,I was so careless 0,0

> [The document](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/network/esp_mesh.html#self-organized-networking) said application should not call Wi-Fi APIs with self organized [...]

closed 2020-01-19 by Gesiz