espressif/esp-idf #4635: stdatomic.h incompatible with PSRAM

szmodz, 2020-01-16:
GCC atomic builtins are not compatible with PSRAM addresses, as they rely on s32c1i.

This in turn makes the compiler incompatible with the C standard (stdatomic.h), and will cause difficult to find problems with valid, standard C code.

Worst part is, using stdatomic.h with PSRAM will not cause any exceptions. It will simply result in difficult to find race conditions.

esp-idf is also affected by this issue. Examples include the ethernet and SPI drivers.

https://github.com/espressif/esp-idf/blob/647cb628a11a9ebd6952276fb7e6f354d4e3171d/components/esp_eth/src/esp_eth.c#L364

(no MALLOC_CAP_INTERNAL)

https://github.com/espressif/esp-idf/blob/647cb628a11a9ebd6952276fb7e6f354d4e3171d/components/esp_eth/src/esp_eth.c#L174

https://github.com/espressif/esp-idf/blob/647cb628a11a9ebd6952276fb7e6f354d4e3171d/components/driver/spi_master.c#L242

(no MALLOC_CAP_INTERNAL)

https://github.com/espressif/esp-idf/blob/647cb628a11a9ebd6952276fb7e6f354d4e3171d/components/driver/spi_master.c#L559

szmodz, 2020-01-16:
```
#include <stdatomic.h>

void xfaa(atomic_int *x) {
    atomic_fetch_add(x, 1);
}

```

> xtensa-esp32-elf-gcc -O2 -std=gnu11 -S test.c

```
	.file	"test.c"
	.text
	.align	4
	.global	xfaa
	.type	xfaa, @function
xfaa:
	entry	sp, 32
	memw
	l32i.n	a8, a2, 0
.L2:
	mov.n	a9, a8
	addi.n	a10, a8, 1
	wsr	a8, SCOMPARE1
	**s32c1i	a10, a2, 0**
	mov.n	a8, a10
	bne	a10, a9, .L2
	retw.n
	.size	xfaa, .-xfaa
	.ident	"GCC: (crosstool-NG esp-2019r2) 8.2.0"

```

https://github.com/espressif/esp-idf/blob/647cb628a11a9ebd6952276fb7e6f354d4e3171d/components/freertos/port.c#L439

suda-morris, 2020-01-16:
Thanks @szmodz Well spotted!

HudsonProdigy, 2020-01-22:
having issues building 3.3.1 and 4.0-rc getting all kinds of stdatomic.h errors. Im guessing it is related to this issue.
`In file included from /Users/user/esp/esp-idf-v4.0-rc/components/esp_eth/src/esp_eth.c:15:0:
/Users/user/esp/esp-idf-v4.0-rc/components/newlib/include/stdatomic.h: In function 'atomic_flag_test_and_set_explicit':
/Users/user/esp/esp-idf-v4.0-rc/components/newlib/include/stdatomic.h:271:31: error: request for member '__val' in something not a structure or union
  __atomic_exchange_n(&(object)->__val, desired, order)
                               ^
/Users/user/esp/esp-idf-v4.0-rc/components/newlib/include/stdatomic.h:387:10: note: in expansion of macro 'atomic_exchange_explicit'
  return (atomic_exchange_explicit(&__object->__flag, 1, __order));
          ^
/Users/user/esp/esp-idf-v4.0-rc/components/newlib/include/stdatomic.h:387:2: error: incompatible type for argument 1 of '__atomic_exchange_n'
  return (atomic_exchange_explicit(&__object->__flag, 1, __order));
  ^
/Users/user/esp/esp-idf-v4.0-rc/components/newlib/include/stdatomic.h: In function 'atomic_flag_clear_explicit':
/Users/user/esp/esp-idf-v4.0-rc/components/newlib/include/stdatomic.h:285:28: error: request for member '__val' in something not a structure or union
  __atomic_store_n(&(object)->__val, desired, order)
                            ^
/Users/user/esp/esp-idf-v4.0-rc/components/newlib/include/stdatomic.h:394:2: note: in expansion of macro 'atomic_store_explicit'
  atomic_store_explicit(&__object->__flag, 0, __order);
  ^
/Users/user/esp/esp-idf-v4.0-rc/components/newlib/include/stdatomic.h:394:2: error: incompatible type for argument 1 of '__atomic_store_n'
/Users/user/esp/esp-idf-v4.0-rc/components/esp_eth/src/esp_eth.c: In function 'esp_eth_driver_install':
/Users/user/esp/esp-idf-v4.0-rc/components/newlib/include/stdatomic.h:91:47: error: request for member '__val' in something not a structure or union
 #define atomic_init(obj, value)  ((void)((obj)->__val = (value)))
                                               ^
/Users/user/esp/esp-idf-v4.0-rc/components/esp_eth/src/esp_eth.c:174:5: note: in expansion of macro 'atomic_init'
     atomic_init(&eth_driver->ref_count, 1);
     ^
/Users/user/esp/esp-idf-v4.0-rc/components/esp_eth/src/esp_eth.c: In function 'esp_eth_driver_uninstall':
/Users/user/esp/esp-idf-v4.0-rc/components/newlib/include/stdatomic.h:283:27: error: request for member '__val' in something not a structure or union
  __atomic_load_n(&(object)->__val, order)
                           ^
/Users/user/esp/esp-idf-v4.0-rc/components/esp_eth/src/esp_eth.c:27:15: note: in definition of macro 'ETH_CHECK'
         if (!(a))                                                                 \
               ^
/Users/user/esp/esp-idf-v4.0-rc/components/newlib/include/stdatomic.h:365:2: note: in expansion of macro 'atomic_load_explicit'
  atomic_load_explicit(object, memory_order_seq_cst)
  ^
/Users/user/esp/esp-idf-v4.0-rc/components/esp_eth/src/esp_eth.c:215:15: note: in expansion of macro 'atomic_load'
     ETH_CHECK(atomic_load(&eth_driver->ref_count) == 1,
               ^
/Users/user/esp/esp-idf-v4.0-rc/components/esp_eth/src/esp_eth.c:215:5: error: incompatible type for argument 1 of '__atomic_load_n'
     ETH_CHECK(atomic_load(&eth_driver->ref_count) == 1,
     ^
In file included from /Users/user/esp/esp-idf-v4.0-rc/components/esp_eth/src/esp_eth.c:15:0:
/Users/user/esp/esp-idf-v4.0-rc/components/esp_eth/src/esp_eth.c: In function 'esp_eth_increase_reference':
/Users/user/esp/esp-idf-v4.0-rc/components/newlib/include/stdatomic.h:273:30: error: request for member '__val' in something not a structure or union
  __atomic_fetch_add(&(object)->__val, operand, order)
                              ^
/Users/user/esp/esp-idf-v4.0-rc/components/newlib/include/stdatomic.h:355:2: note: in expansion of macro 'atomic_fetch_add_explicit'
  atomic_fetch_add_explicit(object, operand, memory_order_seq_cst)
  ^
/Users/user/esp/esp-idf-v4.0-rc/components/esp_eth/src/esp_eth.c:341:5: note: in expansion of macro 'atomic_fetch_add'
     atomic_fetch_add(&eth_driver->ref_count, 1);
     ^
/Users/user/esp/esp-idf-v4.0-rc/components/esp_eth/src/esp_eth.c:341:5: error: incompatible type for argument 1 of '__atomic_fetch_add'
In file included from /Users/user/esp/esp-idf-v4.0-rc/components/esp_eth/src/esp_eth.c:15:0:
/Users/user/esp/esp-idf-v4.0-rc/components/esp_eth/src/esp_eth.c: In function 'esp_eth_decrease_reference':
/Users/user/esp/esp-idf-v4.0-rc/components/newlib/include/stdatomic.h:279:30: error: request for member '__val' in something not a structure or union
  __atomic_fetch_sub(&(object)->__val, operand, order)
                              ^
/Users/user/esp/esp-idf-v4.0-rc/components/newlib/include/stdatomic.h:361:2: note: in expansion of macro 'atomic_fetch_sub_explicit'
  atomic_fetch_sub_explicit(object, operand, memory_order_seq_cst)
  ^
/Users/user/esp/esp-idf-v4.0-rc/components/esp_eth/src/esp_eth.c:352:5: note: in expansion of macro 'atomic_fetch_sub'
     atomic_fetch_sub(&eth_driver->ref_count, 1);
     ^
/Users/user/esp/esp-idf-v4.0-rc/components/esp_eth/src/esp_eth.c:352:5: error: incompatible type for argument 1 of '__atomic_fetch_sub'
/Users/user/esp/esp-idf-v4.0-rc/components/esp_eth/src/esp_eth.c: At top level:
cc1: warning: unrecognized command line option '-Wno-frame-address'
make[2]: *** [esp-idf/esp_eth/CMakeFiles/__idf_esp_eth.dir/src/esp_eth.c.obj] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [esp-idf/esp_eth/CMakeFiles/__idf_esp_eth.dir/all] Error 2
make: *** [all] Error 2
make failed with exit code 2`

szmodz, 2020-01-22:
Build errors are unrelated. This is entirely run-time.

Mras2an, 2020-09-09:
Hello,

I think I have this issue on esp-idf 4.0 with a ESP32-WROVER (8mb of RAM)

In file included from /tmp/esp-homekit-sdk/esp-idf/components/esp_eth/src/esp_eth.c:15:0:
/tmp/esp-homekit-sdk/esp-idf/components/newlib/include/stdatomic.h: In function 'atomic_flag_test_and_set_explicit':
/tmp/esp-homekit-sdk/esp-idf/components/newlib/include/stdatomic.h:271:31: error: request for member '__val' in something not a structure or union

Is there any news on it?

thank you