espressif/esp-idf #4636: Mesh implementation using nimBLE host stack

AlpinProgrammer, 2020-01-16:
Hello all,

I would like to know more about the status of nimBLE mesh implementations.

We are considering this option to develop the BLE mesh functionality with hard constraints of resources (low RAM) and Bluedroid-based mesh won't fit.

Would you advice using nimBLE stack to implement Mesh ? Are there larger functionalities to be added in esp-idf in the future ?

Thanks in advance for your insights,
Regards.

dhrishi, 2020-01-16:
Hi @AlpinProgrammer, as of now there are 2 options you can choose from:

1. NimBLE BLE Mesh (part of and maintained by NimBLE upstream code, which works directly in ESP-IDF). See here: https://mynewt.apache.org/latest/network/mesh/index.html#bluetooth-mesh

2. ESP-BLE-Mesh with NimBLE host. Details here: https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/esp-ble-mesh/ble-mesh-index.html

Alvin1Zhang, 2020-03-16:
Thanks for reporting.

closed 2020-03-16 by Alvin1Zhang

AlpinProgrammer, 2020-03-17:
Thank you for your help !