espressif/esp-idf #4638: BLE Causing abort() on core 0

jonhunterking66, 2020-01-16:
## Environment

- Development Kit:      ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): v4
- Module or chip used:  ESP32-WROOM-32D
- IDF version (run ``git describe --tags`` to find it):
    v4.1-dev-1086-g93a8603c5
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System:     Linux
- Using an IDE?: Visual Studio Code 
- Power Supply:         USB

## Problem Description

BLE module causes abort() was called at PC 0x40082ea6 on core 0
0x40082ea6: lock_acquire_generic at /home/jonhunterking/esp/esp-idf/components/newlib/locks.c:143 when I read values from my flash. This happens after case 7 completes. 

### Code to reproduce this issue

Please note that not all the code from each of my files is posted here. Its thousands of lines. This is just the relevant code. So if you think you need more let me know. 

MAIN CODE

```cpp
void app_main()
{
  esp_log_level_set(TAG, ESP_LOG_DEBUG);
  createWifiStruct();
  bleInit();
  xTaskCreate(&main_task, "MAIN", 1024 *15, NULL, 2, NULL);
}

void main_task(void * params){
    uint64_t sleepTime = 5000000;
    int x = 4;
    uint32_t i = 0;
    //char * ssid = "VENTURES";
    //char * password = "2y!ardTypecolumn1";
    int flashStatus = 0;
    bool calibrationStatus = false;
    bool bleStatus = false;
    char * ssid_test;
    char * password_test;
    while(true)
    {
        printf("Oto state is %d\n", x);
        printf("In Main task\n");
        switch(x){
            case 0:
                ESP_LOGI(TAG, "Sending Indiciation");
                ble_indicateTwo(i);
                i++;
                if (i == 360){
                    i = 3;
                }
                calibrationStatus = returnStatus();
                if (calibrationStatus){
                    ESP_LOGI(TAG, "Calibration Complete");
                    x = 3;
                }
                vTaskDelay(100/portTICK_PERIOD_MS);
                break;
            case 1:
                ESP_LOGI(TAG, "Asking for password and SSID");
                ssid_test = returnSSID();
                password_test = returnPassword();
                printf("Returned SSID is: %s\n", ssid_test);
                printf("Returned Password is: %s\n", password_test);
[... 207 lines cut ...]

0x40082fc9: _lock_acquire_recursive at /home/jonhunterking/esp/esp-idf/components/newlib/locks.c:171

0x4019b971: _vfiprintf_r at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/vfprintf.c:853 (discriminator 2)

0x40193fe1: fiprintf at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdio/fiprintf.c:48

0x40193f69: __assert_func at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/assert.c:58 (discriminator 8)

0x4009abb5: vPortCPUAcquireMutexIntsDisabledInternal at /home/jonhunterking/esp/esp-idf/components/freertos/portmux_impl.inc.h:105
 (inlined by) vPortCPUAcquireMutexIntsDisabled at /home/jonhunterking/esp/esp-idf/components/freertos/portmux_impl.h:100
 (inlined by) vTaskEnterCritical at /home/jonhunterking/esp/esp-idf/components/freertos/tasks.c:4200

0x400d8b1c: periph_module_enable at /home/jonhunterking/esp/esp-idf/components/driver/periph_ctrl.c:33 (discriminator 2)

0x400df376: esp_phy_rf_init at /home/jonhunterking/esp/esp-idf/components/esp_wifi/src/phy_init.c:163

0x400df469: esp_modem_sleep_exit at /home/jonhunterking/esp/esp-idf/components/esp_wifi/src/phy_init.c:324

0x400e142b: btdm_sleep_exit_phase3_wrapper at /home/jonhunterking/esp/esp-idf/components/bt/controller/bt.c:873

0x4011af96: btdm_controller_task at ??:?

0x40099745: vPortTaskWrapper at /home/jonhunterking/esp/esp-idf/components/freertos/port.c:143
```

bitmakk, 2020-01-17:
I am not sure, but if I remember correctly, the flash operations is the main cause we use IRAM_ATTR modifier on every task we create. In my opinion your task is in critical state, and locked the flash (because it is not running from the Instruction RAM), but you try to use the (same locked) flash.
Have a look at this: https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/general-notes.html

jonhunterking66, 2020-01-21:
Was not able to fix issue. Work around was to not use structures but to save my two char strings and my int individually in flash.

closed 2020-01-21 by jonhunterking66

Alvin1Zhang, 2020-01-21:
@jonhunterking66 Thanks for reporting and sharing the updates.

svdrummer, 2021-09-06:
Surprised you can get thousands of lines. ESP32 is 75% full just from the library and 20 lines of code.