espressif/esp-idf #4640: esp_websocket_client    example      The connection to the server is normal, and the server can receive the information sent by the server, but the server cannot receive the information reported by the ESP32.

chenxiahuaxu, 2020-01-17:
esp32 runs normally without error prompts. However, the server cannot receive the message reported by esp32. It is normal for the server to test with other tools.

There is no error message on the ESP32 side. The sending procedure is as follows:
    char data[32];
    int i = 0;
    while (1)
    {
        if (esp_websocket_client_is_connected(client))
        {
            int len = sprintf(data, "hello %04d", i++);
            ESP_LOGI(TAG, "Sending:%s", data);
            int err = esp_websocket_client_send_bin(client, data, len, portMAX_DELAY);
            if (err == ESP_FAIL)
                printf("ESP_FAIL\r\n");
            else
                printf("len:%d\r\n",err);
        }
        vTaskDelay(1000 / portTICK_RATE_MS);
    }
//////////////////////////////////////////////////////////////////////////////////////////////////////////////
ESP32 side running log is as follows:
W (252089) user_main: Received=kkjk
W (252099) user_main: Total payload length=4, data_len=4, current payload offset=0

I (252599) user_main: WEBSOCKET_EVENT_DATA
I (252599) user_main: Received opcode=1
W (252599) user_main: Received=kkjk
W (252609) user_main: Total payload length=4, data_len=4, current payload offset=0

I (252759) user_main: Sending:hello 0245
len:10
I (253759) user_main: Sending:hello 0246
len:10
///////////////////////////////////////////////////////////////////////////////////////////////////////////////
The server-side operation log is as follows:
<----- 接收：2020-01-17 14:28:18
连接：ws://192.168.191.3:8443/ - 192.168.191.7:61271
<----- 接收：2020-01-17 14:28:18
connection:Upgrade
host:192.168.191.3:8443
sec-websocket-key:HudC7xAIDs7v7lNg99CDdg==
sec-websocket-version:13
upgrade:websocket
user-agent:ESP32 Websocket Client
-----> 发送：2020-01-17 14:32:22
192.168.191.7:61271: kkjk
-----> 发送：2020-01-17 14:32:23
192.168.191.7:61271: kkjk
-----> 发送：2020-01-17 14:32:23
192.168.191.7:61271: kkjk

david-cermak, 2020-01-17:
Hi @chenxiahuaxu 

Could you please share the exact version of the esp-idf you use? There's an automated test in our CI that checks basic send/receive operation every master commit, so it seems unlikely to have the  ws-send broken. Also as you're saying there's no error message in the ESP log then the server must accept the TCP packet (assuming that the packet was sent).
* Could you please post the ESP log when debug level enabled?
* Also it would be good to have a wireshark log of the communication, as as:
[image]

chenxiahuaxu, 2020-01-17:
> 嗨@chenxiahuaxu
> 
> 您能否分享使用的esp-idf的确切版本？我们的CI中有一个自动测试，可以检查每个主提交的基本发送/接收操作，因此ws-send似乎不太可能中断。同样，正如您所说的，ESP日志中没有错误消息，则服务器必须接受TCP数据包（假设已发送数据包）。
> 
> * 启用调试级别后，能否请您发布ESP日志？
> * 同样，建立通信的wireshk日志也将是一件好事，例如：
>   [image]

I used the master version. I later confirmed that the message server sent by esp32 received it and successfully returned it. But the server is not displayed. I have used at least three tools to test that the server can display the reported messages, but the esp32 messages cannot be displayed.

chenxiahuaxu, 2020-01-17:
> Hi @chenxiahuaxu
>
> Could you please share the exact version of the esp-idf you use? There's an automated test in our CI that checks basic send/receive operation every master commit, so it seems [...]

I used the master version. I later confirmed that the message server sent by esp32 received it and successfully returned it. But the server is not displayed. I have used at least three tools to test that the server can display the reported messages, but the esp32 messages cannot be displayed.

david-cermak, 2020-01-17:
So do I understand correctly the the messages are correctly sent by the websocket client and received by the server?

What do you mean by "not displayed"? What kind of websocket server you use?

chenxiahuaxu, 2020-01-18:
> So do I understand correctly the the messages are correctly sent by the websocket client and received by the server?
> 
> What do you mean by "not displayed"? What kind of websocket server you use?
///////////////////////////////////////////////////////////////////////////////////////////////
Download address of the server I use：（https://gitee.com/openownworld/WebsocketMan）
//////////////////////////////////////////////////////////////////////////////////////////////////
I chose the postback function on the server:
[image]
/////////////////////////////////////////////////////////////////////////////////////////////////
Use（http://www.websocket-test.com/） to test the server
[image]
The server received the message, successfully parsed it, and displayed it in a window. The client received a message from the server
////////////////////////////////////////////////////////////////////////////////////////////
Use （http://coolaf.com/tool/chattest） to test the server
[image]
The server received the message, successfully parsed it, and displayed it in a window. The client received a message from the server
/////////////////////////////////////////////////////////////////////////////////////////////////////////
Use （http://tongxinmao.com/App/Detail/id/49#） to test the server
[image]
The server received the message, successfully parsed it, and displayed it in a window. The client received a message from the server
////////////////////////////////////////////////////////////////////////////////////////////////////////////
The above is my test process
Next I used esp32 to connect to the server to test
[image]
The server receives the message and the client receives a message from the server. But the server's display window did not parse out the message reported by esp32。
This is questionable to me, I don't know which side has the bug。
At least the messages reported by the several testing tools I used above can be parsed by the server and displayed on the window。
esp32 messages can also be received, but they cannot be parsed

david-cermak, 2020-01-29:
Hi @chenxiahuaxu 

Sorry for the late reply, I cannot tell why the particular message is not displayed on your server GUI, but as if you're saying it's echo'ed back and received correctly that the server must treat it as a correct websocket message.

As for the display issue, have you tried 
* to use the text interface `esp_websocket_client_send_text()` for sending the message? (maybe the server prints only messages that are marked as TEXT)
* to add some line separators at the end of the message?  e.g. `sprintf(data, "hello %04d\r\n", i++);`

chegewara, 2020-02-08:
Hi, i am starting to experiment with websockets and i dont know if this is the case, but i found that my websockets server is receiving data from esp32 as binary data (at least when i resend message from esp32 client in browser displays as binary data):
```
00000000: 6865 6c6c 6f20 3030 3039                 hello 0009
```
Maybe your server cant parse such message. This is how javascript is parsing it:
```
RESPONSE: WebSocket rocks // normal text message

RESPONSE: [object Blob]  // text message from esp32 received as binary data
```

chenxiahuaxu, 2020-02-08:
> Hi, i am starting to experiment with websockets and i dont know if this is the case, but i found that my websockets server is receiving data from esp32 as binary data (at least when i resend message from esp32 client in browser displays as binary data):
> 
> ```
> 00000000: 6865 6c6c 6f20 3030 3039                 hello 0009
> ```
> 
> Maybe your server cant parse such message. This is how javascript is parsing it:
> 
> ```
> RESPONSE: WebSocket rocks // normal text message
> 
> RESPONSE: [object Blob]  // text message from esp32 received as binary data
> ```
> 
> PS adding trailing 0x0 does not help

Thank you for your reply. After reading Espressif's IDF source code, I have found this problem and reflected it to the server provider. The server is not compatible with binary data parsing and has been fixed. The default sending format of Espressif's SDK is binary file data. It also has a function for sending text data. I did not check the API carefully, which caused this problem.

chegewara, 2020-02-08:
Yes, youre correct, i didnt study API too and didnt found additional functions.
Maybe it would be good to add some info in example README?

chenxiahuaxu, 2020-02-08:
> Yes, youre correct, i didnt study API too and didnt found additional functions.
> Maybe it would be good to add some info in example README?
When reading the demo, I didn't see any instructions. The sending function used in the demo is sending in binary form. I saw this problem only when checking the source code. The api has three send functions, namely send, send_bin and send_text, where the send and send_bin functions are the same

chenxiahuaxu, 2020-02-08:
> Yes, youre correct, i didnt study API too and didnt found additional functions.
> Maybe it would be good to add some info in example README? Or changing `esp_websocket_client_send` to send uint8_t data, since its:
> 
> > Generic write data to the WebSocket connection; defaults to binary send
> 
> https://github.com/espressif/esp-idf/blob/release/v4.0/components/esp_websocket_client/include/esp_websocket_client.h#L143-L154

When reading the demo, I didn't see any instructions. The sending function used in the demo is sending in binary form. I saw this problem only when checking the source code. The api has three send functions, namely send, send_bin and send_text, where the send and send_bin functions are the same

david-cermak, 2020-03-05:
@chenxiahuaxu Any suggestion what could be changed or could the issue be closed?
The recommendation to use `uint8_t` for binary data makes sense, but we would like to keep it backward compatible in `v4.x`. Could be updated in `v5.0` though.

chegewara, 2020-03-05:
@david-cermak I think that websocket example can be updated to use `esp_websocket_client_send_text` instead of generic `esp_websocket_client_send` or some comments in example to point that `esp_websocket_client_send` is sending binary data, not string/text/char.

closed 2020-03-10 by espressif-bot